realsenseai/librealsense #9925: How to

brunovollmer, 2021-11-05:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 
| Firmware Version                   | 2.49.0 | 
| Operating System & Version |   Mendel Linux  | 
| Kernel Version (Linux Only)    |  4.14.98-imx                                       | 
| Platform                                 | Coral Dev Board |

### Issue Description
Hey everybody,

I was wondering what the suggested way is to use the compiled version of pyrealsense2 on multiple devices with exactly the same architecture. I compiled everything on one of our Coral Dev Boards and everything works great. Now we have to setup a bunch more and I was wondering if there is a way to avoid the compilation process every time.

So far I've tried to copy only the files in the `site-packages/pyrealsense2` folder but then some .so libraries are missing. If I copy them as well to the right location it seems something else is off. 

I'm not an expert in C++ and Linux so I was wondering if there is an easier way (e.g. create a python wheel)?

MartyG-RealSense, 2021-11-05:
Hi @brunovollmer   Would it be possible for you to create an OS backup image of the Coral board that could be written to an SD card so that the complete image could be flashed onto the other boards with a tool such as BalenaOS?  If so, the link below provides instructions for how this might be accomplished.

https://stackoverflow.com/questions/60894434/how-do-i-boot-into-mendel-os-on-the-sdcard-with-the-coral-dev-board

BalenaOS provide a Coral Dev Board guide for deploying an image to multiple devices by defining a device 'fleet'.

https://www.balena.io/docs/learn/getting-started/coral-dev/python/

brunovollmer, 2021-11-08:
Hey @MartyG-RealSense,
thanks for the quick reply. Your first idea makes sense to me. What I'm wondering though is how to integrate the intelrealsense library with the image that move to the sdcard?

Bruno

MartyG-RealSense, 2021-11-08:
A complete backup image of a system should also contain whatever is installed on it as well as the OS and its settings.  Intel demonstrate this principle in their open-source networking white paper, where they provide a complete image file for Raspberry Pi 4 that installs the SDK and all the networking-related files when the image is flashed onto a Pi's SD card.

https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras#section-2-3-preparing-the-sd-card

MartyG-RealSense, 2021-11-14:
Hi @brunovollmer  Do you require further assistance with this case, please?  Thanks!

brunovollmer, 2021-11-15:
@MartyG-RealSense I will look further into this in the future. Thanks

closed 2021-11-15 by brunovollmer