realsenseai/librealsense #9926: box_dimensioner_multicam python does not capture the volume of the box

ivanvalverdebell, 2021-11-05:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 2.49.0 | 
| Operating System & Version |  Windows 10 |                                    
| Platform                                 | PC  |
| SDK Version                            |  2.0  | 
| Language                              |  python |             

**Issue Description**

Hey everybody,

I am working with an L515 camera with the example of box_dimensioner_multicam in python. When I start the program I get the following calibration error:

**RMS error for calibration with device number f1180787 is : 0.00466779243858435 m**

Then the demo is run but it does not capture the volume of the box.
Is it due to the calibration error I mentioned earlier?

MartyG-RealSense, 2021-11-05:
Hi @ivanvalverdebell   Could you try moving the camera further away from the observed box, please?  Problems can occur if the camera is too close to the box.

The correct procedure for using box_dimensioner_multicam is to point the camera at an empty checkerboard surface and then launch the application.  The camera(s) will automatically calibrate and then ask you to place your box onto the checkerboard.  So it is important that there is not a box currently on the checkerboard when the application is launched.

Also, if you are using a checkerboard with a custom size that is larger than 6x9" then the checkerboard size values in the script should be edited.

ivanvalverdebell, 2021-11-05:
> Hi @ivanvalverdebell Could you try moving the camera further away from the observed box, please? Problems can occur if the camera is too close to the box.
>
> The correct procedure for using [...]

I tried moving the camera, moving it away, bringing it closer, but still I keep getting that error.
Could it be because I am only using one camera?

MartyG-RealSense, 2021-11-05:
Despite its name, box_dimensioner_multicam can work with a single camera.

ivanvalverdebell, 2021-11-09:
I am currently working this way and it keeps throwing me the error:

**RMS error for calibration with device number f1180787 is : 0.00466779243858435 m**

any suggestions? I have changed the angles, heights of the camera and it does not capture the volumetry.
I attach a picture of the camera settings

[image]

jerrydegroot, 2021-11-10:
The line you are getting is not an error, it is specifying that the calibration has succeeded and that the remaining difference between "true" values and "measured" values (also known as "Root Mean Square error": "RMS error") is the value specified. In your case 0.004657... meters (that is: 4.66 mm). 

So, after this message, the program will ask you to place an object on the checkerboard. You will notice that a new window has opened (you have to select it yourself), and in that window you see the camera view of the checkerboard. When you place an object on top of the checkerboard, the program should draw a green bounding box around the object. However, this may not be working for the L515 camera - this is the problem we are currently having.

ivanvalverdebell, 2021-11-10:
> The line you are getting is not an error, it is specifying that the calibration has succeeded and that the remaining difference between "true" values and "measured" values (also known as "Root Mean [...]

Is it feasible or safe for it to work with the D400 series?
for example: CAMERA D435

MartyG-RealSense, 2021-11-10:
Hi @ivanvalverdebell  The box_dimensioner_multicam program was originally created for the 400 Series cameras so yes, it will work with those camera models.

jerrydegroot, 2021-11-10:
Hello @MartyG-RealSense,

Although the code runs for L515, we found that the chessboard plane was at z=2.4 meters instead of at z=0. As a consequence the box placed on the chessboard was not found at z<-depth_threshold.

We seem to get it to work for the L515 by changing helper_functions.py line 180  
(in "def convert_depth_frame_to_pointcloud(depth_image, camera_intrinsics )" :

old:   z = depth_image.flatten() / 1000
new:  z = depth_image.flatten() / 4000

We only don't understand why this works... Do you have any ideas why this fix works?

MartyG-RealSense, 2021-11-10:
Thanks very much @jerrydegroot for sharing your fix!

I do not have a specific explanation for why the change would make a difference, though in general the L515 lidar camera does have some characteristics that provide different performance to a 400 Series stereo depth camera observing the same scene.  As one example, the L515 can see straight through transparent surfaces such as empty plastic bottles to the background detail behind it without registering the transparent surface.  Using the **Short Range** preset to reduce the camera's Laser Power and Receiver Gain values can make a positive difference.

ivanvalverdebell, 2021-11-10:
> Hello @MartyG-RealSense,
>
> Although the code runs for L515, we found that the chessboard plane was at z=2.4 meters instead of at z=0. As a consequence the box placed on the chessboard was not found [...]

Hey! your solution just worked for me, thanks for the help

jerrydegroot, 2021-11-12:
z = depth_image.flatten()/4000  gives "reasonable" results for single L515 camera setup (the bounding box is too large)

We expect that multicam setup will not work for L515 cameras in default asynchronous mode: both cameras will emit IR pulses at the same time, thus interfering each other's depth readings. There is a [whitepaper ](https://dev.intelrealsense.com/docs/lidar-camera-l515-multi-camera-setup) on using multiple L515 with a hardware trigger/sync. cable which should work for static object/scenery measurements. We are looking for a L515 multicam setup for dynamic object/scenery. Any suggestions welcome.

MartyG-RealSense, 2021-11-12:
Hi @jerrydegroot  You may be able to reduce multiple-camera L515 interference without using hardware sync if you can space the cameras apart so that their fields of view do not overlap, as shown in the illustration from the paper below.

[image]

MartyG-RealSense, 2021-11-26:
Hi @ivanvalverdebell  and @jerrydegroot  Do you require further assistance with this case, please?  Thanks!

closed 2021-12-02 by MartyG-RealSense

MartyG-RealSense, 2022-12-10:
Hi @Ankush1909IIT  I responded to your question at https://github.com/IntelRealSense/librealsense/issues/11197#issuecomment-1345192312 - thanks!