realsenseai/librealsense #9931: RealSense error calling: failed Last Error: Protocol error

zxl5344, 2021-11-07:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 01.05.08.01 | 
| Operating System & Version |    Linux (Ubuntu 20.04.3 LTS)  | 
| Kernel Version (Linux Only)    |  5.11.0-38-generic                                         | 
| Platform                                 | Dell Laptop  |
| SDK Version                            |  { legacy / 2.<29>.<0> }                      | 
| Language                            |  {C/python/ }                          | 
| Segment			|  Robot                        | 

### Issue Description
When I turn on L500 depth sensor, I got error: 

Backend in rs2_get_option(options:0x564c57dc0bd0, option:Receiver Gain):
get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Protocol error

How do I solve it?

MartyG-RealSense, 2021-11-08:
Hi @zxl5344   Could you unplug the micro sized end of the USB cable from the base of the camera, turn the connector around the opposite way and re-insert it into the base of the camera, please (USB Type-C cables are two way insertion at the micro sized end).  Does the error still occur after doing so?

zxl5344, 2021-11-08:
I turn the USB Type-C and re-insert it into the base. I still get same error

Backend in rs2_get_option(options:0x7f8d94001060, option:Receiver Gain):
set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Protocol error

MartyG-RealSense, 2021-11-08:
There was a previous case in https://github.com/IntelRealSense/realsense-ros/issues/1844 where this prootcol error occurred with an L515 when using the camera in a Docker container.  Are you using Docker, please?

zxl5344, 2021-11-08:
I don't think so. My laptop only installed Ubuntu.

MartyG-RealSense, 2021-11-08:
Error messages involving **xioctl(UVCIOC_CTRL_QUERY)** tend to be related to a conflict with the kernel.  The RealSense SDK does not yet officially support kernel 5.11 (the newest supported is currently 5.8 at the time of writing this) and so although the SDK can work with newer unsupported kernels, there may be unpredictable consequences in terms of stability.

[image]

Which method did you use to build the RealSense SDK on Linux, please?  If you could build the SDK from source code using the **RSUSB backend** method then this would help to confirm or eliminate the possibility of the problem being caused by a kernel issue.  This is because an RSUSB build is not dependent on Linux versions or kernel versions and does not require patching.

zxl5344, 2021-11-08:
I follow the distribution_linux.md in this repo to build the SDK. Do you have tutorial for RSUSB backend method?

MartyG-RealSense, 2021-11-09:
1.  Obtain the source code for the librealsense SDK.  You can do this by going to the 'Assets' file list at the bottom of the listing for each version on the SDK **Releases** page and left-clicking on the 
to download it as a zipped file.

https://github.com/IntelRealSense/librealsense/releases

[image]

2.  Extract the zipped file to obtain the librealsense SDK folder and go to the folder's 'root' directory.  This is the one with a list of files and folders in, and will typically be at the folder location **librealsense > librealsense**

[image]

3.  From this root directory with the CMakeLists.txt file in, input the command **mkdir build && cd build** to create a sub-directory called 'build' and navigate to it.

4.  Input the command below to use CMake to build the librealsense SDK from source code with example programs and tools such as the RealSense Viewer included.

**cmake ../ -DFORCE_RSUSB_BACKEND=ON  -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true**

5.  Finally, input the command below.

**sudo make uninstall && make clean && make -j8 && sudo make install**

zxl5344, 2021-11-09:
I followed your procedure @MartyG-RealSense. After step 5, I tried to run **realsense-viewer** but I got: **bash: /usr/bin/realsense-viewer: No such file or directory**. How to run it?

MartyG-RealSense, 2021-11-09:
On Linux, realsense-viewer and also the other example programs and tools should be located in the **usr/local/bin** directory if it as been installed successfully.  Do you have realsense-viewer in this folder, please?

zxl5344, 2021-11-09:
Yes, I have realsense-viewer in this folder. What is the next step?

MartyG-RealSense, 2021-11-09:
If you have navigated to this folder, are you able to launch realsense-viewer from there?

zxl5344, 2021-11-09:
No. I **cd /usr/local/bin** first and run **realsense-viewer**. I still got error: 

bash: /usr/bin/realsense-viewer: No such file or directory

zxl5344, 2021-11-09:
I reboot my laptop. And now realsense-viewer is working. But when I turn on L500 depth sensor, I got error:

UNKNOWN in rs2_get_option(options:0x7fdec00473a0, option:Pre Processing
Sharpening):
usb device disconnected

When I turn on RGB camera first and then turn on L500 depth sensor, I got error:

UNKNOWN in rs2_get_option(options:0x56531cdc81c0, option:Laser Power):
usb device disconnected

I turned the USB Type-C and I still got same error. Everytime I got error, system said **L515 is disconnected** and then connected again.

MartyG-RealSense, 2021-11-09:
After having gone through the diagnostic process during this case and considered all of the information, it is likely that there is a hardware fault in your camera and it should be returned to the retailer that you purchased it from using the **Return Material Authorization (RMA)** returns procedure of that particular retailer.

zxl5344, 2021-11-09:
Thanks for your help.

zxl5344, 2021-11-09:
Sorry for reopen it. One of my lab mate use this camera previously on a windows laptop. And it is working well. So I think it should be a software issues. What are the possible reason that L515 is working on windows but not Linux?

MartyG-RealSense, 2021-11-10:
It could be an issue with the USB port hardware on the Linux computer.  Not all USB ports are created equally, and a camera that works fine on one computer may have problems with some or all of the ports on a different make / model of computer.  Different computers may use USB controllers by different providers that may have a bearing on how well the camera works with it.  You could test whether it is an issue with the controller on the Linux computer by attaching the camera to a USB hub if it is not already, as the hub's ports will have their own independent USB controller separate from the computer's ports.

zxl5344, 2021-11-10:
I directly connect my camera on laptop without a hub and now I can turn on L500 sensor. Connection changed from USB 3.2 to USB 2.1. 
But I continuous get this warning when I turn on L500:

09:16:50 [Warn] /home/zxl5344/Realsense SDK/librealsense-2.49.0/src/libusb/messenger-libusb.cpp:42 - control_transfer returned error, index: 768, error: No data available, number: 61

How to solve it? I got same error when I use roslunch to lunch camera

MartyG-RealSense, 2021-11-10:
If you can access the camera in the RealSense Viewer, does it make a difference if you reset the camera from within the Viewer by going to the **More** option in the options side-panel and selecting **Hardware Reset** from its drop-down menu.

[image]

zxl5344, 2021-11-10:
No and No. I still get same error and USB 2.1

MartyG-RealSense, 2021-11-11:
In the Viewer, does the number of **control_transfer returned** errors reduce if you set the Preset to **Short Range** instead of the default Max Range?

zxl5344, 2021-11-11:
No, I feel like the frequency of **control_transfer returned** errors are almost same.

MartyG-RealSense, 2021-11-11:
RealSense SDK version **2.50.0** is now available.  Could you test whether the issues that you have been experiencing in librealsense 2.49.0 are also present in 2.50.0 please?

https://github.com/IntelRealSense/librealsense/releases/tag/v2.50.0

zxl5344, 2021-11-11:
All my codes and repo are based on ROS noetic so I don't want to upgrade to 2.50.0. 

My laptop only have type C port so I am using a type C to type C data cable now. Is it possible that my error and usb 2.1 is caused by data cable? 

Also I plan to set up my camera a little far from my laptop. Do you have any recommendation for a long cable, approximately 10 meter?

MartyG-RealSense, 2021-11-11:
It's fine, I totally understand why using 2.50.0 with the current ROS1 wrapper designed for 2.48.0 would be impractical for you.

When choosing your own USB cable there is a risk that it can lead to reduced performance or disconnections if it is not a high quality cable (for example, if it is a cable designed for device charging instead of high volume data transfer).

The company **Newnex** offer high quality USB cables that are validated for use with RealSense.  They have been featured in the official Intel data sheet document for the 400 Series cameras as a recommended supply source for USB cables for RealSense.

https://www.newnex.com/realsense-3d-camera-connectivity.php

zxl5344, 2021-11-11:
Thank you. I will buy a new USB cable and check if it will solve my issues.

MartyG-RealSense, 2021-11-11:
Thanks very much.  Please do update here when you have been able to test with a new cable.  Good luck!

MartyG-RealSense, 2021-11-15:
A ROS1 wrapper for use with Noetic and SDK 2.50.0 is now available.

https://github.com/IntelRealSense/realsense-ros/releases/tag/2.3.2

zxl5344, 2021-11-18:
I have solved my issue with a new cable. Also, could you show me how to upgrade SDK to latest version? I tried sudo apt-get but it didn't work. I guess it might be caused by RSUSB backend method

MartyG-RealSense, 2021-11-18:
Mixing package builds (apt-get) and source-code builds such as RSUSB on the same computer has the potential to cause conflicts.  You should be able to remove the source-code build if you go to the 'build' directory that you created during the RSUSB installation process and then input the command below.

`sudo make uninstall && make clean`

This version of the CMake instruction performs uninstall of the SDK files and cleans up the CMake cache but does not start building the SDK from source code again.

From that point onwards, use the package installation process on the **distribution_linux.md** page.

https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md

If you have previously performed a package build on your computer though, I recommend uninstalling previous package builds first with the command below.

`dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge`

zxl5344, 2021-11-18:
Thank you. I will try it when I have time.

closed 2021-11-18 by zxl5344