realsenseai/librealsense #9932: About box_dimensioner_multicam_demo

gjdeorl, 2021-11-08:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D400 } | 
| Operating System & Version |  Win (10) | 
| Platform                                 | PC  |
| Language                            |  C/python  | 

Hello
I am using box_dimensioner_multicam_demo. Can I get this file to run directly without using the chessboard? If yes, please tell me how, thanks.

MartyG-RealSense, 2021-11-08:
Hi @gjdeorl   RealSense users have in the past succeeded in separating the multiple camera code script - **realsense_device_manager.py** - from the box_dimensioner_multicam project and using it in their own Python project, though it is complicated to achieve.  I am not aware of a case where box_dimensioner_multicam has been adapted to work with multiple cameras without a chessboard though.

However, in https://github.com/IntelRealSense/librealsense/issues/3073 the question of whether it was possible to use the project without a chessboard for a **single** camera was asked (since with one camera it would not be necessary to calibrate intrinsics and extrinsics between multiple cameras).  A RealSense team member confirmed that it would be possible to modify the script to use the camera's own internal intrinsics and extrinsics instead of chessboard calibration.

A range of alternatives to box_dimensioner_multicam for box measurement with RealSense are also highlighted in https://github.com/IntelRealSense/librealsense/issues/9057#issuecomment-878840095

MartyG-RealSense, 2021-11-08:
You are very welcome.  If your interest was just in using multiple cameras with Python instead of measuring boxes with them, the **multiple_realsense_cameras.py** project in the link below may be helpful.

https://github.com/ivomarvan/samples_and_experiments/tree/master/Multiple_realsense_cameras

MartyG-RealSense, 2021-11-15:
Hi @gjdeorl  Do you require further assistance with this case, please?  Thanks!

gjdeorl, 2021-11-15:
FIne.
Thank you..

MartyG-RealSense, 2021-11-15:
Thanks very much for the update, @gjdeorl   :)   As you are fine, I will close this case.  Thanks again!

closed 2021-11-15 by MartyG-RealSense