realsenseai/librealsense #9934: Window + realsense2 SDK: wrong_api_call_sequence_error

HedgeHao, 2021-11-08:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.12.03.00 | 
| Operating System & Version |  Win10  | 
| Kernel Version (Linux Only)    |                                                               | 
| Platform                                 | PC                                                           |
| SDK Version                            |  2                                     | 
| Language                            |  C++                          | 
| Segment			|                           | 

### Issue Description
I am trying to run the basic hello program. And I get `wrong_api_call_sequence_error` error. Same camera, same code and cmake file can work on Linux but no on Windows 10. 

```c++
#include <iostream>
#include <librealsense2/rs.hpp>

int main()
{
    rs2::pipeline p = rs2::pipeline();
    p.start();

    rs2::frameset frames = p.wait_for_frames();
    rs2::depth_frame frame = frames.get_depth_frame();
    float width = frame.get_width();
    float height = frame.get_height();
    float dist_to_center = frame.get_distance(width / 2, height / 2);

    std::cout << "The camera is facing an object " << dist_to_center << " meters away \n";
    return 0;
}
```
Here's CMakeLists.txt
```cmake
cmake_minimum_required(VERSION 3.8)

project(hello-realsense CXX)

if (WIN32)
	include_directories("C:/Program Files (x86)/Intel RealSense SDK 2.0/include")
	link_directories("C:/Program Files (x86)/Intel RealSense SDK 2.0/lib/x64")
endif(WIN32)

add_executable( ${PROJECT_NAME} hello-realsense.cpp)
set_property(TARGET ${PROJECT_NAME} PROPERTY CXX_STANDARD 11)

target_link_libraries( ${PROJECT_NAME} PRIVATE realsense2)
```

MartyG-RealSense, 2021-11-08:
Hi @HedgeHao   Does it build on Windows if you add **realsense2.dll** to the link path?

`link_directories("C:/Program Files (x86)/Intel RealSense SDK 2.0/lib/x64/realsense2.dll")`

HedgeHao, 2021-11-09:
still not working. If I link realsense2.dll, visual studio says `cannot open file realsense2.lib`. I also try add both link directories. Not working with same error `wrong_api_call_sequence_error`.

MartyG-RealSense, 2021-11-09:
Are you able to run the pre-built executable .exe version of the **rs-hello-realsense** example?  If you have installed the full Windows version of the RealSense SDK, you can find this exe file by right-clicking on the RealSense Viewer launch icon on the Windows desktop and selecting the **Open file location** menu option to open the SDK's 'Tools' folder.  Then locate in this folder the file **rs-hello-realsense.exe** and launch it with a double-leftclick whilst the D415 is plugged in.

[image]

HedgeHao, 2021-11-10:
Yes. Pre-built executable works fine. And realsense-viewer can see the video output, too.

MartyG-RealSense, 2021-11-10:
I carried out further research of **rs2::wrong_api_call_sequence_error** but there was no clear solution, unfortunately.  One RealSense user dealt with the error by inserting a 'catch' into the C++ scripting to handle the error instead of letting the program crash.

```
catch (const rs2::wrong_api_call_sequence_error &) 
{ 
/*pass*/ 
}
```

Further information about handling errors in C++ can be found in the SDK documentation link below,

https://dev.intelrealsense.com/docs/librealsense-error-handling-scheme

closed 2021-11-23 by MartyG-RealSense