realsenseai/librealsense #9935: librealsense2-udev-rules debian package has wrong post install script

samuelba, 2021-11-08:
The debian package `librealsense2-udev-rules_2.49.0-0_realsense0.5307_amd64.deb` has a wrong check in the post install script.

```
#!/bin/sh -e
 
echo "\nPostinst script activated..."
# Copy the preset files to the actual $HOME location
target_dir="/home/${SUDO_USER}/Documents/librealsense2"
mkdir -p "${target_dir}"
#Installing presets is optional and requiers the above target library as well as the `rsync` utility
#Verify and proceed
echo "Presets deployment:"
echo "rsync processes found - " `ps ax | grep rsync | wc -l`
[ -d "usr/share/librealsense2/presets" ] && echo "src dir exists" || echo "no src dir"
[ -d "${target_dir}" ] && echo "tgt dir exists" || echo "no tgt dir"

if ( [ $((`ps ax | grep rsync | wc -l`)) -gt 0 ] && [ -d "${target_dir}" ] && [ -d "usr/share/librealsense2/presets" ] );
then
	echo "Deploy presets files"
	rsync -avh --ignore-errors usr/share/librealsense2/presets "${target_dir}"
else
	echo "Presets deployment is skipped"
fi
```

It check if `rsync` exists, but if you execute `echo $(ps ax | grep rsync | wc -l)` with any string instead of `rsync` it will always return `1` since the command always finds itself :smile:.

A better check would be e.g. `if [ ! -z $(command -v rsync) ]`

MartyG-RealSense, 2021-11-08:
Thanks very much @samuelba for your suggestion.  Please feel free to submit a **Pull Request** for this code change at the link below so that it can be considered for merging into the RealSense SDK.

https://github.com/IntelRealSense/librealsense/pulls

samuelba, 2021-11-08:
Hi @MartyG-RealSense 

I would have done so, but I could not find the specific file here in this repository and I don't know where else you keep that file.

MartyG-RealSense, 2021-11-08:
Where did the script that you quoted in your opening message come from, please?

samuelba, 2021-11-08:
It comes from the debian package `librealsense2-udev-rules_2.49.0-0_realsense0.5307_amd64.deb` which is from your APT repository https://librealsense.intel.com/Debian/apt-repo

MartyG-RealSense, 2021-11-09:
I see, thanks very much for the information.  I am not involved in SDK development activities (including package creation) and there is apparently not anything that you could edit yourself to create a pull request if the code is part of a package.  So this check will likely have to be left unchanged in its current form.

closed 2021-11-26 by MartyG-RealSense