realsenseai/librealsense #9938: Coordinate transformation between camera sensors

Esperer105, 2021-11-09:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | **L515**  | 
| Firmware Version                   | 01.05.05.00 | 
| Operating System & Version |  Win10  | 
| Platform                                 | PC  |
| SDK Version                            |   2.43.0.3013                          | 
| Language                            |  python                  | 
       

### Issue Description
When I try to use the **rs.rs2_deproject_pixel_to_point()**function to get the 3D coordinate from the pixel coordinate, I am wondering whether the intrinsic of depth should be used or not, because there is another intrinsic of color module. Then I do an experiment with two intrinsics, it seems to be different. I check the  [API Reference](https://intelrealsense.github.io/librealsense/python_docs/_generated/pyrealsense2.html#module-pyrealsense2) for further explanation for the parameter 
 **intrinsic** of  **rs.rs2_deproject_pixel_to_point()**, but there is no explanation. Is it correct to use the intrinsic of the color module? Are the separate coordinates corresponding to the color sensor and depth sensor separately?
### code
```
import pyrealsense2 as rs
pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 1280, 720, rs.format.bgr8, 30)
profile = pipeline.start(config)
stream_profile_color = profile.get_stream(rs.stream.color)
stream_profile_depth = profile.get_stream(rs.stream.depth)
color_intrs = stream_profile_color.as_video_stream_profile().get_intrinsics()
depth_intrs = stream_profile_depth.as_video_stream_profile().get_intrinsics()
extrinsics = stream_profile_depth.get_extrinsics_to(stream_profile_color)
print("extrins:\n", extrinsics)
align_to = rs.stream.color
align = rs.align(align_to)

# pixel coordinate
x = 766
y = 486
frames = pipeline.wait_for_frames()
aligned_frames = align.process(frames)
aligned_depth_frames = aligned_frames.get_depth_frame()
dis = aligned_depth_frames.get_distance(x, y)

# get the 3D coordinate 
camera_coordinate = rs.rs2_deproject_pixel_to_point(depth_intrs, [x, y], dis)
camera_coordinate1 = rs.rs2_deproject_pixel_to_point(color_intrs, [x, y], dis)

print("distance:", dis)
print("depth intrins:", camera_coordinate)
print("color intrins:", camera_coordinate1)
pipeline.stop()
```
### the results are as follows
```
extrins:
 rotation: [0.999965, 0.00762357, 0.00331248, -0.00754261, 0.999688, -0.0238027, -0.0034929, 0.0237769, 0.999711]
translation: [0.000304107, 0.0142351, -0.00695471]
distance: 0.30650001764297485
depth intrins: [0.30515602231025696, 0.15374036133289337, 0.30650001764297485]
color intrins: [0.03648047521710396, 0.04504003748297691, 0.30650001764297485]
```
### Issue Description
After the experiment mentioned above, I encountered another problem that the transform between the color coordinate and depth coordinate is different between my own transform and API's transform(maybe the API is misused) based on the assumption that **rs.rs2_deproject_pixel_to_point()** function gets the depth coordinate using the intrinsic of depth. **My question is: What is the correct coordinate? Maybe I need more instructions**
```
import pyrealsense2 as rs
import numpy as np
import cv2

pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 1280, 720, rs.format.bgr8, 30)
profile = pipeline.start(config)
stream_profile_color = profile.get_stream(rs.stream.color)
stream_profile_depth = profile.get_stream(rs.stream.depth)
color_intrs = stream_profile_color.as_video_stream_profile().get_intrinsics()
depth_intrs = stream_profile_depth.as_video_stream_profile().get_intrinsics()
extrinsics = stream_profile_depth.get_extrinsics_to(stream_profile_color)
print("extrins:\n", extrinsics)
align_to = rs.stream.color
align = rs.align(align_to)
x = 766
y = 486
frames = pipeline.wait_for_frames()
aligned_frames = align.process(frames)
aligned_depth_frames = aligned_frames.get_depth_frame()
dis = aligned_depth_frames.get_distance(x, y)
camera_coordinate = rs.rs2_deproject_pixel_to_point(depth_intrs, [x, y], dis)
camera_coordinate1 = rs.rs2_deproject_pixel_to_point(color_intrs, [x, y], dis)
# generate the transform matrix on my own
rotation_matrix = np.array(extrinsics.rotation).reshape(3, 3)
translation_matrix = np.array(extrinsics.translation).reshape(3, 1)
tf_matrix = np.vstack((np.hstack((rotation_matrix, translation_matrix)), [0, 0, 0, 1]))
print("tf_matrix:", tf_matrix)
# Convert to homogeneous coordinates
camera_coordinate_my_own = np.dot(tf_matrix, np.append(camera_coordinate, [1]).reshape(4, 1))
camera_coordinate_api = rs.rs2_transform_point_to_point(extrinsics, camera_coordinate)
# the coordinate unit is meter
print("my own:\n", camera_coordinate_my_own)
print("api:\n", camera_coordinate_api)
print("distance:", dis)
print("depth intrins:", camera_coordinate)
print("color intrins:", camera_coordinate1)
pipeline.stop()
```
### the results are as follow
```
extrins:
 rotation: [0.999965, 0.00762357, 0.00331248, -0.00754261, 0.999688, -0.0238027, -0.0034929, 0.0237769, 0.999711]
translation: [0.000304107, 0.0142351, -0.00695471]
tf_matrix: [[ 9.99965429e-01  7.62356725e-03  3.31247575e-03  3.04106681e-04]
 [-7.54260505e-03  9.99688208e-01 -2.38026995e-02  1.42350988e-02]
 [-3.49290436e-03  2.37768926e-02  9.99711215e-01 -6.95471419e-03]
 [ 0.00000000e+00  0.00000000e+00  0.00000000e+00  1.00000000e+00]]
my own:
 [[0.30713556]
 [0.15809526]
 [0.3015423 ]
 [1.        ]]
api:
 [0.30272528529167175, 0.17727510631084442, 0.2963126599788666]
distance: 0.3060000240802765
depth intrins: [0.3046582341194153, 0.15348955988883972, 0.3060000240802765]
color intrins: [0.03642096370458603, 0.04496656358242035, 0.3060000240802765]
```

MartyG-RealSense, 2021-11-09:
Hi @Esperer105  If you are aiming to obtain XYZ 3D coordinates for a single pixel then it may be easier and more processing-efficient to use the instruction **rs2_project_color_pixel_to_depth_pixel** to convert a color pixel to a depth pixel instead of aligning depth to color and using rs2_deproject_pixel_to_point.  An example of scripting for using this instruction with Python can be found at https://github.com/IntelRealSense/librealsense/issues/5603#issuecomment-574019008

If you need to use rs2_deproject_pxel_to_point then in regard to your first question, a RealSense team member states in https://github.com/IntelRealSense/librealsense/issues/1413#issuecomment-375111942 that when aligning depth to color, the intrinsic of a depth-aligned-to-color frame should be identical to the intrinsic of the color frame.  When aligning depth to color using an **align_to** instruction, the depth frame's field of view (FOV) is automatically resized to match the FOV of the color frame.

The SDK's **Projection** documentation in the link below details how each stream has its own coordinate system that is described by its **extrinsic** parameters.

https://dev.intelrealsense.com/docs/projection-in-intel-realsense-sdk-20#section-extrinsic-camera-parameters

The Projection documentation also provides the following information about deprojection: "Deprojection takes a 2D pixel location on a stream's images, as well as a depth, specified in meters, and maps it to a 3D point location within the stream's associated 3D coordinate space".

You can check what the correct transform values for a particular stream should be by launching the SDK's rs-enumerate-devices tool with the command **rs-enumerate-devices -c** ('c' stands for calibration) to list the intrinsics and extrinsics of the camera for all supported stream types.

[image]

[image]

Esperer105, 2021-11-09:
@MartyG-RealSense Thanks a lot! I will try the method you mentioned. But how to launch the SDK's rs-enumerate-devices tool ? Should I input the command **rs-enumerate-devices** in the command line window?

MartyG-RealSense, 2021-11-09:
On Windows, rs-enumerate-devices is run from the Command Prompt interface.  You can do so by following the steps below.

1.  Navigate to the SDK's **Tools** directory by inputting this command into the Command Prompt window:

**cd C:\Program Files (x86)\Intel RealSense SDK 2.0\tools**

2.  Input the command to run rs-enumerate-devices in its -c (Calibration) mode.

**rs-enumerate-devices -c**

3.  The full intrinsics and extrinsics of the camera will be listed in the Command Prompt window.  You can then use the scrollbar at the side of the window to scroll up from the bottom of the listing to find the information that you require.

Esperer105, 2021-11-10:
@MartyG-RealSense Thanks a lot!  I missed the transpose step for **extrinsic.rotation**, after transposing the rotation matrix, the transform is correct! And the information you provided help me a lot to understand the basic knowledge about different coordinates. Above all, My problem has been solved!
```
my own:
 [[0.53991827]
 [0.30514961]
 [0.53416937]
 [1.        ]]
api:
 [0.5399183034896851, 0.305149644613266, 0.5341693758964539]
```

MartyG-RealSense, 2021-11-10:
You are very welcome, @Esperer105 - it's great to hear that you were successful  :)   Thanks very much for the update!

Esperer105, 2021-11-13:
@MartyG-RealSense 
### Issues
Hi, I am back again! I have to say that I was a little sloppy about this problem that day. 
 I try to use the **rs2_project_color_pixel_to_depth_pixel ** to get the XYZ coordinate instead of aligning the color and the depth according to your advice and the example [#5603(comment)](https://github.com/IntelRealSense/librealsense/issues/5603#issuecomment-574019008), but the consequence seems not to be satisfying. The coordinates in two method are not approximatelty same. Do I use the **rs2_project_color_pixel_to_depth_pixel ** in the right way? Do I use the **rs2_project_color_pixel_to_depth_pixel ** as you expected?
Looking forward to your reply!
```
import pyrealsense2 as rs
import numpy as np

# get the handeye matrix
M_camera2base = np.load('D:\\Realsense\\calibrateHandeye\\M_camera2base_eye_to_hand.npy')

pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 1280, 720, rs.format.bgr8, 30)
profile = pipeline.start(config)

depth_scale = profile.get_device().first_depth_sensor().get_depth_scale()
depth_min = 0.11  # meter
depth_max = 1.0  # meter
depth_intrin = profile.get_stream(rs.stream.depth).as_video_stream_profile().get_intrinsics()
color_intrin = profile.get_stream(rs.stream.color).as_video_stream_profile().get_intrinsics()

depth_to_color_extrin = profile.get_stream(rs.stream.depth).as_video_stream_profile().get_extrinsics_to(
    profile.get_stream(rs.stream.color))
color_to_depth_extrin = profile.get_stream(rs.stream.color).as_video_stream_profile().get_extrinsics_to(
    profile.get_stream(rs.stream.depth))

coordinate_collect = []
frames = pipeline.wait_for_frames()

# get the original depth frame
depth_frame = frames.get_depth_frame()
# pixel coordinate
x = 410
y = 92
color_point = [x, y]
depth_point = rs.rs2_project_color_pixel_to_depth_pixel(
    depth_frame.get_data(), depth_scale,
    depth_min, depth_max,
    depth_intrin, color_intrin, depth_to_color_extrin, color_to_depth_extrin, color_point)
distance = depth_frame.get_distance(int(depth_point[0]), int(depth_point[1]))

camera_coordinate_depth = rs.rs2_deproject_pixel_to_point(depth_intrin, depth_point, distance)
print("coordinate relative to depth sensor", camera_coordinate_depth)
camera_coordinate_color = rs.rs2_transform_point_to_point(depth_to_color_extrin, camera_coordinate_depth)
print("coordinate relative to color sensor:", camera_coordinate_color)
camera_coordinate_color1 = [camera_coordinate_color[0] * 1000, camera_coordinate_color[1] * 1000,
                            camera_coordinate_color[2] * 1000]
camera_coordinate_color2 = np.dot(M_camera2base,
                                  np.vstack((np.array(camera_coordinate_color1).reshape(3, 1), [1])))

print("final_coordinate using rs.rs2_project_color_pixel_to_depth_pixel():\n", camera_coordinate_color2)

# align the color and depth streams
align_to = rs.stream.color
align = rs.align(align_to)
aligned_frames = align.process(frames)
aligned_depth_frames = aligned_frames.get_depth_frame()
dis = aligned_depth_frames.get_distance(x, y)
camera_coordinate_align = rs.rs2_deproject_pixel_to_point(color_intrin, [x, y], dis)
camera_coordinate_align = [camera_coordinate_align[0] * 1000, camera_coordinate_align[1] * 1000,
                           camera_coordinate_align[2] * 1000]
camera_coordinate_align = np.dot(M_camera2base,
                                 np.vstack((np.array(camera_coordinate_align).reshape(3, 1), [1])))
print("final_coordinate after alignment:\n", camera_coordinate_align)
pipeline.stop()
```
the results are as follows:
```
coordinate relative to depth sensor [-0.3307822346687317, -0.4967399835586548, 1.1180000305175781]
coordinate relative to color sensor: [-0.3907437324523926, -0.4911433756351471, 1.120328664779663]
final_coordinate using rs.rs2_project_color_pixel_to_depth_pixel():
 [[-293.22494481]
 [-640.72675349]
 [ 266.17722902]
 [   1.        ]]
final_coordinate after alignment:
 [[-302.08847775]
 [-642.12180112]
 [ 268.22364011]
 [   1.        ]]
```

MartyG-RealSense, 2021-11-13:
Hi @Esperer105   https://github.com/IntelRealSense/librealsense/issues/4315 explains that there will be small differences when using rs2_deproject_pixel_to_point because of use of alignment.

Esperer105, 2021-11-13:
@MartyG-RealSense Thanks for your reply! I use the above code to test the accuracy of two method, first is D455, another is L515:
the results are as follows:
### D455
```
coordinate relative to color sensor using rs.rs2_project_color_pixel_to_depth_pixel():
 [-0.3910404443740845, -0.4915826916694641, 1.1213306188583374]
coordinate relative to the color sensor after alignment:
 [-0.40030184388160706, -0.4909859001636505, 1.1200000047683716]
```
except the **X**, the difference of two other coordinates can be ignored. The difference of **X** is 9 mm
### L515
```
coordinate relative to color sensor using rs.rs2_project_color_pixel_to_depth_pixel():
 [0.19322717189788818, 0.029796496033668518, 0.771574079990387]
coordinate relative to the color sensor after alignment:
 [0.2083302140235901, 0.11789297312498093, 0.7825000286102295]
```
The difference of coordinates is huge, that's why I doubt I maybe make a mistake. I did not find the answer in the [example](https://github.com/IntelRealSense/librealsense/issues/4315) you mentioned. Thank you in advance!

MartyG-RealSense, 2021-11-13:
If you are using get_distance(x,y) with Python, it is a known issue that when alignment is being used with it then the values can be accurate at and around the center of the image but start to drift significantly in the coordinates that are towards the outer edges of the image, leading to large inaccuracies at the edges (such as going from millimeters to 2 meters).  There is a list of resources about this at https://github.com/IntelRealSense/librealsense/issues/9088#issuecomment-847583702

It is also worth bearing in mind that the L515 has a different depth unit scale to the 400 Series cameras.   On the 400 Series the default is 0.001 and on L515 it is **0.000250**

Esperer105, 2021-11-13:
@MartyG-RealSense  Ok, I will check the problem and related issues. Thank you!

Esperer105, 2021-11-15:
@MartyG-RealSense Hi, I find the key point in my code: the selection of **depth_max** and **depth_min** parameters are essential to get an accurate result. When I change the two parameters to a proper range, the result obtained by **L515** from**rs2_project_color_pixel_to_depth_pixel** becomes accurate. So that"s the reason of the above inaccuracies. Thank you again!

closed 2021-11-15 by Esperer105

MartyG-RealSense, 2021-11-15:
You are very welcome.  It's great news that you found a solution, @Esperer105 - thanks very much for the update    :)