realsenseai/librealsense #9939: How to get distance in python

Karin-Sugi, 2021-11-09:
Hi, i'm trying to get distance with realsense D435i in python. However, i could find little python example codes which get distance data, not image.

Now, I got an error saying 
```
get_distance(): incompatible function arguments. The following argument types are supported:
    1. (self: pyrealsense2.pyrealsense2.depth_frame, x: int, y: int) -> float
   
Invoked with: <pyrealsense2.frame Z16 #1>, 424.0, 240.0
```
I think this supported argument type is not what i mean...

I know it's too basic question, but can any one tell me how to get distance of the point (x,y) or get distance of all the frame in python??

I put my code below as well.

```
import pyrealsense2 as rs
import numpy as np

pipeline = rs.pipeline()

# Start streaming
pipeline.start()

frames = pipeline.wait_for_frames()
depth = frames.get_depth_frame()

width = depth.get_width()
height = depth.get_height()

dist = depth.get_distance(width/2, height/2)
print(dist)

# Stop streaming
pipeline.stop()
```

MartyG-RealSense, 2021-11-09:
Hi @Karin-Sugi  Your code looks fine to me, so I am not sure why it is erroring.  Does the error still occur if you specifically define width and height as being **int** values?

**dist = depth.get_distance(int(width/2), int(height/2))**

Karin-Sugi, 2021-11-09:
Hi @MartyG-RealSense Your code did work!! I could get the data. Thanks.

On the other hand, in my search, some people use **depth_scale * depth_frame** to get distance data. What is the difference between them?

MartyG-RealSense, 2021-11-10:
The camera produces raw pixel depth values in a format called **uint16_t**.  Multiplying the uint16_t value by the **depth unit scale** of the camera provides a distance value in meters.

For example, the default depth scale of the 400 Series cameras is 0.001 (meters).  If the raw uint16_t pixel depth value was 4500 then multiplying it by a depth unit scale of 0.001 would give 4500 x 0001 = **4.5** (meters distance).

depth.get_distance though is an alternative way to obtain the distance in meters without having to use the above uint16_t x depth scale formula.  There is a discussion at https://github.com/IntelRealSense/librealsense/issues/7264 where the two methods are compared.

MartyG-RealSense, 2021-11-18:
Hi @Karin-Sugi  Do you require further assistance with this case, please?  Thanks!

Karin-Sugi, 2021-11-20:
Thanks for your assistance!!

closed 2021-11-20 by Karin-Sugi

MartyG-RealSense, 2021-11-20:
You are very welcome, @Karin-Sugi - thanks for the update!