realsenseai/librealsense #9940: What is a method that enables volume measurement at a distance of 1.5 m?

HyeYumKim, 2021-11-09:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Operating System & Version |   Win (10)  | 
| Platform                                 | PC |
| Language                            |  python  | 

Hello, I am using box_dimensioner_multicam_demo to measure volume right now. At a distance of about 40cm, the volume measurement is almost accurate, but at a distance of 1.5m, it cannot be measured.
I would like to know the parameters that can measure volume at a distance of 1.5m.
In this case, what can be adjusted and how can it be checked?
And if it is used as D455, is there a value that needs to be modified separately?
thanks.

MartyG-RealSense, 2021-11-09:
Hi @HyeYumKim  May I begin by confirming whether your chessboard is 6x9" in size?  If it is not then the height and width values in the **box_dimensioner_multicam_demo.py** script should be edited.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/box_dimensioner_multicam/box_dimensioner_multicam_demo.py#L40-L42

Having said that, I have seen past reports with this particular example program where moving the camera further than a meter away from the observed box on the chessboard can cause problems.  This is not a problem that is specific to the D455 camera model and as far as I am aware there is not a solution for that.

HyeYumKim, 2021-11-09:
The size of the chessboard was all set, and from my understanding, if the cailbration was successful at a distance of 1.5m, I thought that the volume measurement would be good at that location. Can't it be done by modifying something other than modifying the size of the chessboard?

MartyG-RealSense, 2021-11-09:
Although I don't recall this being attempted before with box_dimensioner_multicam, you could try changing the **depth unit scale** of the camera from the default value of 0.001 to a value of **0.0001**.  This 'zooms in' the depth view a little and can provide a noticable improvement in depth image detail with less empty gaps that have no depth data.

https://github.com/IntelRealSense/librealsense/issues/6230#issuecomment-613720661 has an example of Python code for setting the depth unit scale to 0.0001.

HyeYumKim, 2021-11-09:
Thank you very much.
I'll try. Maybe modifying the values of the provided preset.json file will also help when the distance goes away?

HyeYumKim, 2021-11-10:
After modifying the depth unit scale, the outline is not clear and the volume cannot be measured. Is there any other way?

MartyG-RealSense, 2021-11-10:
In https://github.com/IntelRealSense/librealsense/issues/9926#issuecomment-965135009 two L515 users found that performance improved by making an edit to the project script **helper_functions.py**.  Though you have a D455, it may be worth trying the edit yourself to see whether it is helpful for you too.

jerrydegroot, 2021-11-12:
By changing helper_functions.py line 180 into: z = depth_image.flatten()/4000 , we get "reasonable" results for single L515 camera setup (the bounding box is ~20 % too large with camera distance ~100 cm; shorter camera distance gives worse results)

We expect that multicam setup will not work for L515 cameras in default asynchronous mode: both cameras will emit IR pulses at the same time, thus interfering each other's depth readings. There is a [whitepaper ](https://dev.intelrealsense.com/docs/lidar-camera-l515-multi-camera-setup) on using multiple L515 with a hardware trigger/sync. cable which should work for static object/scenery measurements. We are looking for a L515 multicam setup for dynamic object/scenery. Any suggestions welcome.

closed 2021-11-24 by MartyG-RealSense