realsenseai/librealsense #9942: Acquire frame using L515 via Matlab failed

zjk41, 2021-11-09:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
Here is the error message I received when I tried to use matlab to acquire frames from L515, matlab successfully connect to the lidar as I can see its device name but can't get a frame back. Any idea?

Error using librealsense_mex
Frame didn't arrive within 15000

Error in realsense.pipeline/wait_for_frames (line 48)
                out = realsense.librealsense_mex('rs2::pipeline', 'wait_for_frames', this.objectHandle);

Error in realsense.pointcloud_example (line 17)
        fs = pipe.wait_for_frames();

MartyG-RealSense, 2021-11-10:
Hi @zjk41   May I begin by asking the questions below, please.

1.  Are you using Windows or Linux?

2.  Is your L515 able to stream depth in the RealSense Viewer program?

3.  Does it make a difference if you unplug the camera's USB cable from the base of the camera, turn the micro-sized connector around the opposite way and then re-insert it into the base of the camera (as USB-C cables are two-way insertion on the micro-sized end).

4.  Which firmware driver version is currently installed in your camera?  If it is older than firmware version **1.5.1.3** and the camera was inserted in a USB 2 port (or mis-detected as being on a USB 2 connection even though it is inserted in a USB 3 port) then this could prevent the L515 from streaming, as USB 2 support was not added until firmware 1.5.1.3 onwards

zjk41, 2021-11-11:
Hi Marty,

Actually, it fixed itself after I left it there overnight. But I also wonder how do you convert the depth frame captured from the lidar into a 2D depth map, where a depth frame should be an array of 1024x768 and each pixel has a depth value. 
In the depth example, it only colorizes the depth frame that doesn't preserve the depth value. 

Thanks!
Jingkai

MartyG-RealSense, 2021-11-11:
If you are aiming to obtain depth values from a MATLAB wrapper script, a RealSense MATLAB user shared their script in https://github.com/IntelRealSense/librealsense/issues/9773#issuecomment-925553900 for getting distance using the **depth.get_distance** instruction.

zjk41, 2021-11-12:
Hi Marty,
This works out very well! Thank you for your help!

closed 2021-11-12 by zjk41

MartyG-RealSense, 2021-11-12:
You are very welcome.  It's great to hear that the script was helpful to you - thanks very much for the update!