realsenseai/librealsense #9943: Build on Raspberry Pi OS Bullseye

davidscherf, 2021-11-09:
Raspberry Pi OS 32 bit Bullseye
Kernel 5.10
gcc version 10.2.1
cmake version 3.18
Language C# but not getting past the librealsense compile

Steps at https://github.com/IntelRealSense/librealsense/blob/development/doc/installation_raspbian.md
include packages that no longer exist in the newer OS, all that end in -dbgsym
Not installing those packages I continued

sudo udevadm control --reload-rules && udevadm trigger 
gets Permission denied, no issue I guess a reboot will reload that.

cmake .. -DBUILD_EXAMPLES=true -DCMAKE_BUILD_TYPE=Release -DFORCE_LIBUVC=true
fails with missing open ssl missing, so installed libssl-dev

libssl-dev
Warning dynamic exception specifications are deprecated in C++11
and then fails with

/usr/bin/ld: ../../librealsense2.so.2.49.0: undefined reference to `__atomic_load_8'
/usr/bin/ld: ../../librealsense2.so.2.49.0: undefined reference to `__atomic_store_8'
/usr/bin/ld: ../../librealsense2.so.2.49.0: undefined reference to `__atomic_fetch_add_8'
collect2: error: ld returned 1 exit status
make[2]: *** [examples/hello-realsense/CMakeFiles/rs-hello-realsense.dir/build.make:110: examples/hello-realsense/rs-hello-realsense] Error 1
make[1]: *** [CMakeFiles/Makefile2:1219: examples/hello-realsense/CMakeFiles/rs-hello-realsense.dir/all] Error 2
make: *** [Makefile:149: all] Error 2

Any ideas how to fix this error?

MartyG-RealSense, 2021-11-10:
Hi @davidscherf  Intel's official Raspbian installation guide is designed for Raspberry Pi 3.  If you are using a Pi 4 then the similar guide by Acrobotic in the link below is recommended.

https://github.com/acrobotic/Ai_Demos_RPi/wiki/Raspberry-Pi-4-and-Intel-RealSense-D435

You may also have a smoother build experience if you update protobuf to a newer version than the ones recommended in the Intel and Acrobotic guides - 3.14 should be a suitable one.  You should be able to install this by setting the protobuf version to **v3.14.0** in the guide's protbuf install instruction.

As **-DFORCE_LIBUVC=true** is being used, this means that the build should not be dependent on Linux versions or kernel versions and does not require patching.  So the 5.10 kernel (which is not yet officially supported by the RealSense SDK) and changes to Raspbian since Buster should not in theory be influencing factors on whether the build succeeds.  

Installing the SDK with the libuvc backend method does require an internet connection though.  Can you confirm please whether you had an active internet connection at the time of your installation attempt, please?

davidscherf, 2021-11-11:
I did not install protobuf in buster, so was not trying to install it on bullseye, nor am i trying to compile any of the samples

Yes the pi had internet while trying to compile the SDK, It should not need the internet to compile as I did not need it to compile in buster.

I have compiled the SDK in buster, and worked out that I need the 2 binary files and some files from the Documents folder, and have made a setup to just cp the binaries and ln them, and copying this to bullseye works.

No idea why there is not a deb file or something to do this instead of compiling, have not gotten in to deb files yet so no idea how to make one myself.

Hope the steps are updated for when Raspberry PI OS gets x64.

MartyG-RealSense, 2021-11-11:
Thanks very much for the further information, @davidscherf - it is excellent to hear that you had success with Bullseye using your binary copying method!

MartyG-RealSense, 2021-11-17:
Hi @davidscherf  Do you require further assistance with this case, please?  Thanks!

davidscherf, 2021-11-17:
Although I have a workaround, compiling under an older OS, this does not fix the compile errors with the newer gcc

MartyG-RealSense, 2021-11-18:
In https://github.com/IntelRealSense/librealsense/issues/9962 there are a couple of Pi users who have also been having Raspbian installation problems recently.  I connected them to the information that you shared in this case.

MartyG-RealSense, 2021-11-30:
Case closed due to no further comments received.

closed 2021-11-30 by MartyG-RealSense

Sashraa, 2023-03-27:
> 

Kindly share the details about the two binary files that need to be added... kindly support. Thank you

Sashraa, 2023-03-29:
Can you kindly on this method ....#Bullseye using your binary copying method! your support is required.

pfcouto, 2023-11-08:
Hey @davidscherf can you explain in more detail which binaries did you copy, and how did you copy them? Thanks!

davidscherf, 2023-11-09:
These instructions are as per arm64 and a user of pi, make modification as per your requirements
Any file outside your home directory will obviously need sudo to copy it there
I hope i did not miss anything

udev rule
file is: /etc/udev/rules.d/99-realsense-libusb.rules
sudo udevadm control --reload-rules
sudo udevadm trigger

- make sure the following folder exists

/home/pi/Documents/librealsense2/presets

- the following files go in there

L535_Max_Range.preset
L535_Max_Range_With_Binning.preset
L535_Medium_Range.preset
L535_Min_Range.preset

- make sure the following folder exists

/usr/local/lib/aarch64-linux-gnu

- the following files go in there

libfw.a
libglfw3.a
librealsense-file.a

Following files in the folder above are version specific e.g here it is 2.49
librealsense2.so.2.49.0
librealsense2-gl.so.2.49.0

Setup sym links
sudo ln -s /usr/local/lib/aarch64-linux-gnu/librealsense2.so.2.49.0 /usr/local/lib/aarch64-linux-gnu/librealsense2.so.2.49
sudo ln -s /usr/local/lib/aarch64-linux-gnu/librealsense2.so.2.49.0 /usr/local/lib/aarch64-linux-gnu/librealsense2.so
sudo ln -s /usr/local/lib/aarch64-linux-gnu/librealsense2-gl.so.2.49.0 /usr/local/lib/aarch64-linux-gnu/librealsense2-gl.so.2.49
sudo ln -s /usr/local/lib/aarch64-linux-gnu/librealsense2-gl.so.2.49.0 /usr/local/lib/aarch64-linux-gnu/librealsense2-gl.so

- If you want the viewer

put it in: /usr/local/bin/realsense-viewer
sudo chmod +x /usr/local/bin/realsense-viewer
export PATH=$PATH:/usr/local/bin
sudo ldconfig -p