realsenseai/librealsense #9944: distance data is strange

Karin-Sugi, 2021-11-10:
Hi, I'm trying to get depth data of all frame with D435i. With get_distancec() command, I could get depth data. However, the data seems strange. The image below is the 3D graph (x,y : pixels, z : max depth - each depth) of table. I took it from the above. It must be plane, but at many point, it isn't.
[image]

I got data with python, and made the graph with matlab. I post both code here.
```
###
# python code to get depth data
###
import pyrealsense2 as rs # realsense package
import numpy as np # numpy package

pipeline = rs.pipeline() 
# Start streaming 
pipeline.start()

frames = pipeline.wait_for_frames()
depth = frames.get_depth_frame()

### size of frame
width = depth.get_width()
height = depth.get_height()

dist = np.zeros((height,width))
data = np.zeros((height-100,width-100)) 

## distance of (x,y)
for i in range(int(width)): 
    for j in range(int(height)):
        dist[j][i] = 1000 * depth.get_distance(i,j) 

# Stop streaming
pipeline.stop()

# remove outlier
for i in range(50, int(width-50), 1):
    for j in range(50, int(height-50), 1):
        if dist[j][i] > dist[j+50][i] + dist[j-50][i]:
            data[j-50][i-50] = (dist[j+50][i] + dist[j-50][i])/2
        elif dist[j][i] > (dist[j+50][i] + dist[j-50][i])/4:
            data[j-50][i-50] = (dist[j+50][i] + dist[j-50][i])/2
        elif dist[j][i] > dist[j][i+50] + dist[j][i-50]:
            data[j-50][i-50] = (dist[j][i+50] + dist[j][i-50])/2
        elif dist[j][i] > (dist[j][i+50] + dist[j][i-50])/4:
            data[j-50][i-50] = (dist[j][i+50] + dist[j][i-50])/2

# save
np.savetxt(r'C:\Users\kacha\Documents\MATLAB\DepthData\distance_data.txt', data, fmt='%.5e')
```

```
//////////
// matlab code to make graph
///////////
a = readmatrix('distance_data.txt');
b = max(a);
c = max(b)
distance = c - a;

figure(1)
mesh(distance)
```

How can I fix it?? If I should use filter, then tell me what kind of filter?

MartyG-RealSense, 2021-11-10:
Hi @Karin-Sugi  Is the surface of the table glossy / reflective?  If so then this may make it more difficult for the camera to analyze for depth detail, especially if there is a strong light source being cast onto the table surface that creates glare from reflections.

Also, what color is the table surface?  If it is dark grey or black then this would also make it difficult for the camera to read depth detail from, as dark grey / black shades absorb light.

Karin-Sugi, 2021-11-10:
Hi @MartyG-RealSense 
Sorry about luck of information. The table is wood table, and there is a light brown table cross on the table, so i don't think it is reflective and dark.

MartyG-RealSense, 2021-11-10:
Could you try disabling the camera's infrared semi random dot-pattern projection please to test whether the spikes on the above image correspond to dots projected onto the surface of the table.  If the scene is well lit then the dot pattern projection should not be needed by the camera to analyze the table's depth, as the camera is able to alternatively use the ambient light in the scene for depth analysis instead.

A code snippet that demonstrates setting the emitter to disabled ('0') in Python is shown below.

```
depth_sensor = device.query_sensors()[0]
depth_sensor.set_option(rs.option.emitter_enabled, 0)
```

MartyG-RealSense, 2021-11-18:
Hi @Karin-Sugi  Do you require further assistance with this case, please?  Thanks!

Karin-Sugi, 2021-11-20:
Hi @MartyG-RealSense 
My pc is crushed so i haven't check the code does work. I'll check in the next week and when it works, i'll close this topic. 

Thanks.

MartyG-RealSense, 2021-11-20:
Thanks very much, I look forward to your update.  Good luck!

MartyG-RealSense, 2021-12-08:
Case closed due to no further comments received.

closed 2021-12-08 by MartyG-RealSense

Karin-Sugi, 2021-12-15:
Sorry I couldn't send any conmments... I was too busy with my papers.

Now I tried your suggestion and added this code below to my code.
`depth_sensor = device.query_sensors()[0]
depth_sensor.set_option(rs.option.emitter_enabled, 0)`

And error came up saying "device is not defined", so I changed "device" to "pyrealsense2", which is defined as rs.
The modified code is below
`depth_sensor = rs.query_sensors()[0]
depth_sensor.set_option(rs.option.emitter_enabled, 0)`

Then, another error came up saying "module pyrealsense2 has no attribute query_sensors". How can I fix it?

MartyG-RealSense, 2021-12-15:
when defining depth_sensor with the method that I suggested, have you defined 'device' as shown below, please?

**device = pipeline_profile.get_device()**
depth_sensor = device.query_sensors()[0]
depth_sensor.set_option(rs.option.emitter_enabled, 0)