realsenseai/librealsense #9946: Unable to build pyrealsense2_net

colinmsnow, 2021-11-10:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 05.12.15.50 | 
| Operating System & Version |   Linux  | 
| Kernel Version (Linux Only)    |  4.9.253-tegra                                         | 
| Platform                                 | NVIDIA Jetson Nano B01  |
| SDK Version                            | 2.47.0 / 2.49.0 | 
| Language                            |  Python | 
| Segment			| Robot | 

### Issue Description

Hi,

I am trying to build the Python realsense library pyrealsense2_net and run the [net_viewer.py](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/net_viewer.py) example to get frames from a networked realsense device. The remote device is a Raspberry Pi 4 setup via the [guide to ethernet networking](https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras).

I am building realsense from source using the commands:

```
cmake .. -DBUILD_NETWORK_DEVICE=ON -DBUILD_PYTHON_BINDINGS=ON -DPYTHON_EXECUTABLE=$(which python3) -DBUILD_SHARED_LIBS:BOOL=ON
make
sudo make install
```

I have also tried with only the first two flags but added shared_libs and python_executable based on comments on some similar issues.

Running the above commands successfully builds pyrealsense2 and a folder for pyrealsense2_net but the network library has almost no methods and does not implement net_device which is critical for using networked cameras.

Here is what happens if I import the two libraries and see their methods:

```
>>> import pyrealsense2
>>> dir(pyrealsense2)
['BufData', 'D400', 'DEPTH', 'L500', 'SR300', 'STAEControl', 'STAFactor', 'STCensusRadius', 'STColorControl', 'STColorCorrection', 'STDepthControlGroup', 'STDepthTableControl', 'STHdad', 'STRauColorThresholdsControl', 'STRauSupportVectorControl', 'STRsm', 'STSloColorThresholdsControl', 'STSloPenaltyControl', 'T200', 'TRACKING', '__doc__', '__file__', '__loader__', '__name__', '__package__', '__spec__', '__version__', 'align', 'any', 'any_intel', 'auto_calibrated_device', 'calib_target_type', 'calibrated_sensor', 'calibration_change_device', 'calibration_status', 'calibration_type', 'camera_info', 'color_sensor', 'colorizer', 'composite_frame', 'config', 'context', 'debug_protocol', 'debug_stream_sensor', 'decimation_filter', 'depth_frame', 'depth_huffman_decoder', 'depth_sensor', 'depth_stereo_sensor', 'device', 'device_calibration', 'device_list', 'disparity_frame', 'disparity_transform', 'distortion', 'dsm_params', 'enable_rolling_log_file', 'event_information', 'extension', 'extrinsics', 'filter', 'filter_interface', 'firmware_log_message', 'firmware_log_parsed_message', 'firmware_logger', 'fisheye_sensor', 'format', 'frame', 'frame_metadata_value', 'frame_queue', 'frame_source', 'hdr_merge', 'hole_filling_filter', 'intrinsics', 'l500_visual_preset', 'log', 'log_message', 'log_severity', 'log_to_callback', 'log_to_console', 'log_to_file', 'max_usable_range_sensor', 'motion_device_intrinsic', 'motion_frame', 'motion_sensor', 'motion_stream', 'motion_stream_profile', 'non_intel', 'notification', 'notification_category', 'option', 'option_range', 'options', 'pipeline', 'pipeline_profile', 'pipeline_wrapper', 'playback', 'playback_status', 'pointcloud', 'points', 'pose', 'pose_frame', 'pose_sensor', 'pose_stream', 'pose_stream_profile', 'processing_block', 'product_line', 'quaternion', 'recorder', 'region_of_interest', 'reset_logger', 'roi_sensor', 'rs2_deproject_pixel_to_point', 'rs2_fov', 'rs2_project_color_pixel_to_depth_pixel', 'rs2_project_point_to_pixel', 'rs2_transform_point_to_point', 'rs400_advanced_mode', 'save_single_frameset', 'save_to_ply', 'sensor', 'sequence_id_filter', 'serializable_device', 'software_device', 'software_motion_frame', 'software_notification', 'software_pose_frame', 'software_sensor', 'software_video_frame', 'spatial_filter', 'stream', 'stream_profile', 'syncer', 'temporal_filter', 'terminal_parser', 'texture_coordinate', 'threshold_filter', 'timestamp_domain', 'tm2', 'units_transform', 'updatable', 'update_device', 'vector', 'vertex', 'video_frame', 'video_stream', 'video_stream_profile', 'wheel_odometer', 'yuy_decoder', 'zero_order_invalidation']
>>> import pyrealsense2_net
>>> dir(pyrealsense2_net)
['__doc__', '__loader__', '__name__', '__package__', '__path__', '__spec__']
```

I have tried SDK versions 2.47.0 and 2.49.0 and they seem to work the same.

Is there something I am missing here? Additional flags I need to set or libraries to install?

Thanks,
Colin

MartyG-RealSense, 2021-11-10:
Hi @colinmsnow   The ethernet guide recommends including the CMake build flag **-DFORCE_RSUSB_BACKEND=ON** when building the SDK from source on Raspberry Pi.  This advice also applies to building the SDK from source on **Jetson**, as demonstrated by the Jetson Nano compatible installation guide at https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049

colinmsnow, 2021-11-10:
I tried again building with:

```
cmake .. -DBUILD_NETWORK_DEVICE=ON -DBUILD_PYTHON_BINDINGS=ON -DPYTHON_EXECUTABLE=$(which python3) -DBUILD_SHARED_LIBS:BOOL=ON -DFORCE_RSUSB_BACKEND=ON -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true
```
It builds successfully but I still have the same problem with missing methods in pyrealsense_net. I have been following the instructions in [ #20](https://github.com/IntelRealSense/meta-intel-realsense/issues/20#issuecomment-687180484) and have moved the __init__.py file to the directory that holds the python package. This makes pyrealsense2 work but there is no equivalent __init__.py for the net module. I tried it with the same init file as pyrealsense2 and also a blank init and neither caused any change.

The build put the python packages in `/usr/local/lib/python3.6` and the `pyrealsense_net` folder contains the following files:

```
pyrealsense2-net.cpython-36m-aarch64-linux-gnu.so  pyrealsense2-net.cpython-36m-aarch64-linux-gnu.so.2.47  pyrealsense2-net.cpython-36m-aarch64-linux-gnu.so.2.47.0
```

This seems like it should be correct but I have no way of verifying whether these files are flawed or whether they are not being imported properly.

MartyG-RealSense, 2021-11-11:
The instructions in https://github.com/IntelRealSense/meta-intel-realsense/issues/20#issuecomment-687180484 relate to a branch of Linux called **Yocto** and not Ubuntu.

After having installed the pyrealsense2 wrapper and included net support in the build with **-DBUILD_NETWORK_DEVICE=ON**, what happens if you then attempt to launch the **net_viewer.py** example?

colinmsnow, 2021-11-11:
I built it using the flags:
```
cmake .. -DBUILD_NETWORK_DEVICE=ON -DBUILD_PYTHON_BINDINGS=ON  -DBUILD_SHARED_LIBS:BOOL=ON -DFORCE_RSUSB_BACKEND=ON -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true
```
Running `make` and `sudo make install` results in the folder `/usr/local/lib/python3.6/pyrealsense2` that contains the files:
```
pybackend2.cpython-36m-aarch64-linux-gnu.so         pyrealsense2.cpython-36m-aarch64-linux-gnu.so         pyrealsense2-net.cpython-36m-aarch64-linux-gnu.so
pybackend2.cpython-36m-aarch64-linux-gnu.so.2       pyrealsense2.cpython-36m-aarch64-linux-gnu.so.2.49    pyrealsense2-net.cpython-36m-aarch64-linux-gnu.so.2.49
pybackend2.cpython-36m-aarch64-linux-gnu.so.2.49.0  pyrealsense2.cpython-36m-aarch64-linux-gnu.so.2.49.0  pyrealsense2-net.cpython-36m-aarch64-linux-gnu.so.2.49.0
```

If I run `net_viewer.py` in this folder I get an import error saying that pyrealsense2_net does not exist (though it successfully imports pyrealsense2 first):
```
Traceback (most recent call last):
  File "test_realsense_net.py", line 12, in <module>
    import pyrealsense2_net as rsnet
ModuleNotFoundError: No module named 'pyrealsense2_net'
```

This makes sense because there is no folder for pyrealsense2_net and python does not know how to find it, but the folder for pyrealsense2 appears to contain the net libraries. Is this the expected behavior? And if so how do I get python to recognize these pyrealsense2-net.cpython files as a library?

MartyG-RealSense, 2021-11-11:
Another RealSense user in https://github.com/IntelRealSense/librealsense/issues/7606 also encountered the **ModuleNotFoundError: No module named 'pyrealsense2_net'** error when trying to import pyrealsense2-net.  A RealSense team member asked whether they had **rs-server** running on Linux.  I therefore wonder whether rs-server should be active first before the net-viewer.py example program is run.

Thanks very much for your patience during this case!

colinmsnow, 2021-11-12:
After reading the link you provided it seems like the problem encountered in the issue was that rs-server only supports Linux and so it is not possible to compile pyrealsense2_net on windows. I am working on Linux and building the SDK successfully builds rs-server and puts it in `/usr/local/bin/rs-server`. I assume this is what is needed to build pyrealsense2_net. I can run rs-server and it works fine but it doesn't do anything because there is no realsense connected to the local device. 
```
/usr/local/bin$ ./rs-server
Rs-server is running
Waiting for Device...
```
Running rs-server has no effect on the python modules and I still get the same error.

MartyG-RealSense, 2021-11-12:
If this were the open source ethernet networking project then I would expect to have to make an SSH connection to rs-server from the host computer before being able to stream data from the remote camera to the host.  The **net-viewer.py** Python example has no accompanying documentation to indicate what prerequisites and procedures are needed in order to establish a network connection though.

As the available investigative leads about this subject have been covered already in this discussion, since you are using Linux and Python it may be worth considering using **EtherSense** instead, which was a RealSense networking project designed for Python that was the predecessor of rs-server.

https://github.com/krejov100/EtherSense

colinmsnow, 2021-11-12:
I think my last post might have been a bit confusing... I have a remote device (Raspi 4) which is running rs-server and streaming data to the Jetson. I know this works as I can access the stream both through the realsense-viewer and the c++ API. The `net_viewer.py` file is failing before it even attempts to make this connection when it tries to import the library `pyrealsense2_net`. The issue is not with anything at runtime but with either the library files or the way they are set up on the computer when it is built by the SDK. Is there any prebuilt version of librealsense which includes net libraries I could take a look at to see how the files are arranged? Or any other documentation on pyrealsense_net besides that one example file? I suspect it might be a problem with how the SDK places the build files after building.

I would like to avoid using EtherSense as my server computer is already set up and working reliably and future plans for this project will likely depend on using both c++ and python to access the remote device which is only possible with realsense-net.

MartyG-RealSense, 2021-11-12:
In the ethernet networking white-paper, Intel provide a pre-built 3 GB Raspberry Pi 4 SD card image containing librealsense 2.34.0 and the networking files for testing purposes and to compare the file structure to your own project.  

https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras#section-2-3-preparing-the-sd-card

colinmsnow, 2021-11-15:
I am using that image on my remote device and to my knowledge, while it does build the net device, it does not build python bindings so does not have pyrealsense2 or pyrealsense2_net.

MartyG-RealSense, 2021-11-16:
I reviewed the case again from the beginning but did not find any further leads that mat be helpful unfortunately, as rs_viewer.py was introduced relatively recently in SDK 2.49.0 and so there are few references about it at the time of writing this.  I do apologize.

colinmsnow, 2021-11-17:
I tried the same set of actions on another computer (Dell Latitude running Ubuntu 18.04) and got exactly the same result so I suspect it is not an issue with my machine or configuration but rather in either how the installer places files or a lack of documentation on how to use them after they are built. I noticed that @maloel wrote the net_viewer example and a few others including @lramati seem to be actively developing the python net module. Would any of you be able to provide some clarity on this or @MartyG-RealSense do you know anyone else who might have knowledge about it? Assuming this is not a computer-specific issue it seems like it is something that is pretty essential to the use of the pyrealsense2_net project.

MartyG-RealSense, 2021-11-17:
I have highlighted this case to Intel to seek advice.  Thanks very much for your patience.

MartyG-RealSense, 2021-11-30:
Hi @colinmsnow and @michezio  Intel are still investigating this issue.  Thanks very much for your patience!

MartyG-RealSense, 2021-12-23:
There is currently no further information from Intel to report about this issue.

MartyG-RealSense, 2022-02-09:
Hi @Adamskiiiiii  The most recent discussion of the **ModuleNotFoundError: No module named 'pyrealsense2_net'** error was a week ago at the time of writing this at https://github.com/IntelRealSense/librealsense/issues/10218 though that was again an Ubuntu related case.

The next version of the RealSense networking API is being worked on and due "soon", though I do not have an estimate of a release timetable to provide.

MartyG-RealSense, 2022-02-22:
I have no further information from Intel to provide at the time of writing this.

@colinmsnow  @michezio  @Adamskiiiiii  Could each of you update me about whether you are still experiencing problems with pyrealsense2.net please.  Thanks!

MartyG-RealSense, 2022-02-22:
@michezio  Thanks for the information!

Adamskiiiiii, 2022-02-23:
@MartyG-RealSense Hello, I have also switched to C++ but now I am having another issue. I have been following the instructions of the ethernet networking whitepaper, up to section 3.4 of the ethernet networking whitepaper. After adding the following line:

#include <librealsense2-net/rs_net.hpp>

to my cubemos cpp_realsense code sample on visual studio, these 2 errors appeared:

C1083	Cannot open include file: 'librealsense2-net/rs_net.hpp': No such file or directory	
E1696	cannot open source file "librealsense2-net/rs_net.hpp"	

[image]
(This is to show that the .hpp file does exist, I think)

[image]
(I then added the highlighted line, as instructed under section 3.4 of the same paper, the errors persist still)

If it wasn't obvious enough by now, I am a complete beginner and don't really know what I am doing. I would appreciate any help I can get, perhaps in the form of more detailed, explicit instructions since I have slowly come to realize I am in fact, borderline computer illiterate.

Thank you for your help in advance!

Adamskiiiiii, 2022-02-23:
@MartyG-RealSense pyrealsense2.net still doesn't work, so I have switched to C++ networking instead (and encountered a different issue, at #10262). Thank you for the update anyway!

MartyG-RealSense, 2022-02-23:
Thanks very much @Adamskiiiiii  for your update!  I have replied to https://github.com/IntelRealSense/librealsense/issues/10262

@colinmsnow  I will close this case after 7 days from the time of writing this if I do not hear from you.  Thanks!

colinmsnow, 2022-02-23:
I was never able to resolve this issue or get pyrealsense2.net working in any capacity. We have moved on to another solution.
Thanks so much for trying!

MartyG-RealSense, 2022-02-23:
Thanks very much @colinmsnow for the update!  As everyone involved in this case has either moved to C++ or chose another solution, I will close this case.  Thanks again!

closed 2022-02-23 by MartyG-RealSense

deepak-asink, 2022-03-10:
Hello MartyG, 
I am also facing the same problem, Is there any update from intel side.

MartyG-RealSense, 2022-03-10:
Hi @deepak-asink   There is no update to provide regarding issues with pyrealsense2.net.  Intel are working on the next version of RealSense networking, although I do not have a time estimate available for its release.

You could explore whether the Python **EtherSense** ethernet networking project in the link below that was the predecessor of the current networking system can meet your needs.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python/examples/ethernet_client_server

deepak-asink, 2022-03-14:
Thanks @MartyG-RealSense , I am able to use Ethersense for my project but the frames I am getting are at very low fps on client side. Can you suggest some fix for that.

MartyG-RealSense, 2022-03-14:
A general rule for RealSense ethernet networking will be that the connection will have a faster data transfer rate than USB 2 but slower than USB 3.  This is described in the table below,.

[image]

At https://github.com/IntelRealSense/librealsense/issues/8213#issuecomment-769048556 there was also a case with a Python ethernet setup resembling the EtherSense project where connection problems were found to be caused by use of USB 2 cabling instead of USB 3.

Adamskiiiiii, 2022-04-08:
I solved the problem by replacing `import pyrealsense2_net as rsnet` with `from pyrealsense2 import pyrealsense2_net as rsnet`.

MartyG-RealSense, 2022-04-08:
Excellent to hear, @Adamskiiiiii - thanks so much for sharing the solution that worked in your particular case  :)

CharmingZh, 2023-05-08:
> from pyrealsense2 import pyrealsense2_net as rsnet

I tried to replacing the code with `from pyrealsense2 import pyrealsense2_net as rsnet`, but I got an error:
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: cannot import name 'pyrealsense2_net' from 'pyrealsense2' (.../Anaconda/anaconda3/envs/KAN-Forest/lib/python3.10/site-packages/pyrealsense2/pyrealsense2.cpython-310-x86_64-linux-gnu.so)
```
Could you please tell me how you managed to install the `pyrealsense2_net` module?

maloel, 2023-05-09:
Hi @CharmingZh 
The realsense2_net library, and its accompanying Python module, have been deprecated and are no longer available except in older versions of the library. It is no longer supported.

CharmingZh, 2023-05-09:
@maloel Okay, that's soooo sad ; (
 thanks bro, it seems that I have to switch my project into C++.

elliotstronge, 2025-03-03:
@MartyG-RealSense why was it deprecated? It was very useful.

MartyG-RealSense, 2025-03-03:
Hi @elliotstronge  The realsense2_net networking was an experimental feature rather than a core part of the librealsense SDK and it was removed in librealsense version 2.54.1 as part of a clean-up initiative to minimize the size and complexity of the SDK.  Versions 2.53.1 and earlier still have support for it.

elliotstronge, 2025-03-03:
@MartyG-RealSense I have seen in other discussion threads here there is/was a plan to replace the networked devices, is that still planned?

MartyG-RealSense, 2025-03-04:
A replacement was planned for the near future after the networking tool was removed but it did not go ahead.  I don't have information on future plans for RealSense networking support at the time of writing this.  Typically, new features are announced on the day of an SDK update's release and not in advance.