realsenseai/librealsense #9947: Depth stream start failure - Right MIPI error

MoaazAbdulrahman, 2021-11-10:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435i | 
| Firmware Version                   | (05.12.15.50 | 
| Operating System & Version |  windows 10  | 
| Platform                                 | PC  |
| SDK Version                            |  2.49                         | 

I am running realsense viewer and I have problem with the depth sensor I am getting "No frames received" issue. It says in the debugger "righ MIPI error" and not able to start streaming. The RGB and camera and motion module work well. I have tried to calibrate the camera using dynamic calibration tool but the stream does not start as well.

MartyG-RealSense, 2021-11-11:
Hi @MoaazAbdulrahman  When you have experienced the Right MIPI error, have you enabled three streams (the Depth, RGB and Motion Module) all at the same time?  If you have then I recommend not enabling three streams if one of them is the Motion Module.  For some RealSense users, this can result in No Frames Received in the depth or RGB stream.  It is a known issue and there is no fix for it in the Viewer.  It does not matter which order the streams are enabled in, as the Motion Module activation is the trigger.

If you need to display depth, RGB and Motion Module simultaneously, the RealSense SDK's example program **rs-multicam** (which works with a single camera) can do so without the No Frames Received issue occurring.  

If you have installed the full RealSense SDK on Windows then you can find a pre-built executable version of rs-multicam and other example programs by right-clicking on the Viewer's launch shortcut icon on the Windows desktop and selecting the 'Open file location' menu option to be taken to the SDK's Tools folder.

When creating your own application, a workaround is to use **callbacks** or to define two separate pipelines, with depth and RGB on one pipeline and the Motion Module alone on the other pipeline.

If you have experienced the Right MIPI issue without enabling three streams simultaneously, please see whether going to the 'More' option near the top of the Viewer's options side panel and selecting 'Reset Hardware' from its drop down menu corrects the problem.

MoaazAbdulrahman, 2021-11-11:
Thank you so much for your kind response. I face the error for both cases, three streams and rinning depth only. I have followed the solutions that you mentioned here
https://support.intelrealsense.com/hc/en-us/community/posts/360046810833-Intel-Realsense-D435i-is-detected-but-shows-an-error-no-frames-detected-
But I am still facing the same error. 
Could it be hardware problem ?

MartyG-RealSense, 2021-11-11:
As you have already installed the Dynamic Calibration tool, please try performing a **gold reset** of the camera to its factory defaults using the **CustomRW** tool that is installed along with the Dynamic Calibration one.

CustomRWthat can be launched from the Windows Command Prompt text interface.  You can find the Command Prompt by typing **command** into the text box at the base of the Windows screen, next to the button in the corner where the PC is shut down.

Once the Command Prompt window is open, you can activate the factory reset function of CustomRW by launching it in the Command Prompt with the command below:

**Intel.RealSense.CustomRW -g**

MoaazAbdulrahman, 2021-11-11:
I have successfully run a golden reset as seen below.

[image: image.png]

But the same error still exists. I have tried to change the resolution but
nothing happens.

I am not able to calibrate the camera using dynamic calibration. I am
getting the below output and the stream does not begin.
[image: image.png]

MoaazAbdulrahman, 2021-11-11:
Here is the error that I get when running the depth module.
[image: image.png]

MartyG-RealSense, 2021-11-12:
Images added by email do not appear on the website.  Could you come to the website and post the image here, please?  you can add an image by copy and pasting it in the comment box, by drag and dropping the image file onto the comment box or selecting it by left-clicking on the bottom of the box.  Thanks!

https://github.com/IntelRealSense/librealsense/issues/9947#issuecomment-966624535

MoaazAbdulrahman, 2021-11-12:
golden reset output

[image]

realsense viewer error:

[image]

dynamic calibration output:

[image]

MartyG-RealSense, 2021-11-12:
Thanks very much for the images.  The new **2.50.0** version of the RealSense SDK was released yesterday.  Would it be possible to install it using the automated installer file **Intel.RealSense.SDK-WIN10** in the 'Assets' file list of that version and see whether it resolves your problem, please?

https://github.com/IntelRealSense/librealsense/releases/tag/v2.50.0

[image]

MoaazAbdulrahman, 2021-11-12:
@MartyG-RealSense Thank you for your advice. I have installed the new version of SDK 2.50 and tried to run the depth module I got the same error. So, I updated the FW version to be 5.13.00.50 and I am getting the same error and nothing changed.

MartyG-RealSense, 2021-11-12:
Has the depth stream ever functioned since the camera was purchased or did it work at first and then develop the problem later?

MoaazAbdulrahman, 2021-11-12:
Yes, when I purchases it the depth modeule was working well. It was a year ago when I stopped using depth and I was using rgb only. When I tried the depth a week ago I encountered this error.

MartyG-RealSense, 2021-11-12:
Have you placed the camera in storage unused for a period of time in a dry, cool place before using the camera again and encountering the depth problem?  Examples of such places could include a desk drawer or cupboard.

MoaazAbdulrahman, 2021-11-12:
Yes, it has been stored in my desk drawer for months.

MartyG-RealSense, 2021-11-12:
There have been a couple of past cases where a camera was stored inactive for weeks or months and then did not work correctly afterwards when first used (for example, having No Frames Received on the RGB stream).  In case https://github.com/IntelRealSense/librealsense/issues/9638 the camera was shipped internationally to Intel in California, US and was working correctly by the time that it arrived.  In the other case https://github.com/IntelRealSense/librealsense/issues/4376 the camera started working again a few weeks after coming out of storage and being initially non-working.  So it may be worth placing the camera in a warm location to see whether it starts working afterwards.

MartyG-RealSense, 2021-11-18:
Hi @MoaazAbdulrahman  Do you have an update about this case that you can provide, please?  Thanks!

MoaazAbdulrahman, 2021-11-20:
unfortunately, I am still facing the same issue.

MartyG-RealSense, 2021-11-20:
Can you confirm that you placed the camera in a warm location for an extended period of time as suggested above and the problem still remains, please?

MoaazAbdulrahman, 2021-11-20:
Yes, I already did.

MartyG-RealSense, 2021-11-20:
Thank you.  I will seek further advice from Intel about your case.  Thanks very much for your patience!

MartyG-RealSense, 2021-11-21:
Intel have requested that you return the camera to the retailer that you purchased it from using the **Return Material Authorization (RMA)** returns process of that retailer.

Could you also email me the serial number of the camera at the address below please.

martyx.grover@intel.com

MoaazAbdulrahman, 2021-11-23:
Thank you for your support. I will send you the serial number on email.

MartyG-RealSense, 2021-11-24:
Thanks very much @MoaazAbdulrahman , I have received the serial number and passed it to Intel.

MartyG-RealSense, 2021-12-05:
Hi @MoaazAbdulrahman  Do you have an update that you can provide about the progress of your camera return, please?  Thanks!

MartyG-RealSense, 2022-01-22:
This case will be closed after a week from the time of writing this if we do not receive further comment.  Thanks!

closed 2022-01-28 by MartyG-RealSense

MartyG-RealSense, 2023-12-11:
Hi @ranjithlesoko  Was this the first time that the camera was used since November 2 2023, and had the camera been stored in a cool dry place such as a cupboard or drawer for weeks before then?