realsenseai/librealsense #9949: UWP Driver Roadmap

electro-logic, 2021-11-10:
Hello,

Will the UWP driver be updated?

I'm specifically looking for BGRA8 Color Stream and CoordinateMapper support.

Thank you,
Leonardo

MartyG-RealSense, 2021-11-11:
Hi @electro-logic  In the past, occasional updates to the UWP driver have arrived unannounced and not followed a schedule.  Whilst I do not have information about future plans for the  UWPdriver, my expectation would be that further features will not be added to it in the foreseeable future.

electro-logic, 2021-11-11:
Hello Marty,

Thank you for the quick reply, 

Any chance the UWP driver will be made open source?

Thanks again,
Leonardo

MartyG-RealSense, 2021-11-11:
You are very welcome.  Intel do not open-source their RealSense camera drivers, firmwares and algorithms.  My understanding is that they are considered to be Intel's private intellectual property.

MartyG-RealSense, 2021-11-17:
Hi @electro-logic  Do you require further assistance with this case, please?  Thanks!

electro-logic, 2021-11-17:
If the UWP driver is getting discontinued, I hope that Intel will open source the driver by keeping only confidential parts in binaries.

Anyway thank you very much for the support.

closed 2021-11-17 by electro-logic