realsenseai/librealsense #9950: SEH exception error

kalai2875, 2021-11-10:
| | |
|---|---|
|**librealsense**|2.49.0 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
Program terminated due to an unrecoverable SEH exception:
Access Violation!
```
I have used the earlier version (2.48 Release) also. It is working with my desktop. When I tried to install it in my laptop, it was giving the above error. 
I tried all the solutions provided in the similar threads.
I have changed the 'Privacy settings' 
I have disabled my web camera to avoid any conflict
I have tried uninstalling and installing the drivers (Both RGB and depth camera)
Nothing is helping me to resolve the issue. Please give me a suitable solution.

MartyG-RealSense, 2021-11-11:
Hi @kalai2875  I will also add from your other posted case at https://github.com/IntelRealSense/librealsense/issues/9948 that you have tried the advice at https://github.com/IntelRealSense/librealsense/issues/7435#issue-710568759 without success.

In cases where the camera works correctly on one computer but not on another and most possible causes have been eliminated through careful testing, I would recommend testing whether the **USB controller** of the USB port is the cause.  This could be done by attaching the camera to a different USB port to the one that it is currently using, perhaps by adding a USB hub if you are not using a hub already (as a hub will have its own separate USB controllers).

kalai2875, 2021-11-11:
> Hi @kalai2875 I will also add from your other posted case at #9948 that you have tried the advice at [#7435 (comment)](https://github.com/IntelRealSense/librealsense/issues/7435#issue-710568759) without success.
> 
> In cases where the camera works correctly on one computer but not on another and most possible causes have been eliminated through careful testing, I would recommend testing whether the **USB controller** of the USB port is the cause. This could be done by attaching the camera to a different USB port to the one that it is currently using, perhaps by adding a USB hub if you are not using a hub already (as a hub will have its own separate USB controllers).

I have tested my both USB ports. They are working fine. I still get the same issue.
I tried using CustomRW to 'gold rest' the camera to its factory defaults, as described in #9947 (comment). This also did not help.

kalai2875, 2021-11-11:
I have one point to mention here. The other system is not having any other camera. My realsense 430 camera works as expected when it is the only camera that is connected.
But I am getting the SEH error when I am connecting it with a laptop which has the integrated webcam in it (Eventhough I have disabled it, I still get SEH error)

MartyG-RealSense, 2021-11-11:
RealSense SDK version **2.50.0** is now available.  Would it be possible to test whether the issues that you have been experiencing in librealsense 2.49.0 are also present in 2.50.0 please?

https://github.com/IntelRealSense/librealsense/releases/tag/v2.50.0

kalai2875, 2021-11-14:
I have tried this. Now I dont get SEH error. My device was detected. But I get only the empty white screen with the text 'No Frames Received'. But in my other computer, I am getting image frames, immediately.
[image]

MartyG-RealSense, 2021-11-14:
Your kindly provided image indicates that the camera is being detected as being on a USB 2 connection.  Can you confirm whether the camera is plugged into a USB 2 or USB 3 port on the affected computer, please?

[image]

kalai2875, 2021-11-16:
Hi MartyG, Thank you very much. I have changed the port (to USB 3.0). Now it works!!! Thanks for your support.

MartyG-RealSense, 2021-11-16:
That's excellent news, @kalai2875 - thanks very much for the update!

closed 2021-11-23 by MartyG-RealSense