realsenseai/librealsense #9951: Is it possible to run box_dimensioner_multicam python after saving the cailbration value?

HyeYumKim, 2021-11-11:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Platform                                 | PC |
| Language                            | python | 

First of all, I want to calibrate the chess board and save the values, and when I run it again under the same conditions, I want to run it with the previously saved calibration values without the chess board.
In this case, what is the way?

MartyG-RealSense, 2021-11-11:
Hi @HyeYumKim  If you are using a single camera with box_dimensioner_multicam then a RealSense team member advises in https://github.com/IntelRealSense/librealsense/issues/3073#issuecomment-456722091 that you could use the program without a chessboard by providing the intrinsics and extrinsics of that single camera instead of calibrating multiple cameras together.  If the same camera will always be used, you could hard-code the values into the program instead of retrieving them in real-time, like the example of Python code for doing so that is provided in https://github.com/IntelRealSense/librealsense/issues/4061

MartyG-RealSense, 2021-11-18:
Hi @HyeYumKim  Do you require further assistance with this case, please?  Thanks!

HyeYumKim, 2021-11-23:
It's a different topic, can you tell me what's the difference between on chip cailbration and box measurement python code?
Could it be more accurate with on chip cailbration?

MartyG-RealSense, 2021-11-23:
On-Chip calibration and the calibration that box_dimensioner_multicam performs are unrelated.  box_dimensioner_multicam calculates the 3D world coordinates of each attached camera in relation to its position near the chessboard object.  Unlike the On-Chip calibration tool, it is not calibrating the camera's sensors to improve the quality of the images that they generate.

If you wish to improve depth measuring accuracy then I recommend performing a **Tare calibration** instead of On-Chip.

https://dev.intelrealsense.com/docs/self-calibration-for-depth-cameras#section-3-depth-accuracy

closed 2021-12-05 by MartyG-RealSense