realsenseai/librealsense #9958: When the d435i camera is in use, because the external power is off or the USB cable falls off, and then it is powered on again or when the USB cable is inserted, the program will perform the operation of connecting the camera again. Occasionally, there will be an abnormal error of unreasonably blocking for a long time, and finally report mfcreatedevicesource... How to eliminate this blocking and throw an exception immediately!!!

2513297244, 2021-11-12:


MartyG-RealSense, 2021-11-12:
An approach that a RealSense user in https://github.com/IntelRealSense/librealsense/issues/8393 used with Python code was to check whether frames were being received, and if they weren't then jump to a **hardware_reset()** routine to reset the camera.  Presumably, the frame receival routine would continue to check whether frames were now being received, and repeat the jump to the reset routine if they were not.

If the problem in your application generates an exception then you could also add an excepton 'catch' routine to your application to do something if an exception occurs instead of allowing the program to crash and exit.  Examples of error-handling catch code are provided in the links below.

**C++**
https://dev.intelrealsense.com/docs/librealsense-error-handling-scheme

**Python**
https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/python-tutorial-1-depth.py#L49

MartyG-RealSense, 2021-11-18:
Do you require further assistance with this case, please?  Thanks!

2513297244, 2021-11-18:
> 请问您在这个案例中需要进一步的帮助吗？谢谢！

No, thank you. Thank you very much for your answer. I wish you a happy mood!

MartyG-RealSense, 2021-11-18:
Thanks very much.  As you are satisfied with the answer, I will close the case.  Best wishes to you too!

closed 2021-11-18 by MartyG-RealSense