realsenseai/librealsense #9960: Why is my resolution only 320 ? L515

CRYENGINE5, 2021-11-14:
[image]

MartyG-RealSense, 2021-11-14:
Hi @CRYENGINE5   Please unplug the micro-sized end of the USB cable from the base of the L515 camera, turn the connector around the other way and re-insert it into the base of the camera (USB Type-C cables are two way insertion at the micro-sized end).  Does this resolve the problem, please?

CRYENGINE5, 2021-11-14:
@MartyG-RealSense , No, only 320 * 240,  
[image]

MartyG-RealSense, 2021-11-14:
Thanks very much for checking.  Are you using the official 1 meter USB cable that was supplied with the camera or a longer USB cable of your own choice?

CRYENGINE5, 2021-11-14:
@MartyG-RealSense , Is there a sticker on the newly purchased l515 product ? I suspect I bought a fake.
[image]

MartyG-RealSense, 2021-11-14:
I compared the box in your image to a genuine unopened camera and it looks the same, though this is no guarantee as fake product packaging can look identical to the real thing.  There should also be a small sticker on the lower side of the camera stating model name, camera serial number and 'Made in Thailand' if it is genuine.  The serial number on the sticker should match the one shown when you click on the 'i' information option in the RealSense Viewer's options side-panel.

CRYENGINE5, 2021-11-14:
Thank you . 
I found it. The USB cable of the product is unstable.

MartyG-RealSense, 2021-11-14:
Is the cable the official 1 meter one that was supplied with the camera or a longer cable of your own choice?

MartyG-RealSense, 2021-11-20:
Hi @CRYENGINE5  Do you require further assistane with this case, please?  Thanks!

closed 2021-11-26 by MartyG-RealSense