realsenseai/librealsense #9962: Error compiling SDK Raspberry Pi 4

xandrei2772, 2021-11-14:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
Trying to follow this installation guide for the realsense camera [guide](https://github.com/acrobotic/Ai_Demos_RPi/wiki/Raspberry-Pi-4-and-Intel-RealSense-D435)
I keep running into this error, wondering if anyone knows how to fix this?
[Error](https://imgur.com/a/Y7kmM5x)

xandrei2772, 2021-11-14:
I get stuck on the part when trying to install the realsense sdk:

cd ~/librealsense
mkdir  build  && cd build
cmake .. -DBUILD_EXAMPLES=true -DCMAKE_BUILD_TYPE=Release -DFORCE_LIBUVC=true
make -j1
sudo make install

I get to make -j1 and run into the error when its 80% done

MartyG-RealSense, 2021-11-14:
Hi @xandrei2772   You may have a smoother build experience if you use a more recent version of protobuf, such as 3.14

**git clone --depth=1 -b v3.14.0 https://github.com/google/protobuf.git**

Using the -j make instruction can also cause random crashes during the build process on low-power devices.  So you could try simply inputting **make** without the -j1 part, though the build time may be longer.

xandrei2772, 2021-11-15:
> Hi @xandrei2772 You may have a smoother build experience if you use a more recent version of protobuf, such as 3.14
>
> **git clone --depth=1 -b v3.14.0 https://github.com/google/protobuf.git**
>
> Using [...]

Have made the changes you suggested and im still getting the same error.

xandrei2772, 2021-11-15:
> Have made the changes you suggested and im still getting the same error.

`[ 56%] Linking CXX executable rs-hello-realsense
/usr/bin/ld: ../../librealsense2.so.2.50.0: undefined reference to `__atomic_load_8'
/usr/bin/ld: ../../librealsense2.so.2.50.0: undefined reference to `__atomic_store_8'
/usr/bin/ld: ../../librealsense2.so.2.50.0: undefined reference to `__atomic_fetch_add_8'
collect2: error: ld returned 1 exit status
make[2]: *** [examples/hello-realsense/CMakeFiles/rs-hello-realsense.dir/build.make:110: examples/hello-realsense/rs-hello-realsense] Error 1
make[1]: *** [CMakeFiles/Makefile2:1240: examples/hello-realsense/CMakeFiles/rs-hello-realsense.dir/all] Error 2
make: *** [Makefile:149: all] Error 2
`

MartyG-RealSense, 2021-11-15:
I have not seen these **latomic** errors on Raspberry Pi installations since 2019.  The most commoly suggestion solution was:

1.  Input the command below: 

**sudo apt-get install libatomic-ops-dev**

2.  Add this line to the top of the CMakeLists.txt file:

**set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -latomic")**

xandrei2772, 2021-11-15:
> I have not seen these **latomic** errors on Raspberry Pi installations since 2019. An example case that discusses the subject is #4950
> 
> The most commonly suggested solution for the latomic errors was the following two steps:
> 
> 1. Input the command below:
> 
> **sudo apt-get install libatomic-ops-dev**
> 
> 1. Add this line to the top of the CMakeLists.txt file:
> 
> **set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -latomic")**

I've made the suggested changes and still getting the same error.

MartyG-RealSense, 2021-11-16:
As mentioned earlier, libatomic is a rare error that mostly has not been seen since 2019.  This suggests that something may have gone wrong during following the Acrobotics installation guide.  Sometimes if an installation is not working and you have tried every kown fix, just wiping everything and re-installing from the very beginning can cause it to suddenly work.

Before doing that though, you could see whether the **libuvc backend** method (the predecessor of RSUSB) in the link below works for you.

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

debOliveira, 2021-11-17:
Hi @MartyG-RealSense , I'm having the same problem here. 

I've wiped everything twice. Just for info, I'm using protobuf 3.10 as told in #5092 and the flag `-DFORCE_LIBUVC=true` like the [Raspbian guide](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_raspbian.md).

Couple of weeks ago I did the same process on a smaller SD card (not changing a cmd line) and it came out clean. It is possible that this is a problem of the latest release?

xandrei2772, 2021-11-17:
> As mentioned earlier, latomic is a rare error that mostly has not been seen since 2019. This suggests that something may have gone wrong during following the Acrobotic installation guide. Sometimes if an installation is not working and you have tried every known fix, just wiping everything and re-installing from the very beginning can cause it to suddenly work.
> 
> Before doing that though, you could see whether the **libuvc backend** method (the predecessor of RSUSB) in the link below works for you.
> 
> https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

Hello Again, 

I've just started my pi from scratch, installed rasbian again and tried following the guide instructions for installing the realsense camera. Unfortunately I ran into the same issue https://imgur.com/a/vGiFWMB

debOliveira, 2021-11-17:
> Hello Again,
>
> I've just started my pi from scratch, installed rasbian again and tried following the guide instructions for installing the realsense camera. Unfortunately I ran into the same issue [...]

Also had the same issue downgrading the librealsense to 2.49.0 and upgrading protobuf to 3.14.0. 
So that's not a release issue.

xandrei2772, 2021-11-17:
> Also had the same issue downgrading the librealsense to 2.49.0 and upgrading protobuf to 3.14.0. So that's not a release issue.

please let me know if you find a solution

debOliveira, 2021-11-17:
> please let me know if you find a solution

Until now, couldn't solve cmake problem. I wrote `-latomic` in all build lines in root, example, and wrapper folder. As for now, I'm working on an old image I saved from that SD card a couple of weeks ago.

What is strange is that, just two weeks ago, all the commands were working fine. Have no idea which lib updated and crashed the build setup.

MartyG-RealSense, 2021-11-18:
Would it be practical for you to attempt a Pi installation with **Ubuntu Server 20.04** instead of Raspian OS using the information referenced in https://github.com/IntelRealSense/realsense-ros/issues/1724 or is Raspbian a requirement of your projects?

MartyG-RealSense, 2021-11-18:
Another RealSense user with a Pi was having similar installation errors with the new Raspbian OS Bullseye on SDK 2.49.0 in the recent case https://github.com/IntelRealSense/librealsense/issues/9943

They were able to compile the SDK in Raspian Buster and then copy 2 binary files to Bullseye, as described in https://github.com/IntelRealSense/librealsense/issues/9943#issuecomment-966048785 and attributed their compilation problems to their gcc version.

debOliveira, 2021-11-18:
As I said, I managed to workaround copying an old SD card image (with pyrealsense already compiled from two weeks ago).

But this doesn't solve the building problem for Raspbian OS.

MartyG-RealSense, 2021-11-19:
Hi @xandrei2772  Do you have any positive progress to report, please?  Thanks!

xandrei2772, 2021-11-19:
> Hi @xandrei2772 Do you have any positive progress to report, please? Thanks!

Hi i've installed ubuntu and now currently installing all the packages for the camera, will let you know how i get on in a bit.
Thanks

MartyG-RealSense, 2021-11-19:
Thanks very much - good luck!

hervst, 2021-11-24:
Hello,

I ran into the same issue a few days ago too. I also copied an earlier compiled version of the RealSense SDK to my Raspberry Pi as workaround.

Today I installed the Raspberry Pi OS 64 Bit (as far as I know in Beta state) version on my Raspberry Pi to do some tests. Fortunately the RealSense SDK compiled without any error. Maybe it was by chance and the issue with a package is fixed already but maybe this issue only affects the 32 Bit version of Raspberry Pi OS? Another difference between the two builds was that I used the lite version of the Raspberry Pi OS 64 Bit. For 32 bit is used the version with desktop environment (but not "full").

I hope this will help to fix the issue.

xandrei2772, 2021-11-26:
> Thanks very much - good luck!

Hey so im trying to install it on ubuntu and everything was going fine until i ran the following command:
`sudo -H pip3 install pyopengl_accelerate==3.1.3rc1`

I got a lot of errors:

```
(name)@(name)-desktop:~/Work/librealsense/build$ sudo -H pip3 install pyopengl_accelerate==3.1.3rc1
Collecting pyopengl_accelerate==3.1.3rc1
  Using cached PyOpenGL-accelerate-3.1.3rc1.tar.gz (532 kB)
Building wheels for collected packages: pyopengl-accelerate
  Building wheel for pyopengl-accelerate (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /usr/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-04z9ijyb/pyopengl-accelerate_f3e5be8205834820ba90cc2c609cad1c/setup.py'"'"'; __file__='"'"'/tmp/pip-install-04z9ijyb/pyopengl-accelerate_f3e5be8205834820ba90cc2c609cad1c/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-m3zg89pv
       cwd: /tmp/pip-install-04z9ijyb/pyopengl-accelerate_f3e5be8205834820ba90cc2c609cad1c/
  Complete output (175 lines):
  Unable to import numpy, skipping numpy extension building
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.linux-aarch64-3.9
  creating build/lib.linux-aarch64-3.9/OpenGL_accelerate
  copying OpenGL_accelerate/__init__.py -> build/lib.linux-aarch64-3.9/OpenGL_accelerate
  running build_ext
  building 'OpenGL_accelerate.wrapper' extension
  creating build/temp.linux-aarch64-3.9
  creating build/temp.linux-aarch64-3.9/src
  aarch64-linux-gnu-gcc -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/tmp/pip-install-04z9ijyb/pyopengl-accelerate_f3e5be8205834820ba90cc2c609cad1c/.. -I/tmp/pip-install-04z9ijyb/pyopengl-accelerate_f3e5be8205834820ba90cc2c609cad1c/src -I/tmp/pip-install-04z9ijyb/pyopengl-accelerate_f3e5be8205834820ba90cc2c609cad1c -I/usr/include/python3.9 -c src/wrapper.c -o build/temp.linux-aarch64-3.9/src/wrapper.o
  src/wrapper.c: In function ‘__Pyx_modinit_type_init_code’:
  src/wrapper.c:24653:56: error: ‘PyTypeObject’ {aka ‘struct _typeobject’} has no member named ‘tp_print’
  24653 |   __pyx_type_17OpenGL_accelerate_7wrapper_cArgConverter.tp_print = 0;
        |                                                        ^
  src/wrapper.c:24674:57: error: ‘PyTypeObject’ {aka ‘struct _typeobject’} has no member named ‘tp_print’
  24674 |   __pyx_type_17OpenGL_accelerate_7wrapper_pyArgConverter.tp_print = 0;
        |                                                         ^
  src/wrapper.c:24695:61: error: ‘PyTypeObject’ {aka ‘struct _typeobject’} has no member named ‘tp_print’
  24695 |   __pyx_type_17OpenGL_accelerate_7wrapper_cArgumentConverter.tp_print = 0;
        |                                                             ^
  src/wrapper.c:24716:58: error: ‘PyTypeObject’ {aka ‘struct _typeobject’} has no member named ‘tp_print’
  24716 |   __pyx_type_17OpenGL_accelerate_7wrapper_returnConverter.tp_print = 0;
        |                                                          ^
  src/wrapper.c:24737:64: error: ‘PyTypeObject’ {aka ‘struct _typeobject’} has no member named ‘tp_print’
  24737 |   __pyx_type_17OpenGL_accelerate_7wrapper_CArgCalculatorElement.tp_print = 0;
        |                                                                ^
  src/wrapper.c:24758:57: error: ‘PyTypeObject’ {aka ‘struct _typeobject’} has no member named ‘tp_print’
  24758 |   __pyx_type_17OpenGL_accelerate_7wrapper_CArgCalculator.tp_print = 0;
        |                                                         ^
  src/wrapper.c:24769:65: error: ‘PyTypeObject’ {aka ‘struct _typeobject’} has no member named ‘tp_print’
  24769 |   __pyx_type_17OpenGL_accelerate_7wrapper_PyArgCalculatorElement.tp_print = 0;
        |                                                                 ^
  src/wrapper.c:24790:58: error: ‘PyTypeObject’ {aka ‘struct _typeobject’} has no member named ‘tp_print’
  24790 |   __pyx_type_17OpenGL_accelerate_7wrapper_PyArgCalculator.tp_print = 0;
        |                                                          ^
  src/wrapper.c:24801:62: error: ‘PyTypeObject’ {aka ‘struct _typeobject’} has no member named ‘tp_print’
  24801 |   __pyx_type_17OpenGL_accelerate_7wrapper_CArgumentCalculator.tp_print = 0;
        |                                                              ^
  src/wrapper.c:24814:62: error: ‘PyTypeObject’ {aka ‘struct _typeobject’} has no member named ‘tp_print’
  24814 |   __pyx_type_17OpenGL_accelerate_7wrapper_CallFuncPyConverter.tp_print = 0;
        |                                                              ^
  src/wrapper.c:24847:60: error: ‘PyTypeObject’ {aka ‘struct _typeobject’} has no member named ‘tp_print’
  24847 |   __pyx_type_17OpenGL_accelerate_7wrapper_DefaultCConverter.tp_print = 0;
        |                                                            ^
  src/wrapper.c:24866:50: error: ‘PyTypeObject’ {aka ‘struct _typeobject’} has no member named ‘tp_print’
  24866 |   __pyx_type_17OpenGL_accelerate_7wrapper_Wrapper.tp_print = 0;
        |                                                  ^
[... cut ...] setup.py'"'"'; __file__='"'"'/tmp/pip-install-04z9ijyb/pyopengl-accelerate_f3e5be8205834820ba90cc2c609cad1c/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-ibqqr8hy/install-record.txt --single-version-externally-managed --compile --install-headers /usr/local/include/python3.9/pyopengl-accelerate Check the logs for full command output.
```

MartyG-RealSense, 2021-11-26:
Hi @xandrei2772  Does it make any difference if you install pyopengl_accelerate without specifying a version?

**sudo -H pip3 install pyopengl_accelerate**

This version of the command is used by Intel's official Raspbian installation guide for Pi 3.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_raspbian.md#change-pi-settings-enable-opengl

xandrei2772, 2021-11-26:
> Hi @xandrei2772 Does it make any difference if you install pyopengl_accelerate without specifying a version?
> 
> **sudo -H pip3 install pyopengl_accelerate**
> 
> This version of the command is used by Intel's official RealSense Raspbian installation guide for Pi 3.
> 
> https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_raspbian.md#change-pi-settings-enable-opengl

thats perfect thanks, i went ahead and ran that command worked perfectly!

running into a different error now, im trying to enable OpenGL by going into raspi-config > advanced opentiosn > GL driver > GL (Fake KMS) 

soon as i open gl driver tab i get a message saying "Driver and kernel not present on your system. please update." cant seem to find much written about this online

MartyG-RealSense, 2021-11-26:
Before opening the GL driver tab, did you install raspi-config with the instruction below?

**sudo raspi-config**

xandrei2772, 2021-11-26:
> Before opening the GL driver tab, did you install raspi-config with the instruction below?
> 
> **sudo raspi-config**

yes i ran sudo raspi-config

error pic: http://prntscr.com/20xbrkh

MartyG-RealSense, 2021-11-26:
One person who had this error resolved it by updating their kernel.

A procedure for updating a Pi is below.

```
sudo apt update
sudo apt full-upgrade
```

xandrei2772, 2021-11-26:
> One person who had this error on Pi resolved it by updating their kernel.
> 
> A procedure for updating a Pi is below.
> 
> ```
> sudo apt update
> sudo apt full-upgrade
> ```

just after trying out the 2 commands you suggested and im still encountering the same issue

xandrei2772, 2021-11-26:
> One person who had this error on Pi resolved it by updating their kernel.
> 
> A procedure for updating a Pi is below.
> 
> ```
> sudo apt update
> sudo apt full-upgrade
> ```

found a forum where someone had the same issue but im having a hard time understanding the solution,
https://hub.mender.io/t/issue-with-gl-driver-on-raspbian-stretch/906

MartyG-RealSense, 2021-11-26:
I could not determine a clear solution from the linked page either, though as it is related to Mender ("an open source over-the-air (OTA) software updater for embedded Linux devices"), the advice on that page may not be relevant to our particular case.

Next, could you try installing the graphics drivers and then try opening the GL driver tab again afterwards.

`sudo apt-get install mesa-utils`

xandrei2772, 2021-11-26:
> I could not determine a clear solution from the linked page either, though as it is related to Mender ("an open source over-the-air (OTA) software updater for embedded Linux devices"), the advice on that page may not be relevant to our particular case.
> 
> Next, could you try installing the OpenGL graphics drivers and then try opening the GL driver tab again afterwards.
> 
> `sudo apt-get install mesa-utils`

I tried doing that earlier on to see if that'd change anything, but it doesn't, still experiencing the same problem

MartyG-RealSense, 2021-11-26:
Can you test whether it is possible to launch the **realsense-viewer** program that makes use of OpenGL.

`realsense-viewer`

xandrei2772, 2021-11-26:
> Can you test whether it is possible to launch the **realsense-viewer** program that makes use of OpenGL.
> 
> `realsense-viewer`
> 
> This should be done after rebooting the Pi with `sudo reboot`

the application open and just freezes, i keep getting this warning:
 26/11 15:43:37,763 WARNING [281473064272160] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Success, number: 0

MartyG-RealSense, 2021-11-26:
Next, please test a text-only example program to confirm that the problem is with OpenGL and not with communication with the camera in general.

`rs-hello-realsense`

xandrei2772, 2021-11-26:
> Next, please test a text-only example program to confirm that the problem is with OpenGL and not with communication with the camera in general.
> 
> `rs-hello-realsense`

device is not working anymore?

andrei@andrei-desktop:~$ rs-hello-realsense
RealSense error calling rs2_pipeline_start(pipe:0xaaaafceaf650):
    No device connected

MartyG-RealSense, 2021-11-26:
Do you get the same non-detection with `rs-enumerate-devices`

Also, which RealSense camera model are you using, please?

waversrijan, 2021-11-29:
same problem on raspbian. No luck with any solution.

MartyG-RealSense, 2021-11-30:
Hi @waversrijan  Which Raspbian installation guide are you using, please?

waversrijan, 2021-11-30:
> Hi @waversrijan Which guide for installing the SDK on Raspbian are you using, please?

this one:
https://github.com/acrobotic/Ai_Demos_RPi/wiki/Raspberry-Pi-4-and-Intel-RealSense-D435

MartyG-RealSense, 2021-11-30:
@waversrijan   Have you tried the **libuvc backend** installation method yet?

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

waversrijan, 2021-11-30:
@MartyG-RealSense yes I did. It resulted in the same error. I reinstalled the OS and am trying again. Will update if there’s any progress.

waversrijan, 2021-12-01:
>Have you tried the libuvc backend installation method yet?
https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

@MartyG-RealSense it didn’t work. Still the same error.

MartyG-RealSense, 2021-12-02:
There was a RealSense user with these three **undefined reference** errors (load, store. fetch) at https://github.com/alexa/avs-device-sdk/issues/1404#issuecomment-531228489 who resolved them by updating their gcc version to gcc-5.  This gcc update process is also described in the link below.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md#building-librealsense2-sdk

MartyG-RealSense, 2021-12-08:
Hi @waversrijan  Do you require further assistance with this case, please? Thanks!

waversrijan, 2021-12-08:
> Hi @waversrijan Do you require further assistance with this case, please? Thanks!

No

MartyG-RealSense, 2021-12-08:
@waversrijan   Thanks very much for the update!

@xandrei2772  As the creator of this case, do you still have a problem that needs to be resolved, please?  https://github.com/IntelRealSense/librealsense/issues/9962#issuecomment-980130049 was the most recent message that I sent you.  Thanks!

MartyG-RealSense, 2021-12-14:
Hi @waversrijan  Do you have an update about this case that you can provide, please?  Thanks!

zainmehdi, 2021-12-16:
Having the same issue ? Can anyone provide any update on this  ?

waversrijan, 2021-12-16:
> Hi @waversrijan Do you have an update about this case that you can provide, please? Thanks!
I’ve put the project on pause for now. No progress.

MartyG-RealSense, 2021-12-16:
Okay, thanks very much @waversrijan for the update.  I'm sorry that you haven't achieved the hoped-for outcome so far.

waversrijan, 2021-12-16:
@MartyG-RealSense @zainmehdi I got it to work. I downloaded the preloaded sd card img and went into the buil directory and started installation 

https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras?_ga=2.233919451.437300906.1639682000-1239850054.1639682000#section-2-3-preparing-the-sd-card

MartyG-RealSense, 2021-12-16:
I'm pleased that you found a working solution, @waversrijan  :)

waversrijan, 2021-12-16:
> I'm pleased that you found a working solution, @waversrijan :)

I appreciate your help.

devenjarvis, 2021-12-21:
So I've been hitting the `undefined reference to `__atomic_load_8'` error for a few weeks now. After trying all the CMakeLists modifications that should have fixed it but didn't and tracking a similar issue across other libraries for pi users over the past few months I narrowed it down to some sort of issue with the version of gcc that is on the latest Raspbian. I'm not sure exactly the root cause, but I was finally able to successfully build librealsense on Rasbian on a 32bit Raspbian OS by using Clang instead of gcc with the following steps:

(git cloned the repo in my home dir, installed protobufs, etc per normal instructions)
`sudo apt install clang llvm`
`cd ~/librealsense`
`mkdir build`
`cd build`
`export CC=/usr/bin/clang`
`export CXX=/usr/bin/clang++`
`cmake .. -DBUILD_EXAMPLES=true -DCMAKE_BUILD_TYPE=Release -DFORCE_LIBUVC=true -DOTHER_LIBS="-latomic"` (I'm not sure that the DOTHER_LIBS flag is required, but I had it from previous attempts to fixing this and it worked so I'm leaving it)
`make -j4`
`sudo make install`

Hope this helps the original issue creator and/or others who are battling this issue on the Raspberry Pi right now.

zainmehdi, 2021-12-21:
@devenjarvis Thanks alot you are a life saver. Worked for me too. :D

MartyG-RealSense, 2021-12-21:
Thanks so much @devenjarvis for your helpful advice to this discussion!

MartyG-RealSense, 2021-12-27:
@xandrei2772   Can you confirm whether you were able to achieve a solution on this case, please?  Otherwise, the case will be closed after a week from the time of writing this if we do not hear from you.  Thanks!

MartyG-RealSense, 2022-01-02:
Case closed due to no further comments received.

closed 2022-01-02 by MartyG-RealSense

mathklk, 2022-12-02:
@devenjarvis Thanks a lot, this worked for me!

Just make sure to start with an empty build directory. I had an issue that was caused by some cached files laying around.

jeojeda, 2023-05-15:
@mathklk Do you have the code you used to do the correct installation? I'm still having the problem mentioned above.

mathklk, 2023-05-15:
@jeojeda Have a look at this guide I created https://github.com/mathklk/realsense_raspberry_pi4