realsenseai/librealsense #9963: L515: Depth sensor frames don't arrive

Scratchcat1, 2021-11-14:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 01.05.08.01 | 
| Operating System & Version |   Ubuntu Mate 20.04.1  | 
| Kernel Version (Linux Only)    |  5.4.0                                      | 
| Platform                                 | Raspberry Pi 4 |
| SDK Version                            |  2.49.0                | 
| Language                            |  N/A                    | 
| Segment			|  others                          | 

### Issue Description
Depth sensor previously worked on this system.
Attempting to enable the depth sensor fails either with a frame timeout or fatal error message.
Plugging the sensor into a different system doesn't resolve the issue, nor does swapping out the cable.
```
$ rs-hello-realsense
RealSense error calling rs2_pipeline_wait_for_frames(pipe:0xaaaab28cba40):
    Frame didn't arrive within 15000
```

Enabling the depth sensor in `realsense-viewer` outputs `notifications.cpp:514 - Fatal error occured (16)`.
However the RGB camera and motion sensor can be enabled in the viewer and work fine.

### `dmesg` output
```
[102473.739903] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[102473.834610] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[102473.889788] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[102473.945402] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[102474.177094] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[102474.227699] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[102474.232125] uvcvideo: Non-zero status (-71) in video completion handler.
[102474.279071] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[102474.329620] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[102474.419785] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[102474.485905] hid-sensor-hub 0003:8086:0B64.0004: No report with id 0xffffffff found
[102474.485908] hid-sensor-hub 0003:8086:0B64.0004: No report with id 0xffffffff found
[102475.616594] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[102475.675546] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[102476.931092] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[102484.230743] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[102485.490983] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[102486.743140] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[102488.000075] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[102489.262998] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
```

MartyG-RealSense, 2021-11-14:
Hi @Scratchcat1   There have been a few past reports of this L515 error on Ubuntu, Windows and Mac in https://github.com/IntelRealSense/librealsense/issues/9329 https://github.com/IntelRealSense/librealsense/issues/9333 and https://github.com/IntelRealSense/librealsense/issues/9822

Intel's advice in those cases was to return the camera to the retailer that the camera was purchased from, using the **Return Material Authorization (RMA)** procedure of that retailer.

MartyG-RealSense, 2021-11-20:
Hi @Scratchcat1  Bearing in mind Intel's recommendation above to return the camera to the retailer that it was purchased from, do you require further assistance with this case please?  Thanks!

Scratchcat1, 2021-11-24:
I've returned the camera for replacement so I'll close this issue. Thanks for your help.

closed 2021-11-24 by Scratchcat1