realsenseai/librealsense #9965: Acces Violation: IntelRealSenseViewer 2.49.0

beatrizzgz, 2021-11-15:
| | |
|---|---|
|**librealsense**|2.49.0 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
Unhandled exception escaping from a worker thread!
Error type: Access Violation!
```

MartyG-RealSense, 2021-11-15:
Hi @beatrizzgz  Would it be possible to first test whether the same error occurs in the new **2.50.0** version of the RealSense SDK that was released during the past week?

https://github.com/IntelRealSense/librealsense/releases/tag/v2.50.0

[image]

MartyG-RealSense, 2021-11-22:
Hi @beatrizzgz  Do you require further assistance with this case, please?  Thanks!

beatrizzgz, 2021-11-22:
I haven’t yet tested the new version

MartyG-RealSense, 2021-11-22:
Okay, thanks very much for the update!

MartyG-RealSense, 2021-11-28:
Hi @beatrizzgz  Do you have an update about this case that you can provide, please?  Thanks!

beatrizzgz, 2021-11-28:
Yes, I tried the new version for a few hours and it seems to work fine by now. Thanks for asking :)

closed 2021-11-28 by beatrizzgz

MartyG-RealSense, 2021-11-28:
That's great to hear.   Thanks very much for the update. :)