realsenseai/librealsense #9968: Compile errors with kernel 4.4.194

CRCinAU, 2021-11-16:
----------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Ubuntu Bionic (18.04.6)  | 
| Kernel Version (Linux Only)    | 4.4.194                         | 
| Platform                                 | Firefly Face-RK3399  |
| SDK Version                            |  lN/A                          | 
| Language                            |  N/A                          | 
| Segment			|  N/A | 

### Issue Description
Hi guys,

I am trying to apply the patchset for the realsense camera to address the following stack dumps (trimmed to only show the pixelformat error, and not the entire message):
```
[  497.997345] Unknown pixelformat 0x2036315a
[  498.012891] Unknown pixelformat 0x2036315a
[  498.043059] Unknown pixelformat 0x20493859
```
I have added the following to `uvc_driver.c`:
```
+       {
+               .name           = "Depth 16-bit (INVZ)",```
+               .guid           = UVC_GUID_FORMAT_INVZ,
+               .fcc            = V4L2_PIX_FMT_INVZ,
+       },
+       {
+               .name           = "Depth:IR 16:8 24-bit (INZI)",
+               .guid           = UVC_GUID_FORMAT_INZI,
+               .fcc            = V4L2_PIX_FMT_INZI,
+       },
+       {
+               .name           = "Depth 16-bit (INVR)",
+               .guid           = UVC_GUID_FORMAT_INVR,
+               .fcc            = V4L2_PIX_FMT_INVR,
+       },
+       {
+               .name           = "Raw data 8-bit (RAW8)",
+               .guid           = UVC_GUID_FORMAT_RAW8,
+               .fcc            = V4L2_PIX_FMT_GREY,
+       },
+       {
+               .name           = "Raw data 16-bit (RW16)",
+               .guid           = UVC_GUID_FORMAT_RW16,
+               .fcc            = V4L2_PIX_FMT_RW16,
+       },
+       {
+               .name           = "Luminosity data 8-bit (L8)",
+               .guid           = UVC_GUID_FORMAT_L8,
+               .fcc            = V4L2_PIX_FMT_GREY,
+       },
+       {
+               .name           = "Luminosity data 16-bit (L16)",
+               .guid           = UVC_GUID_FORMAT_L16,
+               .fcc            = V4L2_PIX_FMT_Y16,
+       },
+       {
+               .name           = "Depth data 16-bit (D16)",
+               .guid           = UVC_GUID_FORMAT_D16,
+               .fcc            = V4L2_PIX_FMT_Z16,
+       },
+       {
+               .name           = "Packed raw data 10-bit",
+               .guid           = UVC_GUID_FORMAT_W10,
+               .fcc            = V4L2_PIX_FMT_W10,
+       },
+       {
+               .name           = "Depth:IR 16:8 24-bit (INRI)",
[... 401 lines cut ...]

        stream->bulk.payload_size += ret;
        len -= ret;

        if (buf->bytesused == stream->queue.buf_used ||
            stream->bulk.payload_size == stream->bulk.max_payload_size) {
                if (buf->bytesused == stream->queue.buf_used) {
                        stream->queue.buf_used = 0;
                        buf->state = UVC_BUF_STATE_READY;
                        buf->buf.sequence = ++stream->sequence;
                        uvc_queue_next_buffer(&stream->queue, buf);
                        stream->last_fid ^= UVC_STREAM_FID;
                }

                stream->bulk.header_size = 0;
                stream->bulk.payload_size = 0;
        }

        urb->transfer_buffer_length = stream->urb_size - len;
}
```

As such, none of the patches seem to apply...

Is this what I'm missing for the pixelformat errors I keep seeing in the dmesg? Is there something else missing?

CRCinAU, 2021-11-16:
Full function for `uvc_video_decode_bulk` is available at:
https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/tree/drivers/media/usb/uvc/uvc_video.c?h=linux-4.4.y&id=882f8791e1412d81e5cc7a4c379c73195155b40f#n1232

CRCinAU, 2021-11-16:
When plugging in the camera, I see the following via dmesg:
```
[ 3514.275725] usb 7-1: new high-speed USB device number 4 using xhci-hcd
[ 3514.389820] usb 7-1: New USB device found, idVendor=8086, idProduct=0ad4
[ 3514.389850] usb 7-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 3514.389874] usb 7-1: Product: Intel(R) RealSense(TM) Depth Camera 430 
[ 3514.389896] usb 7-1: Manufacturer: Intel(R) RealSense(TM) Depth Camera 430 
[ 3514.396836] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 430  (8086:0ad4)
```
Then looking through the video devices with `v4l2-ctl`, I see the following:
```
# v4l2-ctl --list-formats-ext -d /dev/video12
ioctl: VIDIOC_ENUM_FMT
	Index       : 0
	Type        : Video Capture
	Pixel Format: 'Z16 '
	Name        : Depth data 16-bit (Z16)
		Size: Discrete 256x144
			Interval: Discrete 0.011s (90.000 fps)
		Size: Discrete 480x270
			Interval: Discrete 0.017s (60.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.167s (6.000 fps)
		Size: Discrete 640x360
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 640x480
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.167s (6.000 fps)
		Size: Discrete 848x480
			Interval: Discrete 0.100s (10.000 fps)
			Interval: Discrete 0.125s (8.000 fps)
			Interval: Discrete 0.167s (6.000 fps)
		Size: Discrete 1280x720
			Interval: Discrete 0.167s (6.000 fps)

	Index       : 1
	Type        : Video Capture
	Pixel Format: 'Z16 '
	Name        : Depth data 16-bit (D16)
		Size: Discrete 256x144
			Interval: Discrete 0.011s (90.000 fps)
		Size: Discrete 480x270
			Interval: Discrete 0.017s (60.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.167s (6.000 fps)
		Size: Discrete 640x360
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 640x480
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.167s (6.000 fps)
		Size: Discrete 848x480
			Interval: Discrete 0.100s (10.000 fps)
			Interval: Discrete 0.125s (8.000 fps)
			Interval: Discrete 0.167s (6.000 fps)
		Size: Discrete 1280x720
			Interval: Discrete 0.167s (6.000 fps)
```
The above triggers the `Unknown pixelformat 0x2036315a` error.

```
# v4l2-ctl --list-formats-ext -d /dev/video14
ioctl: VIDIOC_ENUM_FMT
	Index       : 0
	Type        : Video Capture
	Pixel Format: 'GREY'
	Name        : 8-bit Greyscale
		Size: Discrete 480x270
			Interval: Discrete 0.017s (60.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.167s (6.000 fps)
		Size: Discrete 640x360
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 640x480
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.167s (6.000 fps)
		Size: Discrete 848x480
			Interval: Discrete 0.100s (10.000 fps)
			Interval: Discrete 0.125s (8.000 fps)
			Interval: Discrete 0.167s (6.000 fps)
		Size: Discrete 1280x720
			Interval: Discrete 0.167s (6.000 fps)

	Index       : 1
	Type        : Video Capture
	Pixel Format: 'UYVY'
	Name        : UYVY 4:2:2
		Size: Discrete 480x270
			Interval: Discrete 0.017s (60.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.167s (6.000 fps)
		Size: Discrete 640x360
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 640x480
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.167s (6.000 fps)
		Size: Discrete 848x480
			Interval: Discrete 0.100s (10.000 fps)
			Interval: Discrete 0.125s (8.000 fps)
			Interval: Discrete 0.167s (6.000 fps)
		Size: Discrete 1280x720
			Interval: Discrete 0.167s (6.000 fps)

	Index       : 2
	Type        : Video Capture
	Pixel Format: 'GREY'
	Name        : 8-bit Greyscale
		Size: Discrete 480x270
			Interval: Discrete 0.017s (60.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.167s (6.000 fps)
		Size: Discrete 640x360
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 640x480
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.167s (6.000 fps)
		Size: Discrete 848x480
			Interval: Discrete 0.100s (10.000 fps)
			Interval: Discrete 0.125s (8.000 fps)
			Interval: Discrete 0.167s (6.000 fps)
		Size: Discrete 1280x720
			Interval: Discrete 0.167s (6.000 fps)

	Index       : 3
	Type        : Video Capture
	Pixel Format: 'Y8I '
	Name        : Greyscale 8 L/R (Y8I)
		Size: Discrete 480x270
			Interval: Discrete 0.017s (60.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.167s (6.000 fps)
		Size: Discrete 640x360
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 640x480
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.167s (6.000 fps)
		Size: Discrete 848x480
			Interval: Discrete 0.100s (10.000 fps)
			Interval: Discrete 0.125s (8.000 fps)
			Interval: Discrete 0.167s (6.000 fps)
		Size: Discrete 1280x720
			Interval: Discrete 0.167s (6.000 fps)
```
The above triggers a `Unknown pixelformat 0x20493859` error.

MartyG-RealSense, 2021-11-16:
Hi @CRCinAU  Your problem may be related to the RK3399 chipset.  For an RK3399 based board, I would recommend building the RealSense SDK from source code with the **RSUSB backend** installation method if possible by including the build term ** -DFORCE_RSUSB_BACKEND=true** in the CMake build instruction.  An RSUSB build is not dependent on Linux versions or kernel versions and does not require patching.

A RealSense user with an RK3399-based board shared the RSUSB installation procedure that worked for them at https://github.com/IntelRealSense/librealsense/issues/8850#issuecomment-826303351

CRCinAU, 2021-11-16:
Hi @MartyG-RealSense - It seems that the majority of the kernel changes have worked - however we run librealsense in a docker container - which seems to work fine...

The `Unknown pixelformat` error in the host kernel comes with ~2-3Kb of stack trace - which ends up coming through our entire logging stack - and it happens each time we open the camera device. Multiple that by a few hundred devices and we're talking about a massive combined logspam.

The patches I'm trying to do is purely for silencing the `Unknown pixelformat` logspam...

MartyG-RealSense, 2021-11-16:
In the latest 2.50.0 SDK version Intel released a Docker tutorial with a demo docker file to generate and deploy a containerized version of the SDK.  It may be worth using the guide to produce an updated Docker installation to see whether it rectifies the **Unknown pixelformat** message spam.

https://github.com/IntelRealSense/librealsense/tree/master/scripts/Docker

An alternative approach may be changing the logging level with te instruction **sudo dmesg -n 1** to try to stop a flood of kernel messages in the console.

CRCinAU, 2021-11-16:
Yeah - after your comments, I ended up re-thinking the approach - as importing the full patch set is a bit of a pain - and we'd have to carry that on top of all our other patches forever.... I came to the conclusion that this patch is much easier to maintain:

```
diff --git a/drivers/media/v4l2-core/v4l2-ioctl.c b/drivers/media/v4l2-core/v4l2-ioctl.c
index 29dd31748f18..7347efc86188 100644
--- a/drivers/media/v4l2-core/v4l2-ioctl.c
+++ b/drivers/media/v4l2-core/v4l2-ioctl.c
@@ -1302,7 +1302,7 @@ static void v4l_fill_fmtdesc(struct v4l2_fmtdesc *fmt)
                case V4L2_PIX_FMT_SE401:        descr = "GSPCA SE401"; break;
                case V4L2_PIX_FMT_S5C_UYVY_JPG: descr = "S5C73MX interleaved UYVY/JPEG"; break;
                default:
-                       WARN(1, "Unknown pixelformat 0x%08x\n", fmt->pixelformat);
+                       //WARN(1, "Unknown pixelformat 0x%08x\n", fmt->pixelformat);
                        if (fmt->description[0])
                                return;
                        flags = 0;
```

That stops the logspam dead - and if we still have everything working from within the docker container with librealsense etc, then we really don't need these messages inside the host OS...

MartyG-RealSense, 2021-11-16:
Thanks so much for sharing your solution with the RealSense community   :)

closed 2021-11-16 by CRCinAU