realsenseai/librealsense #9969: Blurry infrared

ZiSo89, 2021-11-16:
| | |
|---|---|
|**librealsense**|2.48.0 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense D430|
|**Serial Number**|031522070161|
|**Firmware Version**|05.13.00.50|
|**Advanced Mode**|YES|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|D400|
|**Asic Serial Number**|052623021751|
|**Firmware Update Id**|052623021751|

The left infrared camera is blurry. And the depth data are not enough to calibrate, I also tried to do factory reset.
[image]

MartyG-RealSense, 2021-11-16:
Hi @ZiSo89  848x100 resolution is the **High Speed Capture** mode, which restricts the vertical field of view in order to be able to provide high FPS without seriously impacting performance.  Intel's white-paper document in the link below about High Speed Capture mode recommends using **300 FPS**. 

https://dev.intelrealsense.com/docs/high-speed-capture-mode-of-intel-realsense-depth-camera-d435

 Could you try selecting 300 FPS instead of 100 in the RealSense Viewer to see whether it improves your results, please?

[image]

ZiSo89, 2021-11-16:
I tried the 300 fps and is not only blur but also flickering.

I have two cameras the other one does not have the same issue.
[image]

MartyG-RealSense, 2021-11-16:
Next, could you swap over the USB cables to test the affected camera with the cable from the correctly performing camera to eliminate the possibility of a bad USB cable?

ZiSo89, 2021-11-16:
I tested it with different USB cables, Unfortunately is not a cable issue.

MartyG-RealSense, 2021-11-16:
Do you experience the same problems with the affected camera if you stream depth and infrared in 848x480 resolution

ZiSo89, 2021-11-16:
Yes the problem is the same.
[image]

MartyG-RealSense, 2021-11-16:
What do the images look like if you increase the **Laser Power** setting under the Controls sectio of the Stereo Module controls to its maximum?

[image]

ZiSo89, 2021-11-16:
[image]

MartyG-RealSense, 2021-11-16:
Next, please try updating your camera's firmware driver to version **5.12.14.50**.  You can do this by following the steps below.

1.  Visit the firmware releases page at the link below and download the firmware file for 5.12.14.50

https://dev.intelrealsense.com/docs/firmware-releases#section-d400-series-firmware-downloads

[image]

2.  After the firmware file has been downloaded, go to the **More** option near the top of the options side-panel of the RealSense Viewer and select the **Update Firmware** option.  A file selector window will pop up.

[image]

3.  Select the downloaded firmware file.  The firmware should now begin installing.

ZiSo89, 2021-11-16:
I update the Firmware, the left infrared is still blur.
[image]

MartyG-RealSense, 2021-11-16:
The depth image looks improved compared to the earlier ones.

You mentioned that you had performed calibration.  Was this the **On-Chip** calibration in the Viewer or with the **Dynamic Calibration** software tool in the link below?

https://www.intel.com/content/www/us/en/download/645988/intel-realsense-d400-series-dynamic-calibration-tool.html

The Dynamic Calibration can provide a robust calibration of the camera.

ZiSo89, 2021-11-16:
I tried only the On-Chip calibration

MartyG-RealSense, 2021-11-16:
Okay, it may therefore be worth downloading and installing the Dynamic Calibration tool from the link above.  

Before doing that though, as you had an improved depth image after the firmware update you coud try the On-Chip calibration again to see whether it can now detect sufficient depth information to perform the procedure.

ZiSo89, 2021-11-17:
Unfortunately I don't have the time to do the Dynamic Calibration. Although the Infrared frames are still blur or flikering at 300fps (this probably is a hardware issue?) the On-Chip calibration succeed this time. I think now I have depth values I can work with. Thanks a lot.
[image]

MartyG-RealSense, 2021-11-17:
It's great to hear that you achieved satisfactory results!

Does flickering reduce if you enable the **Emitter Always On** option under the Controls section of the Stereo Module options?  The projector on the D430 / D435 / D435i models pulses in line with exposure, whilst on the D415 model the projector is always on.  Enabling 'Emitter Always On', as the name suggests, keeps the projector always on instead of pulsing.

[image]

Flickering may also be caused if there are fluorescent lights such as ceiling strip lights in the location where the camera is being used.  Such lights contain a heated gas that flickers at frequencies that are difficult to see with the human eye.  The negative effects on the image caused by the lights can be reduced by matching the FPS as closely as possible with the operating frequency of the lights.  For some lights it may be 30 FPS and for others it may be 60 FPS.  Fluorescent lights in European regions may operate on 50hz.  

There was also a case with a medical monitoring project that ran for multiple hours including during the night, and the RealSense user in that case addressed flickering under fluorescent lights by doing camera calibration and performing post-processing filtering.

https://support.intelrealsense.com/hc/en-us/community/posts/360047112993/comments/360011849194

closed 2021-11-17 by ZiSo89