realsenseai/librealsense #9974: When will v2.50.0 be exiting pre-release?

nkalupahana, 2021-11-16:
The latest release of `realsense-ros` requires librealsense v2.50.0, but it looks like that release here is still in pre-release. When will it become a normal release?

MartyG-RealSense, 2021-11-16:
Hi @nkalupahana   I do not have information available about when the release status of 2.50.0 might change.

GitHub's documentation states that the pre-release status is to "notify users that the release is not ready for production and may be unstable".  If you would be using 2.50.0 for a non-commercial purpose such as project prototyping then it is fine to use it.  And if you had a production-level project such as a commercial product or a critical project that requires solid stability then I would recommend not using the latest SDK version in it anyway.

A good rule to apply for production-level projects is 'one version back' - using one SDK and firmware version older than the most recent one.  This will provide time to see whether any issues arise from a new release.  If you need to use ROS though then I would recommend using SDK 2.48.0 when applying this strategy, as 2.49 does not have a matching RealSense ROS wrapper.

MartyG-RealSense, 2021-11-23:
Hi @nkalupahana  Do you require further assistance with this case, please?  Thanks!

nkalupahana, 2021-11-23:
I was hoping someone would know when the release status would change, but it doesn't seem like anyone does, so that's okay, I'll close.

closed 2021-11-23 by nkalupahana

MartyG-RealSense, 2021-12-20:
Hi @nkalupahana  SDK version 2.50.0 is now updated from Pre-Release to Latest.

[image]