realsenseai/librealsense #9975: "rs2_pipeline_wait_for_frames - pipe:0x2326020 :Frame didn't arrive within 5000" after running normally several hours.

hsyzhxy, 2021-11-17:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415                                                       | 
| Firmware Version                   | 05.12.07.150                                           | 
| Operating System & Version | 20.04.1-Ubuntu                                      | 
| Kernel Version (Linux Only)    |  5.11.0-38-generic                                 | 
| Platform                                 | NUC                                                       |
| SDK Version                           |  { legacy / 2.<?>.<?> }                          | 
| Language                              |  C++                                                      | 
| Segment		    	       |      others }                          | 

 Issue Description:
I copied directly a librealsense2.so.2.32 lib to my NUC and there are 4 USB3.1 port on the NUC. 
The app call pipe.wait_for_frames() to get the frames. At first it worked normal, but it caught error "rs2_pipeline_wait_for_frames - pipe:0x2326020 :Frame didn't arrive within 5000" after running several hours. the code as attached.
Sometimes I powered down and restart the NUC , restart the APP, or switch to another USB3.1 port, it can work normally again.

But  It's so annoying. I could add more information if you need. Any help or other suggestions would greatly appriciated!

hsyzhxy, 2021-11-17:
[image]

MartyG-RealSense, 2021-11-17:
Hi @hsyzhxy  If you are using librealsense SDK version **2.32.0** as your reference to librealsense2.so.2.32 suggests then using Ubuntu 20.02 and kernel 5.11 would not be a good match for 2.32.  This is because official support for Ubuntu 20.02 was not added until SDK 2.40.0 and kernel 5.11 is not officially supported by the SDK yet (it can work with newer kernels but there may be unpredictable consequences for stability).

Your firmware driver version is an appropriate choice for the 2.32 SDK though.

More information about 20.02 support from SDK 2.40 onwards is provided at the link below.

https://github.com/IntelRealSense/librealsense/wiki/Release-Notes#release-2400

You may be able to improve stability if you build librealsense from source code instead of from packages by using the **RSUSB backend** installation method, which is not dependent on Linux versions or kernel versions and does not require patching.  RSUSB was introduced in SDK 2.30.0, so you should be able to install 2.32 with that method.  The procedure for doing so is described in https://github.com/IntelRealSense/librealsense/issues/9872#issuecomment-948727294

It is also worth mentioning that when a failure occurs in a long-run session, it may be due to a problem with the computer hardware, operating system or USB equipment rather than the camera hardware system itself.  RealSense cameras have been regularly tested by Intel for long-running and can operate for an indefinite period of time so long as they remain within recommend operating temperature tolerances.

hsyzhxy, 2021-11-17:
Thanks a lot!
I will  build librealsense from source code to install it. 
Do I need to change the version？  Do you mean  2.32 can match my Ubuntu 20.04 environment with this  installation method?

hsyzhxy, 2021-11-17:
Also， What is your recommended temperature tolerance？ The temperature of Our NUC computer will be a little high in  a long-run session.

MartyG-RealSense, 2021-11-17:
I have not seen a case of Ubuntu 20.02 being used with SDK 2.32 but since an RSUSB build is not dependent on Linux versions, it should be able to work in theory.

The maximum recommended operating temperature in the data sheet document for the 400 Series cameras used to be 35 degrees C (though in real-world conditions the camera could likely go to 42 degrees before problems started manifesting beyond that).  The data sheet now expresses the maximum in terms of a case temperature of 50 degrees C.

[image]

hsyzhxy, 2021-11-18:
I met “Unsupported distribution focal kernel version 5.11.0-40-generic . The patches are maintained for Ubuntu16/18 (Xenial/Bionic) with LTS kernels 4-[4,8,10,13,15,18]” while running“ ./scripts/patch-realsense-ubuntu-lts.sh”， the screenshot as attached.
Is this our expectation？ What should I do next？

Thanks！

hsyzhxy, 2021-11-18:
[image]

hsyzhxy, 2021-11-18:
I met this problem using sdk2.32

hsyzhxy, 2021-11-18:
I use SDK version 2.50 instead, but I met “Replacing videodev  -
	Applying the patched module ... modprobe: ERROR: could not insert 'videodev': Exec format error” while run “ ./scripts/patch-realsense-ubuntu-lts.sh”， the detail as attached

closed 2021-11-30 by MartyG-RealSense