realsenseai/librealsense #9977: Minimum focusing distance of D455

vikove, 2021-11-17:
What is the minimum object distance of RealSense D455? I am planning to buy RealSense D455 for my project where the required minimum object distance has to be around 30mm to 50mm. I also read in another post (regarding D405) that we can reduce the minimum distance by increasing the **Disparity Shift** setting. Does the same apply here too?

Thanks

MartyG-RealSense, 2021-11-17:
Hi @vikove  The minimum distance of the D455 is 0.4 meters / 40 cm.  You can use Disparity Shift with the D455 to reduce minimum distance, yes.  This method has the trade-off of also reducing maximum observable depth distance, though this will not matter so much if you are performing close-range sensing.

You could consider using the D435i model instead.  It has an IMU like the D455 does and the benefit of a minimum depth sensing distance of 0.1 meters / 10 cm.  You would therefore not need to use disparity shift.

The benefits of a D455 over the D435i are increased depth sensing range, improved depth-color alignment and a fast global shutter on the RGB sensor.  The D435i's RGB sensor has a slower 'rolling' shutter that is more vulnerable to blurring during camera motion or when observing motion from a fixed position.  If your project only requires depth sensing though then the D435i could meet your minimum distance requirements.

vikove, 2021-11-17:
Hi @MartyG-RealSense , thanks for your detailed response! I also have a query regarding the contrast at the Minimum Distance (0.1 m/ 10 cm). Can you help me with providing the datasheet of the sensors to look up the MTF parameters?

MartyG-RealSense, 2021-11-17:
If you are referring to **Modulation Transfer Function (MTF)**, I could not locate any references in relation to RealSense cameras or the specific OmniVision sensors that the cameras use.

 As a rough indicator though, I captured an RGB image of a laptop keyboard at 10 cm distance.

[image]

The contrast controls of the SDK (including in its RealSense Viewer program) can be used to alter contrast in real-time, darkening the image as contrast is increased and lightening it as it is reduced.

**MINIMUM CONTRAST**

[image]

**MAXIMUM CONTRAST**

[image]

vikove, 2021-11-17:
@MartyG-RealSense , Appreciate the prompt reply! Thank you very much for your insight. Although your explanation pertains to the visible camera, I need more information about the Near-Infrared Camera at _850 nm_. I tried looking up for the OmniVision's OV9282 camera in the internet but failed to locate it. I have written to them 5 days ago, and haven't heard from them yet. I wish you could help in this regard.

MartyG-RealSense, 2021-11-17:
The left and right infrared sensors of the 400 Series cameras can observe any visible light frequencies as the sensors do not have an IR Cut filter.   Technical information about this is provided at https://github.com/IntelRealSense/librealsense/issues/1070

The RGB sensor does have an IR Cut filter.

closed 2021-11-29 by MartyG-RealSense