realsenseai/librealsense #9978: Task '' not found in project ':librealsense'.

Papiryk, 2021-11-17:
Hello,

I tried to build the RealSense SDK for Android OS following [this](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/android/readme.md) tutorial. I got to the fifth step where you need to run :librealsense-->build-->assembleRelease:, but it's missing from gradle scripts. I added the ndk path in local.properties and also ndkVersion to build.gradle, but when I try to run assembleRelease with Execute Gradle Task (it's showing there for some reason) I get this error: 

Task '' not found in project ':librealsense'.

* Try:
Run gradle tasks to get a list of available tasks. Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

I am new to Android Studio and I would appreciate any help. I am trying to make a 3D facial recognition app with Realsense Camera for my Bachelor's thesis. Hope this makes sense. 

Thank you.
 
[image]
[image]

MartyG-RealSense, 2021-11-17:
Hi @Papiryk   Someone in a discussion linked to below had the same problem with not having the **assembleRelease** task available.  They found that syncing their projects with gradle files resolved the problem.

https://stackoverflow.com/questions/58099197/assemblerelease-not-found-in-task-gradle-in-jenkins

This action is mentioned in a guide that I recently created about step 5 of the android installation guide at https://github.com/IntelRealSense/librealsense/issues/9753#issuecomment-919903084

Papiryk, 2021-11-17:
Hi @MartyG-RealSense I synced my project with gradle files after every little change so that was not the issue and I had "do not build gradle task link..." already unticked. But I tried to do it all over again and assembleRelease was still not there, but I ran it through Execute Gradle Task and it worked!

Thank you for your help!

MartyG-RealSense, 2021-11-17:
You are very welcome.  It's great to hear that you achieved a solution.  Thanks very much for the update!

closed 2021-11-23 by MartyG-RealSense