realsenseai/librealsense #9979: L515 Python Enable Stream Configs (resolution)

ccomkhj, 2021-11-18:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 01.050.08.01 | 
| Operating System & Version |   Ubuntu 20.04  | 
| Platform                                 | PC |
| SDK Version                            |  2.50.0.3812                           | 
| Language                            |  python }                          | 
----------------------------------------------------------------------------------------------------

### Issue Description

When I type, config.enable_stream(rs.stream.color, rs.format.bgr8, 30) it is working, but it only support QVGA (320x240).
When I type, config.enable_stream(stream_type= rs.stream.depth, width = 640, height = 480 , format=rs.format.z16, framerate = 30), 

it generates error 

pipeline_profile = config.resolve(pipeline_wrapper)
RuntimeError: No device connected

However, I need higher resolution such as VGA, XGA.

It works fine with Intel RealSense Viewer v2.49.0

MartyG-RealSense, 2021-11-18:
Hi @ccomkhj   Could you unplug the micro-sized end of the USB cable from the base of the camera, reverse the connector's orientation and plug it back into the camera, please (USB-C cables are two-way insertion at the micro-sized end).  Does doing so make any difference?

ccomkhj, 2021-11-18:
Hi @MartyG-RealSense 
I have inserted USB-C micro cable at both side, and there is other error message.

Side 1:
pipeline.start(config)
RuntimeError: Couldn't resolve requests

Side 2:
pipeline_profile = config.resolve(pipeline_wrapper)
RuntimeError: No device connected

MartyG-RealSense, 2021-11-18:
Please try replacing your two config.enable_stream instructions with the ones below.

```
config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 1280, 720, rs.format.bgr8, 30)
```

ccomkhj, 2021-11-19:
I have changed the same lines, but it gives same error.

Side 1:
pipeline.start(config)
RuntimeError: Couldn't resolve requests

Side 2:
pipeline_profile = config.resolve(pipeline_wrapper)
RuntimeError: No device connected

=============================================================================================
Here is the modified example code.

class AppState:

    def __init__(self, *args, **kwargs):
        self.WIN_NAME = 'RealSense'
        self.pitch, self.yaw = math.radians(-10), math.radians(-15)
        self.translation = np.array([0, 0, -1], dtype=np.float32)
        self.distance = 2
        self.prev_mouse = 0, 0
        self.mouse_btns = [False, False, False]
        self.paused = False
        self.decimate = 1
        self.scale = True
        self.color = True

    def reset(self):
        self.pitch, self.yaw, self.distance = 0, 0, 2
        self.translation[:] = 0, 0, -1
                      
    @property
    def rotation(self):
        Rx, _ = cv2.Rodrigues((self.pitch, 0, 0))
        Ry, _ = cv2.Rodrigues((0, self.yaw, 0))
        return np.dot(Ry, Rx).astype(np.float32)

    @property
    def pivot(self):
        return self.translation + np.array((0, 0, self.distance), dtype=np.float32)

state = AppState()

pipeline = rs.pipeline()
config = rs.config()

pipeline_wrapper = rs.pipeline_wrapper(pipeline)
pipeline_profile = config.resolve(pipeline_wrapper)
device = pipeline_profile.get_device()

found_rgb = False
for s in device.sensors:
    if s.get_info(rs.camera_info.name) == 'RGB Camera':
        found_rgb = True
        break
if not found_rgb:
    print("The demo requires Depth camera with Color sensor")
    exit(0)

config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 1280, 720, rs.format.bgr8, 30)

pipeline.start(config)

MartyG-RealSense, 2021-11-19:
At the begnning of this case you said: "When I type, config.enable_stream(rs.stream.color, rs.format.bgr8, 30) it is working, but it only support QVGA (320x240)".

The L515 only supports a minimum of 640x360 color (it applies 1280x720 color by default if there is not a custom stream configuration).  So I was not sure how it could be working with a color resolution of 320x240 (**depth** resolution of 320x240 is supported).

ccomkhj, 2021-11-22:
Firstly, my first question was wrong due to my lack of understanding.

I checked **depth_intrinsics.width** and **depth_intrinsics.height**. and I found that it's maximum resolution is upto 320x240.
I can't still assign higher color resolution except for 640x480.

When I print, **profile.get_stream(rs.stream.color)**
<pyrealsense2.[video_]stream_profile: Color(0) 640x480 @ 30fps BGR8>

If I assign **config.enable_stream(rs.stream.color, 1280, 720, rs.format.bgr8, 30)**,
pipeline.start(config)
*** RuntimeError: Couldn't resolve requests

Is there available combination of resolution between depth and color?

MartyG-RealSense, 2021-11-22:
Your description suggests to me that your camera is being identified as being on a USB 2.1 connection instead of USB 3.2.  On USB 2.1, the maximum L515 resolutions available are 320x240 depth and 640x480 color, as shown in the image below.

[image]

The error **RunTimeError: Couldn't resolve requests** means that a resolution and FPS combination was requested that was not currently supported by the camera, perhaps because of the USB mode that it was currently in, and so that resolution / FPS could not be provided.

On a USB 3.2 connection, the following depth and color resolutions should be available.

**Depth**
320x240 
640x480
1024x768

**Color**
640x360
640x480
960x540
1280x720

Can you confirm please that your L515 camera is plugged into a USB 3 port.  If it is, are you using the official 1 meter USB cable supplied with the camera or a longer cable of your own choice (which may unknowingly be a USB 2 cable that puts the L515 in USB 2 mode).

ccomkhj, 2021-11-23:
Thank you for your support!! 
The usb cable was the reason.
Now it works well.

closed 2021-11-23 by ccomkhj

MartyG-RealSense, 2021-11-23:
That's excellent news, @ccomkhj - thanks for the update!   :)