realsenseai/librealsense #9981: Intel RealSense Viewer crash,how to fix?

MichaelMingCao, 2021-11-19:
Windows 11 64bit

| | |
|---|---|
|**librealsense**|2.49.0 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
Program terminated due to an unrecoverable SEH exception:
Access Violation!
```

MartyG-RealSense, 2021-11-19:
Hi @MichaelMingCao   Whilst you are the first RealSense user that I know of to be using the camera with Windows 11, it is not necessarily the cause of this error as it occurs in the RealSense Viewer tool on Windows 10 too.

Could you first try downloading and running version **2.50.0** of the RealSense Viewer by going to the 'Assets' file list for SDK 2.50.0 and clicking on the **Intel.RealSense.Viewer.exe** link.  Once the file is downloaded, launch it from the folder location that you downloaded it to and not with the launch shortcut icon on the Windows desktop.  Does the error still occur with this version?

https://github.com/IntelRealSense/librealsense/releases/tag/v2.50.0

closed 2021-12-02 by MartyG-RealSense

MartyG-RealSense, 2021-12-24:
Hi @MichaelMingCao  After highlighting your report to Intel, they performed Windows 11 testing with a range of example programs such as rs-capture, rs-hello and rs-callback but were unable to replicate the problem that you experienced.