realsenseai/librealsense #9982: No device detected, Librealsense Docker

tiandeyu, 2021-11-19:
realsense camera works fine in host os, not detected in docker container.
Host os: Ubuntu: 18.04
Device: D435i
commond : docker run -it --rm \
    -v /dev:/dev \
    --device-cgroup-rule "c 81:* rmw" \
    --device-cgroup-rule "c 189:* rmw" \
    librealsense/librealsense

MartyG-RealSense, 2021-11-19:
Hi @tiandeyu  The procedure that you list is the same as the one provided in Intel's new Docker tutorial in the link below, released with SDK 2.50.0.  Are these the instructions that you have been using, please?

https://github.com/IntelRealSense/librealsense/tree/master/scripts/Docker

If you are using this guide, are you using the guide's pre-built Dockerfile that is retrieved with the command **docker pull librealsense/librealsense**

tiandeyu, 2021-11-19:
yes, I'm following this guide.

MartyG-RealSense, 2021-11-19:
Bearing in mind that the description of the Dockerfile that is retrieved with the **docker pull** command defines Ubuntu 20.02 as its base system and you are using Ubuntu 18.04 ... I wonder whether it may be better to build the Dockerfile from script yourself using the guidance in the link below, setting the **BASE_IMAGE** parameter in line 1 of the script to 18.04 instead of 20.04 before running the image building script.

https://github.com/IntelRealSense/librealsense/tree/master/scripts/Docker#building-librealsense-docker-image

MartyG-RealSense, 2021-11-26:
Hi @tiandeyu  Do you require further assistance with this case, please?  Thanks!

tiandeyu, 2021-11-26:
Thanks for your help, I build the image success using this guide(https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md).

closed 2021-11-26 by tiandeyu

MartyG-RealSense, 2021-11-26:
Great to hear that you succeeded, @tiandeyu - thanks very much for the update!