realsenseai/librealsense #9983: error when ran 'npm install' under the nodejs/examples

caixiaoniweimar, 2021-11-20:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Ubuntu 20.04 | 
| Kernel Version (Linux Only)    |  (5.4.0-54-generic)                                         | 
| Platform                                 | PC|
| SDK Version                            |  2.50                          | 
| Language                            |  nodejs                          | 
| Nodejs                            |  v10.22.0                          | 
| Segment			|  Robot                      | 

### Issue Description
hello, I followed by the official documentation for installation, however, I got the error when I ran 'npm install' under the wrappers/nodejs/examples folder. I have been struggling for several days but couldn't figure out the problems. Could you please give me some hints? Thanks in advance!

`CMake Error at CMakeLists.txt:228 (message):
  The Xf86VidMode library and headers were not found

-- Configuring incomplete, errors occurred!
See also "/home/librealsense-2.50.0/wrappers/nodejs/examples/node_modules/node-12-glfw-3/deps/glfw-3.0.4/CMakeFiles/CMakeOutput.log".
See also "/home/librealsense-2.50.0/wrappers/nodejs/examples/node_modules/node-12-glfw-3/deps/glfw-3.0.4/CMakeFiles/CMakeError.log".
make: *** [native_glfw_build.target.mk:13: /home/librealsense-2.50.0/wrappers/nodejs/examples/node_modules/node-12-glfw-3/deps/glfw-3.0.4/src/libglfw.so] Error 1
make: Leaving directory '/home/librealsense-2.50.0/wrappers/nodejs/examples/node_modules/node-12-glfw-3/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:191:23)
gyp ERR! stack     at ChildProcess.emit (events.js:198:13)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:248:12)
gyp ERR! System Linux 5.4.0-54-generic
gyp ERR! command "/usr/local/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/librealsense-2.50.0/wrappers/nodejs/examples/node_modules/node-12-glfw-3
gyp ERR! node -v v10.22.0
gyp ERR! node-gyp -v v5.1.0
gyp ERR! not ok 
npm WARN node-librealsense2-examples@0.0.1 No description
npm WARN node-librealsense2-examples@0.0.1 No repository field.

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! node-12-glfw-3@git+https://github.com/whsol/node-glfw.git#7b28c1d63d1cb60a1557e51a9868a30c281c916a install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the node-12-glfw-3@git+https://github.com/whsol/node-glfw.git#7b28c1d63d1cb60a1557e51a9868a30c281c916a install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/.npm/_logs/2021-11-20T05_53_24_905Z-debug.log
`
[2021-11-20T05_53_24_905Z-debug.log](https://github.com/IntelRealSense/librealsense/files/7574623/2021-11-20T05_53_24_905Z-debug.log)

MartyG-RealSense, 2021-11-20:
Hi @caixiaoniweimar   Have you tried running the npm install command from the **wrappers/nodejs** directory (the parent folder of the Node.js wrapper) instead of the wrappers/nodejs/examples sub-directory, please?

caixiaoniweimar, 2021-11-20:
Hi Marty, Thanks for your quick reply.
Yes, after running
```
make 
cmake ../
cmake ../ -DBUILD_EXAMPLES=true
sudo make uninstall && make clean && make && sudo make install
```
under the librealsense/build folder, 

I have tried firstly 'npm install' under the librealsense/wrappers/nodejs folder and it works. Then ran 'npm install' under the librealsense/wrappers/nodejs/examples folder. And it showed the error.

MartyG-RealSense, 2021-11-20:
Thanks very much.  So I understand that you have completed wrapper installation but have problems when trying to install the examples using the **2.  Run Examples** stage of the instructions.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/nodejs#2-run-examples

May I confirm if you installed the wrapper from source code using the instructions in the **Manually Build** section, since you mentioned using 'npm build' to build the wrapper and not the 'npm install --save node-librealsense' method at the top of the instructions page?

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/nodejs#manually-build

If you did build the wrapper from source code using the Manually Build section then the instructions state that 

My understanding is that section 1.3 of the instructions - that offers the choice between building the wrapper with npm or CMake - is a choice between one or the other.  So if you used npm install in the wrappers/nodejs directory then you should **not** then perform the Build With CMake instructions in the next sectin 'Build With CMake'.  Can you confirm please that you did not perform the Build With CMake instructions to build the wrapper after already having done so with 'npm install' please.

caixiaoniweimar, 2021-11-21:
Hi, @MartyG-RealSense,
I installed librealsense not from the git repository but from the source code, the way RSUSB you have metioned #[9872](https://github.com/IntelRealSense/librealsense/issues/9872#issuecomment-948727294). At the beginning, I used the kernel version 5.11 and it failed, so I found the suggestions from previous issues to install librealsense in the way of RSUSB, without dependency on Linux version, kernel version or patches. 

By the way, I have also tried to downgrade my kernel version to 5.4(the current one I am using) and installed librealsense from the git repository, however,  after running `./scripts/patch-realsense-ubuntu-lts.sh`, I found it took a few days and got time out error (I opened vpn in China). And that's why in the end I still used RSUSB to download librealsense.

One thing to confirm, after running make and Cmake from librealsense/build, I was able to run the ./sc_capture from librealsense/build/examples/capture.

For the Node.js, I ran `npm install --save node-librealsense` from the root folder (not from the librealsense folder), and it successed, according to the section [Notice: Before you go further](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/nodejs#notice-before-you-go-further). Then I ran the command mentioned in the section [manually-build](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/nodejs#manually-build). Did I do something wrong here? Since I have run 'npm install node-librealsense', do I still need to do the manually build? 

And once I have done the manually build, yes, I didn't do the 'Build with Cmake'.

caixiaoniweimar, 2021-11-21:
Hi @MartyG-RealSense, I redownloaded the librealsense repository, and tried to do only the 'Build with Cmake'.
`~/librealsense-2.50.0/wrappers/nodejs$ cmake -DBUILD_NODEJS_BINDINGS=1
CMake Warning:
  No source or binary directory provided.  Both will be assumed to be the
  same as the current working directory, but note that this warning will
  become a fatal error in future CMake releases.
-- The C compiler identification is GNU 9.3.0
-- The CXX compiler identification is GNU 9.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring done
-- Generating done
CMake Warning:
  Manually-specified variables were not used by the project:
    BUILD_NODEJS_BINDINGS
-- Build files have been written to: /home/librealsense-2.50.0/wrappers/nodejs
`
and 
`~/librealsense-2.50.0/wrappers/nodejs$ make
Scanning dependencies of target BUILD_NODE_ADDON
make[2]: *** No rule to make target 'realsense2', needed by 'build/Release/node_librealsense.node'.  Stop.
make[1]: *** [CMakeFiles/Makefile2:104: CMakeFiles/BUILD_NODE_ADDON.dir/all] Error 2
make: *** [Makefile:84: all] Error 2`

MartyG-RealSense, 2021-11-21:
If you have used the **npm install --save node-librealsense** method from the **Notice: Before You Go Further** section in the 'wrappers/nodejs' parent folder of the wrapper then you do not need to manually build afterwards using the methods in the **Build From Source** section.  

All of the Node.js wrapper building cases I have handled that had a successful outcome have used the 'Notice: Before You Go Further' method instead of building from source code.

caixiaoniweimar, 2021-11-21:
Hi @MartyG-RealSense Thanks for your hints and sharing your experience. I downloaded the librealsense from the git repository. and then ran  `npm install --save node-librealsense`, and yes, it worked. However, then I would like to try the examples, so I went to 

```
~/librealsense$ cd wrappers/nodejs/examples/
~/librealsense/wrappers/nodejs/examples$ npm install
```
it showed me the same build error as before. Did I do somewhere wrongly?

Once I installed node-librealsense successfully, does it mean I can create librealsense object in my other nodejs project even if I am not able to run the examples?

MartyG-RealSense, 2021-11-21:
I checked back through the nodejs cases I have handled and in almost all of them, problems with running the nodejs examples or the nodejs version of the RealSense Viewer were experienced.  So it is not only you that has had problems, suggesting that your nodejs wrapper build is indeed okay.

It may be worth trying a newer version of nodejs than v10.22.0.  The most recent update was support for nodejs 14 in SDK 2.44.0.

caixiaoniweimar, 2021-11-21:
Hi @MartyG-RealSense Thanks for your info and suggestions. We upgrade our node version to v14.18.1.
And my team member Philip tried 
`sudo apt-get install xorg-dev libxcb1-dev libglfw3 libglfw3-dev`
and solved the problem. So now we are able to run the nodejs examples via 'npm install'. Benefits for others, if they have the same issues. Maybe they could try. 

Thanks for your help again!

closed 2021-11-21 by caixiaoniweimar

MartyG-RealSense, 2021-11-21:
Great news, @caixiaoniweimar - thanks very much for sharing the method that worked for you with the RealSense community  :)