realsenseai/librealsense #9984: No device connected

kongxin22, 2021-11-21:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D455} | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Linux (Ubuntu 16) } | 
| Kernel Version (Linux Only)    |  (5.4.0)                                         | 
| Platform                                 | PC |
| SDK Version                            |  {librealsense2.49.0 }                          | 
| Language                            |  {C/C# }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
Hello!When I try to collect the data,there is a fatal error:
RealSense error calling rs2_pipeline_start_with_config(pipe:0x560b767d4380, config:0x560b769168a0):
    No device connected
What should I do?Thanks in advance!

MartyG-RealSense, 2021-11-21:
Hi @kongxin22  Could you provide more information please about the script or program that you are using that produces this error?  If it is a script that you have created, please paste it into a comment if possible.  Thanks!

kongxin22, 2021-11-21:
@MartyG-RealSense I find that if I pull out the camera and connect it again,the problem will not occur.Is it a big problem?

MartyG-RealSense, 2021-11-21:
Unplugging and re-inserting the camera resets it.  It is fine to resolve the problem like this, though it can become understandably annoying if it has to be done often. 

The camera can also be reset without unplugging in the RealSense Viewer program by going to the 'More' option near the top of the options side-panel and selecting the 'Hardware Reset' menu option.  

[image]

If you are writing your own script then you can also reset the camera when the script launches by placing a **hardware_reset()** instruction in the script.

MartyG-RealSense, 2021-12-04:
Case closed due to no further comments received.

closed 2021-12-04 by MartyG-RealSense

kongxin22, 2021-12-12:
Hello.Long  time no see.The problem was shelved by me.Today when I ran my script,which called pipe to start,the problem was still here:
RealSense error calling rs2_pipeline_start_with_config(pipe:0x556234ce8e80, config:0x556234ca8280):
    No device connected
This time I used a new camera.I thought maybe my script  has errors.But sometimes it did run after reinserting the camera today.And I tried it on many computers with no luck.I am confused.What is the problem?

MartyG-RealSense, 2021-12-12:
Hi @kongxin22  Are the multiple computers that you used all the same manufacturer / model?