realsenseai/librealsense #9985: Resource temporarily unavailable, number: 11

kongxin22, 2021-11-21:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D455 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Linux (Ubuntu 16)  | 
| Kernel Version (Linux Only)    |  (e.g. 5.4.0)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.<49>.<0> }                          | 
| Language                            |  {C/C# }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
I'm so sorry that a new problem occurs.At first everything went well,but suddenly something happened like these:
 21/11 16:43:00,257 WARNING [140271777982208] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 21/11 16:43:01,343 ERROR [140271650064128] (ds5-options.cpp:88) Asic Temperature value is not valid!
 21/11 16:29:51,418 WARNING [140271777982208] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: No data available, number: 3d

Is my camera broken?I'm so worried!
PS:I have not installed the librealsense-ros.

MartyG-RealSense, 2021-11-21:
Hi @kongxin22  There is no need to apologize  :)

**control_transfer returned** errors are usually not caused by a camera hardware failure but instead by a communication problem such as camera data temporararily not being received by the computer'sUSB port.  An 'Asic Temperature is not valid!' warning can similarly be caused by an interruption in receival of data.

If the messages are only occurring occasionally (for example, once or twice a minute) and are not disrupting the correct operation of the camera then they can be safely ignored.

kongxin22, 2021-11-22:
@MartyG-RealSense But the statements on my terminal are only "Resource temporarily unavailable, number: 11",which appear one by one.And there are other problems on another computer:

22/11 22:51:26,811 WARNING [140535454770944] (backend-v4l2.cpp:1358) Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!
 22/11 22:51:31,730 WARNING [140535035463424] (hdr-merge.cpp:38) HDR Merge filter cannot process frames because relevant metadata params are missing
 22/11 22:51:33,982 ERROR [140535079630592] (http-downloader.cpp:143) Download error from URL: https://librealsense.intel.com/Releases/rs_versions_db.json, error info: Timeout was reached
 22/11 22:53:31,783 WARNING [140535729716672] (ds5-motion.cpp:473) IMU Calibration is not available, default intrinsic and extrinsic will be used.
 22/11 22:53:31,787 WARNING [140534081386240] (backend-hid.cpp:715) HID set_power 1 failed for /sys/devices/pci0000:00/0000:00:14.0/usb4/4-2/4-2:1.5/0003:8086:0B5C.0002/HID-SENSOR-200073.2.auto/iio:device0/buffer/enable

I have tried to reinstall the SDK,but there was no luck.I am worried because the troubles occurred so suddenly.
Please forgive me for my broken English!

MartyG-RealSense, 2021-11-22:
Your English is very good, so don't worry!

The warning message that you experienced is of the 'control_transfer returned' type.  'Resource temporarily unavailable' describes the kind of control_transfer returned warning that it is.

WARNING [140271777982208] (messenger-libusb.cpp:42) **control_transfer returned error,** index: 768, error: Resource temporarily unavailable, number: 11

The 'IMU Calibration is not available' warning suggests that the D455's internal IMU hardware component (called the Motion Module in the RealSense Viewer) has not been calibrated using Intel's IMU calibration tool since the camera was purchased.  This is not a serious problem though.

https://dev.intelrealsense.com/docs/imu-calibration-tool-for-intel-realsense-depth-camera

The warning 'Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!' suggests that if you built the SDK using Linux and you built it from source code then the Linux kernel was not patched during the installation process.  I note though that the SDK was working normally for you at first and then suddenly having problems.  This suggests that the SDK was installed correctly.

What are your problems on the other computer, please?

MartyG-RealSense, 2021-11-23:
If the camera streams appear to be operating normally then you could ignore the control_transfer returned messages.  If you wish to try to reduce them though then you could test whether the number of messages reduce if you use a lower resolution.  For example, setting 640x480 and 30 FPS for the depth stream in the RealSense Viewer tool.

[image]

kongxin22, 2021-11-23:
Yes,it does work normally.Only the statements on the terminal worry me:
23/11 22:52:54,990 WARNING [139858452395776] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 23/11 22:53:03,046 WARNING [139858452395776] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 23/11 22:53:13,102 WARNING [139858452395776] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 23/11 22:53:15,154 WARNING [139858452395776] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 23/11 22:53:16,205 WARNING [139858452395776] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
And sometimes:
Resource temporarily unavailable, number: 11
But I found that the problem occurs only on one computer no matter which the USB port is.Does it mean I can ignore the problem?

MartyG-RealSense, 2021-11-23:
Yes, if it is not causing you problems then you can ignore the warnings without worry.  They may be related to the USB hardware on that particular computer.

MartyG-RealSense, 2021-11-29:
Hi @kongxin22  Do you require further assistance with this case, please?  Thanks!

wilsonz91, 2021-11-29:
Hi, I'm having a similar problem on a Lenovo T480s and a VR capable NUC (although I am running L515). Tried both v2.48 and v2.5 also and still getting the same problem. The messages appear constantly, both on realsense-viewer and ROS. 
The error pops up when the depth sensor stream is enabled; for rgb only it works fine. 
Tried switching the original USB C cable around at the L515 side but still getting the error. 
Set up: Ubuntu 18 + ROS melodic
(Not sure if I should open a new thread, thought i'll try here first)

MartyG-RealSense, 2021-11-29:
Hi @wilsonz91  In ROS, do the messages still occur if you add **initial_reset:=true** to your roslaunch instruction to reset the camera at launch.  For example:

**roslaunch realsense2_camera rs_camera.launch initial_reset:=true**

wilsonz91, 2021-11-29:
Hi @MartyG-RealSense, sadly, yes it does

MartyG-RealSense, 2021-11-29:
@wilsonz91  Are you able to stream the camera images in the Viewer and in RViz in ROS without apparent negative consequences to performance?  If so then I would repeat the advice given to @kongxin22 above to simply ignore the messages.

wilsonz91, 2021-11-29:
I'll have to check on that. 

Could I ask regarding the ROS timestamp seen on the image/depth messages: are they from the camera time or the time they are published? 
So for eg if the camera image is published at fixed rate, and images are lost due to say USB hardware fault, would the timestamp show up as follows: 
No. | Timestamp 
#1  | 0001
#2  | 0002
#3  | 0003 (this image is lost)
#4  | 0004 (this image received would have this time stamp)

or would image #4 shows up timestamp 0003?

MartyG-RealSense, 2021-11-29:
In the RealSense ROS wrapper, **Global Time** is enabled by default for 400 Series cameras (including the D455).  **Doronhi** the RealSense ROS wrapper developer provides advice about Global Time in regard to timestamps and how it aligns device time with system time at https://github.com/IntelRealSense/realsense-ros/issues/796

Global Time is discussed further in detail at https://github.com/IntelRealSense/librealsense/issues/4505

In regard to your specific question about timestamp numbering, I don't know unfortunately.

kongxin22, 2021-11-30:
Hi @MartyG-RealSense ,I have taken your advice to ignore the "Resource temporarily unavailable, number: 11" and "No data available, number: 61".But I find that when I run a script,which need to collect data from the camera,similar problem occurred:No device connected.
Sometimes it can be solved by  re-inserting the camera.However,mots of the time the script can't run ,no matter on which computer.Does it mean my camera broken?

MartyG-RealSense, 2021-11-30:
Hi @kongxin22  If the problem can be solved sometimes by USB unplug-replug then it suggests that the camera hardware is okay and the problem may be elsewhere, such as in the USB system (either the USB port of the computer, the port's USB controller or the USB cable).  

Are you using the official 1 meter USB cable supplied with the camera or a longer USB cable of your own choice, please?

MartyG-RealSense, 2021-12-06:
Hi @kongxin22 and @wilsonz91  Do you require further assistance with this case, please?  Thanks!

wilsonz91, 2021-12-07:
Hi @MartyG-RealSense, thanks for the information. I'm ignoring the warnings for now.

MartyG-RealSense, 2021-12-07:
Thanks very much @wilsonz91 for the update of your particular situation!

closed 2021-12-13 by MartyG-RealSense