realsenseai/librealsense #9986: Calibration error

Nishanthiruthayaraj, 2021-11-21:
I am getting the following error when I run on-chip calibration. 
"invalid value in rs2_run_on_chip_calibration........"
Are there any other possibilities to do calibration of Intel Realsense D435i by using python codes?

| | |
|---|---|
|**librealsense**|2.49.0 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense D435I|
|**Serial Number**|038322070705|
|**Firmware Version**|05.12.15.50|
|**Advanced Mode**|YES|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|2.1|
|**Product Line**|D400|
|**Asic Serial Number**|039223050798|
|**Firmware Update Id**|039223050798|

Please provide a description of the problem

MartyG-RealSense, 2021-11-21:
Hi @Nishanthiruthayaraj   The RealSense SDK has a Python example program for on-chip calibration called **depth_auto_calibration_example.py** at the link below.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/depth_auto_calibration_example.py

If you have been working in C++ up until now, https://github.com/IntelRealSense/librealsense/issues/7087 provides an On-Chip calibration script shared by a RealSense user to compare your own code to.

On-chip calibration can additionally be performed in the **RealSense Viewer** tool by going to the 'More' option near the top of the Viewer's options side-panel and selecting **On-Chip Calibration** from its drop-down menu.

whatisor, 2021-11-23:
Note that you have to turn off all streams before run on chip - calibration, otherwise it informs error.

Nishanthiruthayaraj, 2021-11-23:
> Note that you have to turn off all streams before run on chip - calibration, otherwise it informs er

> Hi @Nishanthiruthayaraj The RealSense SDK has a Python example program for on-chip calibration called **depth_auto_calibration_example.py** at the link below. [...]

Thank you very much for the on-time reply. Error resolved

Nishanthiruthayaraj, 2021-11-23:
> Note that you have to turn off all streams before run on chip - calibration, otherwise it informs error.

Thanks for the hints. My problem is already resolved. Will try your idea if I get the same error.

MartyG-RealSense, 2021-11-23:
Excellent news, @Nishanthiruthayaraj that you achieved a solution - thanks for the update!   :)

closed 2021-11-29 by MartyG-RealSense