realsenseai/librealsense #9988: about intrinsics variable

HyeYumKim, 2021-11-22:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D455  | 
| Language                            | python | 

I am trying to measure box volume with python box_dimentioner_multicam code using 2 d455 cameras. I am trying to measure the volume of the box more accurately by modifying the intrinsics variable of the d455 camera.
Currently, I am editing the json file. Is there any other json file or code to refer to?

MartyG-RealSense, 2021-11-22:
Hi @HyeYumKim  Is this question in relation to your earlier question at https://github.com/IntelRealSense/librealsense/issues/9951 please, and you are trying to define fixed values for the intrinsics instead of retrieving them from the cameras in real-time?

HyeYumKim, 2021-11-22:
Okay, I'll go ahead and ask a question. thank you.

MartyG-RealSense, 2021-11-28:
Case closed due to question being asked on https://github.com/IntelRealSense/librealsense/issues/9951 instead.

closed 2021-11-28 by MartyG-RealSense