realsenseai/librealsense #9990: Manual align by factory intrinsic and extrinsic is not accurate upon hardware

whatisor, 2021-11-23:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D450} | 
| Firmware Version                   |  (0.5.12.15.50) |
| Operating System & Version |  Ubuntu 20.04  | 
| Kernel Version (Linux Only)    |  (5.11.0-37-generic)                                     | 
| Platform                                 | PC|
| SDK Version                            | master                          | 
| Language                            | C++ | 
| Segment			|   | 

### Issue Description
I am trying to align manually depth and rgb to create pointcloud using factory intrinsic and extrinsic.
**However, alignment only works accurately with 1 sensor, but not  other ( I have 2 sensors).**
**Notice that realsense-viewer and rs2:align api works well for both sensor.**
Is there something hidden inside API  I need to know except for intrinsic, extrinsic and color coeffs?

[image]

whatisor, 2021-11-23:
Also, trying on-chip calibration did not help.

whatisor, 2021-11-23:
Here are values for 2 sensors.(one is bad, other is good)

[0RT.txt](https://github.com/IntelRealSense/librealsense/files/7585818/0RT.txt)
[1RT.txt](https://github.com/IntelRealSense/librealsense/files/7585819/1RT.txt)

MartyG-RealSense, 2021-11-23:
Hi @whatisor  Could you provide more information please about your project.  I understand that you are attempting to align depth to color and generate a textured point cloud.  

If you simpy need to generate a point cloud and do not need to obtain the coordinate values of the depth points then you would not necessarily need to use alignment to do so.  The RealSense SDK's C++ example program **rs-pointcloud** demonstrates how to generate a textured point cloud with the instruction **pc.map_to**.

https://github.com/IntelRealSense/librealsense/tree/master/examples/pointcloud

whatisor, 2021-11-23:
@MartyG-RealSense  I am using stream from browser, so no api is ready.
Do you think data(depth) from browser( chrome) is same with realsense sdk?

MartyG-RealSense, 2021-11-23:
Are you using the HTML5 browser example in the link below or a different method?

https://github.com/intel/depth-camera-web-demo

If you are then you may need to adjust the program for the D450's configuration, as the D450 was not released when that browser tool was created.  The project's Development page explains how to add support for new RealSense camera models and depth and color resolutions not in the demo's default configuration.

https://github.com/intel/depth-camera-web-demo/blob/master/DEVELOPMENT.md

The browser tool is apparently taking raw depth and color data from the SDK and converting it into a texture using **WebGL**.  This is similar to the approach used by the RealSense wrapper for the Unity game engine, where the camera data is converted into a 'material' to render it visually on-screen.  So I have no reason to think that the data that the browser tool is using is especially inaccurate, though it likely is indeed a 'raw' image that is not having refinements such as post-processing filters applied to it.

whatisor, 2021-11-23:
not really,
I refer this source:
https://github.com/intel/depthcamera-pointcloud-web-demo
And using factory matrix from sdk.

MartyG-RealSense, 2021-11-23:
If you are defining the intrinsics and extrinsics manually, are you using the same set of values for both cameras?  If you are then I recommend obtaining the intrinsics and extrinsics for each individual camera, as no one camera should have identical intrinsic / extrinsic values due to differences introduced during manufacturing in the factory.

whatisor, 2021-11-23:
Yes, I think I still have some own issue. Let me close this.

closed 2021-11-23 by whatisor