realsenseai/librealsense #9991: NodeJS Wrapper error native function rs2_pipeline_start

phixxx5, 2021-11-23:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D415 } | 
| Firmware Version                   | (05.13.00.50) | 
| Operating System & Version |   {Linux (Ubuntu 20.04.3 LTS)  | 
| Kernel Version (Linux Only)    |  (5.4.0-90-generic)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { 2.50 }                          | 
| Language                            | {nodejs }                          | 
| Segment			|  {Robot }                          | 

### Issue Description
I built the RealSense library with the NodeJS wrapper. NodeJS is currently on version v10.24.1. I'm able to run the NodeJS examples like nodejs-capture.js in librealsense/wrappers/nodejs/examples. RealSense-Viewer works perfectly as well. However when I try to use the library in my node-red project (Code below), I receive the following error:

    [realsense-node:dc353c6914411109] Error: Unrecoverable! error native function rs2_pipeline_start: set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Inappropriate ioctl for device

My code:

    // initialize the features
    const rs2 = require('node-librealsense');
    const pipeline = new rs2.Pipeline(); //main work pipeline of realsense camera
    pipeline.start(); //start camera

Any suggestions what could be wrong?

MartyG-RealSense, 2021-11-23:
Hi @phixxx5  Do you experience the same error if you try the full starter script at the link below, please?

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/nodejs/scripts/npm_dist#nodejs-api-for-intel-realsense-depth-cameras

phixxx5, 2021-11-23:
With the starter script I do not experience the error.

MartyG-RealSense, 2021-11-23:
The lines of your node-red code are the same as the initial lines of the starter script (aside from the absent **const colorizer = new rs2.Colorizer();** line), which suggests to me that one of the missing lines may be triggering your error.

A way to test this theory would be to remove lines from the starter script and run it until it replicates the error in node-red.  Removing the final line **rs2.cleanup();** would be a good place to begin.

MartyG-RealSense, 2021-11-30:
Hi @phixxx5  Do you require further assistance with this case, please?  Thanks!

phixxx5, 2021-11-30:
Hi,
I was able to solve the problem. The librealsense wrapper that I built from source was not imported but instead the one from here was imported (https://www.npmjs.com/package/node-librealsense). I copied the folder with the nodejs wrapper to my node-red project and then imported it through `const rs2 = require('./lib/node-librealsense/index.js');`
Thanks for your help :)

MartyG-RealSense, 2021-11-30:
You're very wecome, @phixxx5  - thanks for sharing your solution with the RealSense community!

closed 2021-12-06 by MartyG-RealSense