realsenseai/librealsense #9993: Program terminated due to an unrecoverable SEH exception: Access Violation!

EzEsh44, 2021-11-23:
| | |
|---|---|
|**librealsense**|2.49.0 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
Program terminated due to an unrecoverable SEH exception:
Access Violation!
```

MartyG-RealSense, 2021-11-23:
Hi @EzEsh44   I hope that the advice in https://github.com/IntelRealSense/librealsense/issues/9885 will be helpful in resolving your problem.

MartyG-RealSense, 2021-11-30:
Hi @EzEsh44  Do you require further assistance with this case, please?  Thanks!

EzEsh44, 2021-12-01:
Yes, I went through #9885 and am still getting the same violation. I can access the camera through my windows camera app as well as python scripts

MartyG-RealSense, 2021-12-01:
In a couple of other cases of this error in the past week, upgrading to the latest SDK version **2.50.0** resolved the problem.

closed 2021-12-13 by MartyG-RealSense