realsenseai/librealsense #9994: install SDK on Jetson TX2 Ubuntu18.04

karry-yuan, 2021-11-24:
Is it right to add repository with "**sudo add-apt-repository "deb https://librealsense.intel.com/Debian/apt-repo $(lsb_release -cs) main" -u**"?  After I add this repository, it always says that"the repository does not have a release file". So, I can't install SDK on Jetson TX2 Ubuntu18.04. Thank you very much!

MartyG-RealSense, 2021-11-24:
Hi @karry-yuan  Jetson boards have their own dedicated installation instructions page at the link below.

https://dev.intelrealsense.com/docs/nvidia-jetson-tx2-installation

That page states a 2-step process for package installation on Jetson. 

1.  Register the server's public key.

`sudo apt-key adv --keyserver keys.gnupg.net --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE || sudo apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE`

2.  Add the server to the list of repositories using the instruction below for Ubuntu 18.04.

`sudo add-apt-repository "deb https://librealsense.intel.com/Debian/apt-repo bionic main" -u`

karry-yuan, 2021-11-24:
I did it with your way. But there are still errors.
` 
kumar@nv-tx2:~$ sudo add-apt-repository "deb https://librealsense.intel.com/Debian/apt-repo bionic main" -u
Get:1 http://mirrors.ustc.edu.cn/ros/ubuntu bionic InRelease [4,680 B]
Hit:2 http://mirrors.tuna.tsinghua.edu.cn/ubuntu-ports bionic InRelease                                        
Hit:3 http://mirrors.tuna.tsinghua.edu.cn/ubuntu-ports bionic-updates InRelease                        
Hit:4 http://mirrors.tuna.tsinghua.edu.cn/ubuntu-ports bionic-backports InRelease                      
Err:1 http://mirrors.ustc.edu.cn/ros/ubuntu bionic InRelease                     
  The following signatures were invalid: EXPKEYSIG F42ED6FBAB17C654 Open Robotics <info@osrfoundation.org>
Get:5 https://librealsense.intel.com/Debian/apt-repo bionic InRelease [3,251 B]  
Hit:6 http://mirrors.tuna.tsinghua.edu.cn/ubuntu-ports bionic-security InRelease
Get:7 https://librealsense.intel.com/Debian/apt-repo bionic/main arm64 Packages [20.9 kB]
Reading package lists... Done       
W: GPG error: http://mirrors.ustc.edu.cn/ros/ubuntu bionic InRelease: The following signatures were invalid: EXPKEYSIG F42ED6FBAB17C654 Open Robotics <info@osrfoundation.org>
E: The repository 'http://mirrors.ustc.edu.cn/ros/ubuntu bionic InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.

 `

MartyG-RealSense, 2021-11-24:
It looks as though the GitHub version of the Jetson instructions has been updated more recently than the dev.intelrealsense.com version of the instructions.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md

On that page they recommend using this keyserver instruction first:

`sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE || sudo apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE`

And then the apt-get instruction that you already used in your opening message.

`sudo add-apt-repository "deb https://librealsense.intel.com/Debian/apt-repo $(lsb_release -cs) main" -u`

karry-yuan, 2021-11-24:
The add-apt-repository is OK. But librealsense2-dkms can't be installed.
` kumar@nv-tx2:~$ sudo add-apt-repository "deb https://librealsense.intel.com/Debian/apt-repo bionic main" -u
Hit:1 http://mirrors.tuna.tsinghua.edu.cn/ubuntu-ports bionic InRelease        
Hit:2 http://mirrors.ustc.edu.cn/ros/ubuntu bionic InRelease                   
Hit:3 http://mirrors.tuna.tsinghua.edu.cn/ubuntu-ports bionic-updates InRelease
Ign:4 https://librealsense.intel.com/Debian/apt-repo bionic InRelease          
Get:5 https://librealsense.intel.com/Debian/apt-repo bionic Release [2,718 B]  
Get:6 https://librealsense.intel.com/Debian/apt-repo bionic Release.gpg [498 B]
Hit:7 http://mirrors.tuna.tsinghua.edu.cn/ubuntu-ports bionic-backports InRelease
Hit:8 http://mirrors.tuna.tsinghua.edu.cn/ubuntu-ports bionic-security InRelease
Fetched 3,216 B in 6s (534 B/s)                                                
Reading package lists... Done
kumar@nv-tx2:~$ sudo apt-get install librealsense2-dkms
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package librealsense2-dkms `

MartyG-RealSense, 2021-11-24:
I have not seen the **mirrors.tuna.tsinghua.edu.cn** server location before.  If server location is a factor in the error, in that the installation cannot retrieve librealsense2-dkms from those servers, then it may be worth trying to build librealsense from source code instead.  A method for doing so with Jetson boards is described in https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049

karry-yuan, 2021-11-24:
I successfully install the librealsense 2.50.0, and the realsense-viewer can be launched. But the devcie D455 can't be found when I use it with reaalsense-viewer, it says that
` kumar@nv-tx2:~$ realsense-viewer 
 24/11 20:53:24,007 WARNING [547510624272] (ds5-factory.cpp:1152) DS5 group_devices is empty. `
When I launch it with realsense-ros-2.3.2, there is the same problem
`process[camera/realsense2_camera_manager-2]: started with pid [10228]
process[camera/realsense2_camera-3]: started with pid [10229]
[ INFO] [1637758745.890108985]: Initializing nodelet with 6 worker threads.
[ INFO] [1637758746.264148161]: RealSense ROS v2.3.2
[ INFO] [1637758746.264412291]: Built with LibRealSense v2.50.0
[ INFO] [1637758746.264553124]: Running with LibRealSense v2.50.0
[ WARN] [1637758746.354865111]: No RealSense devices were found!
 24/11 20:59:06,354 WARNING [546928845184] (ds5-factory.cpp:1152) DS5 group_devices is empty.`
Another, the firmware release of D455 is 5.13.0.50. How can I solve this problem?

MartyG-RealSense, 2021-11-24:
What build method did you use to successfully install 250.0 please?

karry-yuan, 2021-11-24:
apt install

MartyG-RealSense, 2021-11-24:
And you used the apt install package installation method on the Jetson instruction page?

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md

The packages on the **distribution_linux.md** instruction page are not compatible with Arm devices such as Jetson as they require an x86 / x64 processor like those in desktop and laptop computers.

https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md

karry-yuan, 2021-11-25:
Yes, I used the apt install package installation method on the Jetson instruction page.

MartyG-RealSense, 2021-11-25:
And the camera cannot be detected in the Viewer if you unplug the USB cable and then plug it back in again?

MartyG-RealSense, 2021-12-01:
Hi @karry-yuan  Do you require further assistance with this case, please?  Thanks!

karry-yuan, 2021-12-01:
With your help, I solved this problem successfully. Thank you very much!

MartyG-RealSense, 2021-12-01:
That's great to hear, @karry-yuan - thanks very much for the update!  As you have achieved a solution, I will close this case.  Thanks again!

closed 2021-12-01 by MartyG-RealSense

myxiaojun, 2021-12-03:
how do you solved this problem? I have met the same problem, install librealsense with this tutorial : https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md

>

myxiaojun, 2021-12-03:
how do you solved this problem: ? I have met the same problem, install librealsense with this tutorial : https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md

> 
I successfully install the librealsense 2.50.0, and the realsense-viewer can be launched. But the devcie D455 can't be found when I use it with reaalsense-viewer, it says that
kumar@nv-tx2:~$ realsense-viewer 24/11 20:53:24,007 WARNING [547510624272] (ds5-factory.cpp:1152) DS5 group_devices is empty.
When I launch it with realsense-ros-2.3.2, there is the same problem
process[camera/realsense2_camera_manager-2]: started with pid [10228] process[camera/realsense2_camera-3]: started with pid [10229] [ INFO] [1637758745.890108985]: Initializing nodelet with 6 worker threads. [ INFO] [1637758746.264148161]: RealSense ROS v2.3.2 [ INFO] [1637758746.264412291]: Built with LibRealSense v2.50.0 [ INFO] [1637758746.264553124]: Running with LibRealSense v2.50.0 [ WARN] [1637758746.354865111]: No RealSense devices were found! 24/11 20:59:06,354 WARNING [546928845184] (ds5-factory.cpp:1152) DS5 group_devices is empty.
Another, the firmware release of D455 is 5.13.0.50. How can I solve this problem?

karry-yuan, 2021-12-03:
I can't solve the problem. So I used D435i and reduced the version of librealsense.

myxiaojun, 2021-12-06:
> I can't solve the problem. So I used D435i and reduced the version of librealsense.

I solved "DS5 group_devices is empty." by refresh my system to jetpack 4.6, use SDK-Manager.

MartyG-RealSense, 2021-12-06:
Thanks very much @myxiaojun for your advice to @karry-yuan  :)

karry-yuan, 2021-12-08:
I will try it! Thank you! @myxiaojun

Jsc-Nuaa, 2021-12-11:
Hello, karry-yuan . I used D435i and librealsense v2.50.0 but I met this problem too. "DS5 group_devices is empty"
And I install librealsense by source code. 
Jetson TX2  
jetpack3.1
ubuntu 18.04

MartyG-RealSense, 2021-12-11:
Hi @karry-yuan  Your JetPack version is very old now.  Intel's installation process for building librealsense from source code on Jetson has been tested with JetPack versions as recent as **4.5.1**.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md#building-from-source-using-native-backend

Would it be possible to try updating your JetPack, as Nvidia's JetPack archive states that modern JetPack versions are compatible with TX2.

https://developer.nvidia.com/embedded/jetpack-archive

Jsc-Nuaa, 2021-12-11:
Wow, thanks! I will try to update my JetPack to a newer version.@MartyG-RealSense

Jsc-Nuaa, 2021-12-11:
HI，I have already update the version of JetPack and my version is 4.5.1 now. What should I do next？@MartyG-RealSense

MartyG-RealSense, 2021-12-11:
@Jsc-Nuaa  Are you able to stream depth successfully in the RealSense Viewer tool (realsense-viewer) after your JetPack version is updated to 4.5.1 please?

Jsc-Nuaa, 2021-12-12:
Hi, I solved the "DS5 group_devices is empty"  problem. 
Thanks!@MartyG-RealSense
But I met new problems.
WARNING[548045828112](ds5-motion.cpp:473)IMU Calibration is not available, default intrinsic and extrinsic will be used!

Jsc-Nuaa, 2021-12-12:
[image]

MartyG-RealSense, 2021-12-12:
Hi @Jsc-Nuaa  That's great to hear that the **DS5 group_devices is empty** issue is resolved for you!

The IMU calibration warning is not necessarily a problem.  It is simply suggesting that the IMU motion tracking component of your D435i may not be calibrated, but the IMU is still usable.  

IMU calibration is an optional step.  Once it is performed then an option is unlocked in the Motion Module section of the Viewer to make a correction to IMU data so that the values are adjusted to compensate for known error factors in the IMU component (which does not have in-built calibration).

If you wish to perform it then it can be done with a Python-based IMU calibration tool at the link below.

https://github.com/IntelRealSense/librealsense/tree/master/tools/rs-imu-calibration

An excellent PDF user guide document for the tool is here:

https://dev.intelrealsense.com/docs/imu-calibration-tool-for-intel-realsense-depth-camera

Jsc-Nuaa, 2021-12-12:
Your help is greatly appreciated!@MartyG-RealSense Thanks!

Jsc-Nuaa, 2021-12-13:
Hi,  I met new problems. I tried to use realsense-ros. When I installed ROS Melodic and tried to check the version of opencv. And the version is 4.1.1. I could find opencv3.2. What should I do ?
@MartyG-RealSense 
[image]

Jsc-Nuaa, 2021-12-13:
When I installed ROS Melodic on my laptop, I could success in finding the version of opencv(3.2.0) after "pkg-config opencv --modversion"