realsenseai/librealsense #9996: How to read gyroscope and acceleration

Aman-he, 2021-11-25:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 }                                                   | 
| Firmware Version                   | (05.12.14.50) | 
| Operating System & Version |   Win10  | 
| Platform                                 | PC |
| SDK Version                           |  { 2.48.0 }                          | 
| Language                               |  {python }                         | 
| Segment			        |  {others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
Who can teach me how to read gyroscope and acceleration data with d455? I didn't get the result after checking the API document, or can you write an example for me to understand

MartyG-RealSense, 2021-11-25:
Hi @Aman-he  There is an example of a Python script for accessing gyroscope and acceleration values (gyro and accel) at https://github.com/IntelRealSense/librealsense/issues/3409#issuecomment-471673994

Aman-he, 2021-11-26:
> 

Thank you very much for your help. Now this problem has been solved.

Aman-he, 2021-11-26:
> Hi @Aman-he There is an example of a Python script for accessing gyroscope and acceleration values (gyro and accel) at [#3409 (comment)](https://github.com/IntelRealSense/librealsense/issues/3409#issuecomment-471673994)

Thank you very much for your help. Now this problem has been solved

MartyG-RealSense, 2021-11-26:
That's great to hear, @Aman-he - thanks very much for the update.  As you have achieved a solution, I will close this case.  Thanks again!

closed 2021-11-26 by MartyG-RealSense