realsenseai/librealsense #10000: Poor point cloud quality

L-Carson, 2021-11-26:
Hello, I use the realsenseD435i to find the wall, at first, the cloud points is normal as photo1 showed, but after 1month used later, the cloud points is missing in somewhere as photo2 showed . So please tell what the problem of the camera？

[image]
[image]
[image]
[image]

MartyG-RealSense, 2021-11-26:
Hi @L-Carson  It looks as though Photo 2 may have lower lighting conditions in the location where the camera is being used, making the dot-pattern projection from the camera that is cast onto the surface less visible.  This could be making it more difficult for the camera to analyze the dots for depth information.  If the depth image quality has reduced over the past month then possibly the lighting conditions in the location has become dimmer as winter approaches.

You could first try setting the **Laser Power** option to its maximum value of '360' to increase the visibility of the dots so that it is easier for the camera to analyze the dots for obtaining depth information.

shiguangzqz, 2021-11-27:
> can you tell me how to  set the Laser Power option to its maximum value of '360' by my code ?

MartyG-RealSense, 2021-11-27:
Which programming language are you using, please (Python, C++, C#, etc)

shiguangzqz, 2021-11-27:
> Which programming language are you using, please (Python, C++, C#, etc)

C++，Thanks

MartyG-RealSense, 2021-11-27:
The laser power can be set in C++ using the **RS2_OPTION_LASER_POWER** instruction.  An example script for doing so is provided in the link below, under the **librealsense2** heading.

https://dev.intelrealsense.com/docs/api-how-to#section-controlling-the-laser

shiguangzqz, 2021-11-27:
> The laser power can be set in C++ using the **RS2_OPTION_LASER_POWER** instruction. An example script for doing so is provided in the link below, under the **librealsense2** heading.
> 
> https://dev.intelrealsense.com/docs/api-how-to#section-controlling-the-laser

OK，Thanks，we will try this code to resolve our problem

L-Carson, 2021-12-03:
Hello，if we set the RS2_OPTION_LASER_POWER to '360', does it damage the life of the infrared laser which resulting in shorter camera lifer？

MartyG-RealSense, 2021-12-03:
Hi @L-Carson  I am not aware of a relationship between maximum laser power and the lifespan of the IR emitter component.  The laser can degrade over time though the camera can continue to function, as described by a RealSense team member in https://github.com/IntelRealSense/librealsense/issues/3452#issuecomment-471492320

L-Carson, 2021-12-03:
OK， Thank you very much!

closed 2021-12-15 by MartyG-RealSense