realsenseai/librealsense #9999: Fail to build on MacBook Pro 14 with M1 Pro

gy2256, 2021-11-26:
| Camera Model                       | D455 | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |  MacOS  Monterey| 
| Platform                                 | MacBook Pro 14 (2021)  |
| SDK Version                            | librealsense-2.49.0 | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  Robot | 

### Issue Description
Hello,

I am following the steps from [https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_osx.md](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_osx.md) to build the SDK on my MacBook Pro. I have done some tweaks according to [https://www.yuque.com/docs/share/e97e0cca-7532-424b-b3ca-b69906651779](https://www.yuque.com/docs/share/e97e0cca-7532-424b-b3ca-b69906651779) and added the following
```
set(CMAKE_THREAD_LIBS_INIT "-lpthread")
set(CMAKE_HAVE_THREADS_LIBRARY 1)
set(CMAKE_USE_WIN32_THREADS_INIT 0)
set(CMAKE_USE_PTHREADS_INIT 1)
set(THREADS_PREFER_PTHREAD_FLAG ON)
```
in CMakeLists.txt above the `build` folder. This is the error message I am getting

```
[ 48%] Linking CXX shared library librealsense2.dylib
ld: library not found for -latomic
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [librealsense2.2.49.0.dylib] Error 1
make[1]: *** [CMakeFiles/realsense2.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
HEAD is now at 2f33be817 RELEASE-NOTES: synced
[ 48%] No update step for 'libcurl'
[ 49%] No patch step for 'libcurl'
[ 49%] Performing configure step for 'libcurl'
-- The C compiler identification is AppleClang 13.0.0.13000029
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - failed
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc - broken
CMake Error at /opt/homebrew/Cellar/cmake/3.22.0/share/cmake/Modules/CMakeTestCCompiler.cmake:69 (message):
  The C compiler

    "/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /Users/du/development_local/intel_realsense/librealsense-2.49.0/build/libcurl/src/libcurl-build/CMakeFiles/CMakeTmp

    Run Build Command(s):/usr/bin/make -f Makefile cmTC_783be/fast && /Applications/Xcode.app/Contents/Developer/usr/bin/make  -f CMakeFiles/cmTC_783be.dir/build.make CMakeFiles/cmTC_783be.dir/build
    Building C object CMakeFiles/cmTC_783be.dir/testCCompiler.c.o
    /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc   -pedantic -g -D_DEFAULT_SOURCE   -mfpu=neon -mfloat-abi=hard -ftree-vectorize -latomic -pthread  -arch arm64 -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX12.0.sdk -MD -MT CMakeFiles/cmTC_783be.dir/testCCompiler.c.o -MF CMakeFiles/cmTC_783be.dir/testCCompiler.c.o.d -o CMakeFiles/cmTC_783be.dir/testCCompiler.c.o -c /Users/du/development_local/intel_realsense/librealsense-2.49.0/build/libcurl/src/libcurl-build/CMakeFiles/CMakeTmp/testCCompiler.c
    clang: warning: -latomic: 'linker' input unused [-Wunused-command-line-argument]
    clang: warning: argument unused during compilation: '-mfpu=neon' [-Wunused-command-line-argument]
    clang: warning: argument unused during compilation: '-mfloat-abi=hard' [-Wunused-command-line-argument]
    Linking C executable cmTC_783be
    /opt/homebrew/Cellar/cmake/3.22.0/bin/cmake -E cmake_link_script CMakeFiles/cmTC_783be.dir/link.txt --verbose=1
    /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc    -pedantic -g -D_DEFAULT_SOURCE   -mfpu=neon -mfloat-abi=hard -ftree-vectorize -latomic -pthread  -arch arm64 -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX12.0.sdk -Wl,-search_paths_first -Wl,-headerpad_max_install_names CMakeFiles/cmTC_783be.dir/testCCompiler.c.o -o cmTC_783be
    ld: library not found for -latomic
    clang: error: linker command failed with exit code 1 (use -v to see invocation)
    make[4]: *** [cmTC_783be] Error 1
    make[3]: *** [cmTC_783be/fast] Error 2

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:48 (project)

-- Configuring incomplete, errors occurred!
See also "/Users/du/development_local/intel_realsense/librealsense-2.49.0/build/libcurl/src/libcurl-build/CMakeFiles/CMakeOutput.log".
See also "/Users/du/development_local/intel_realsense/librealsense-2.49.0/build/libcurl/src/libcurl-build/CMakeFiles/CMakeError.log".
make[2]: *** [libcurl/src/libcurl-stamp/libcurl-configure] Error 1
make[1]: *** [CMakeFiles/libcurl.dir/all] Error 2
make: *** [all] Error 2
```

MartyG-RealSense, 2021-11-26:
Hi @gy2256  Building the RealSense SDK on a modern Mac with an Apple Silicon M1 processor is possible but can be more complicated than a build for the older Intel-based models.  the case https://github.com/IntelRealSense/librealsense/issues/8090 is an excellent hub for finding solutions to M1 build problems.

I note though that you are using the new MacOS Monterey.  USB device problems in general (not just with RealSense cameras) have been reported on Monterey on both Intel and M1 based computers, whereas RealSense worked with earlier MacOS versions such as Catalina and Big Sur.  This is discussed in https://github.com/IntelRealSense/librealsense/issues/9916 
 
However, you are not experiencing errors typically related to the Monterey SUB problem such as **failed to claim usb interface** and so it is more likely to be a build-related problem.  When debugging, a good rule is to start the investigation at the first error, as subsequent errors may be resulting from a cascade of problems caused by the first error and so disappear once the initial error is solved.  

Regarding the initial error **ld: library not found for -latomic ... clang: error: linker command failed with exit code 1 (use -v to see invocation)**: a RealSense M1 user in https://github.com/IntelRealSense/librealsense/issues/8090#issuecomment-829339582 experienced this.  

It looks as though that user, who subsequently achieved a successful build, may have followed the advice given to them to another user to build using the alternative instruction guide in the link below.

https://dev.intelrealsense.com/docs/macos-installation-for-intel-realsense-sdk

gy2256, 2021-11-26:
Thanks for the reply. I realize that should not use `make -j8` to build the project. Instead, I need to use Xcode to build the project. I use the following command before building it in Xcode. 

```
cmake .. -DBUILD_PYTHON_BINDINGS:bool=true -DFORCE_LIBUVC=true -DFORCE_RSUSB_BACKEND=true -DBUILD_NETWORK_DEVICE=ON -DBUILD_EXAMPLES=true -DBUILD_WITH_OPENMP=false -DHWM_OVER_XU=false -G Xcode -DOPENSSL_ROOT_DIR=/opt/homebrew/opt/openssl -DOPENSSL_LIBRARIES=/opt/homebrew/opt/openssl/lib
```

closed 2021-11-26 by gy2256

MartyG-RealSense, 2021-11-26:
Excellent news that you found a solution, @gy2256 - thanks very much for the update and for sharing your method!

Obrepal, 2021-12-07:
Hi, I do not have much experience with CMake and M1 is there easy way to fix this? (I checked other issues but nothing was close or I missed them)
-- Found PythonInterp: /opt/homebrew/bin/python3.9 (found version "3.9.9") 
-- Found PythonLibs: /opt/homebrew/opt/python@3.9/Frameworks/Python.framework/Versions/3.9/lib/libpython3.9.dylib
-- Performing Test HAS_FLTO
-- Performing Test HAS_FLTO - Failed
-- Performing Test HAS_FLTO_THIN
-- Performing Test HAS_FLTO_THIN - Failed
-- Found Python: /opt/homebrew/Frameworks/Python.framework/Versions/3.9/bin/python3.9 (found version "3.9.9") found components: Interpreter Development Development.Module Development.Embed 
CMake Error at /opt/homebrew/Cellar/cmake/3.22.0/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find Threads (missing: Threads_FOUND)
Call Stack (most recent call first):
  /opt/homebrew/Cellar/cmake/3.22.0/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:594 (_FPHSA_FAILURE_MESSAGE)
  /opt/homebrew/Cellar/cmake/3.22.0/share/cmake/Modules/FindThreads.cmake:238 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  examples/measure/CMakeLists.txt:9 (find_package)

-- Configuring incomplete, errors occurred!

MartyG-RealSense, 2021-12-08:
Hi @Obrepal  Which MacOS version are you using (for example, Catalina, Big Sur or Monterey)

Obrepal, 2021-12-08:
> Hi @Obrepal Which MacOS version are you using (for example, Catalina, Big Sur or Monterey)

I am using Monterey (downloaded it yesterday, with Big sur I had the same problem and thought that might help)
Also I have the newest Xcode Version 13.1 (13A1030d)

I downloaded the repo with this command: 
`git clone https://github.com/IntelRealSense/librealsense`

and here the whole error output

`build % cmake .. -DBUILD_EXAMPLES=true -DBUILD_WITH_OPENMP=false -DHWM_OVER_XU=false
-- Checking internet connection...
-- Internet connection identified
-- Info: REALSENSE_VERSION_STRING=2.50.0
-- Setting Unix configurations
-- using RS2_USE_LIBUVC_BACKEND
CMake Error at /opt/homebrew/Cellar/cmake/3.22.0/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find Threads (missing: Threads_FOUND)
Call Stack (most recent call first):
  /opt/homebrew/Cellar/cmake/3.22.0/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:594 (_FPHSA_FAILURE_MESSAGE)
  /opt/homebrew/Cellar/cmake/3.22.0/share/cmake/Modules/FindThreads.cmake:238 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  examples/measure/CMakeLists.txt:9 (find_package)`

I don't want to make it too long so I attached two files mentioned after cmake .. 

 

[CMakeError.log](https://github.com/IntelRealSense/librealsense/files/7678096/CMakeError.log)
[CMakeOutput.log](https://github.com/IntelRealSense/librealsense/files/7678097/CMakeOutput.log)

MartyG-RealSense, 2021-12-08:
Thanks very much @Obrepal for the information.  If you also had the same problem with Big Sur then your issue likely isn't related to the USB problems that have been experienced by upgraders to Monterey, as discussed in https://github.com/IntelRealSense/librealsense/issues/9916

Obrepal, 2021-12-09:
So I should make  a new issue?

MartyG-RealSense, 2021-12-09:
@Obrepal  Yes, please post a new issue using the green **New Issue** button on the front page of this forum at the link below and include your original message from https://github.com/IntelRealSense/librealsense/issues/9999#issuecomment-988188997

https://github.com/IntelRealSense/librealsense/issues/