realsenseai/librealsense #10001: Calibration Failed!

wsjlmq, 2021-11-28:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D415  | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (10)  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC |
| SDK Version                            |  2.49 }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
[image]

1.When I use _realsense.viwer_ , 
[image]
then, I begin _on_chip_calibration_it fails, as shown below
[image]

2. so I try to use Dynamic calibration tool , it can never eliminate all the blue squares! so this failed too.
Why is this happening?

MartyG-RealSense, 2021-11-28:
Hi @wsjlmq   Could you first check your Windows camera privacy settings please using the instructions in https://github.com/IntelRealSense/librealsense/issues/3381#issuecomment-503424789 in order to confirm that access to the camera hardware and permission for applications to access the camera are both set to On.

wsjlmq, 2021-11-29:
> 

Hi @MartyG-RealSense ，thank you very much for your suggestion.
I immediately checked the permissions of the camera.I'm pretty sure that permissions are both set to On.Unfortunately,calibration still failed,same as before.
So I contacted the official technician, who suggested that I return to the factory for maintenance.At the moment, I still don't know what happened to my camera.

MartyG-RealSense, 2021-11-29:
Hi @wsjlmq  Intel do not provide a repair service for RealSense cameras.  Cameras can be returned to the retailer that they were purchased from for a replacement unit.  As it seems from the images above though that the camera hardware is capable of streaming depth, I would not myself recommend a return yet at this point in your support case.

Would it be possible for you to next try installing the latest SDK version **2.50.0** to see whether this resolves your problem?  If you installed 2.49.0 from the **Assets** file list on the SDK 'Releases' page then installing 2.50.0 with the same method will overwrite the 2.49.0 installation.  

https://github.com/IntelRealSense/librealsense/releases/tag/v2.50.0 

[image]

wsjlmq, 2021-11-29:
hi @MartyG-RealSense , you are right! Retailers may  replace  this camera.
Thank you for your suggestion!i will install the latest SDK version 2.50.0 ,and try to align again.

wsjlmq, 2021-11-29:
hi @MartyG-RealSense , it can not work.
[image]

MartyG-RealSense, 2021-11-29:
Okay, thanks very much for testing with the new version.  Can the camera stream depth and RGB normally when the streams are enabled in the options side-panel?

wsjlmq, 2021-11-29:
the camera stream depth and RGB can work normally.just can not align!
[image]
and the camera stream depth has ore noise points than before.

MartyG-RealSense, 2021-11-29:
It is encouraging to confirm that depth and RGB are both able to stream.

The 2D mode of the RealSense Viewer (shown in your image above) does not have a stream alignment function.  You can though map RGB onto depth in the Viewer's 3D point cloud mode.  Go into 3D mode with Depth and RGB both disabled.  Then enable the Depth stream first and RGB second.  The RGB should then automatically map onto the point cloud's depth coordinates.

In regard to the depth image quality, black clothing is not a good depth sensing target.  This is because it is a general physics principle (not specific to RealSense) that dark grey and black color surfaces absorb light, making it difficult for the camera to analyze such surfaces for depth detail.

A surface with low texture detail or no texture such as a plain wall may also be difficult for the camera to read unless the **IR Emitter** of the camera is enabled in order to project a semi-random dot pattern onto the surface.  The camera can then use these dots to generate depth data.  Can you confirm please in the **Controls** section of the 'Stereo Module' side-panel options that the 'Emitter Enabled' option has been set to **Laser** in its drop-down menu to enable the dot pattern projection?

[image]

wsjlmq, 2021-11-29:
the 'Emitter Enabled' option has been set to Laser,it works.The quality of map depth is improved.
[image]
but calibration failed again.

MartyG-RealSense, 2021-11-29:
Judging by the quality of the above depth image, your camera may not need calibration at this time.  It is important though to be able to access calibration when you do actually need to perform it.

Is there a good level of lighting in the location that you are performing calibration in?  

We can also check for whether there may be a corruption in the **calibration table** stored inside the camera.  If there is then performing a 'gold reset' procedure to reset the camera to its factory-new defaults should correct a corruption.

A camera can be reset to factory defaults using a tool called **CustomRW** that is packaged with the RealSense Dynamic Calibration tool that you have already installed.  You can input the command below into the Windows **Command Prompt** interface to launch CustomRW and activate the Gold Reset:

Intel.Realsense.CustomRW.exe -g

wsjlmq, 2021-11-29:
I try to provide a good level of lighting in the location that I am performing calibration in.I tried calibration indoors and provided sufficient lighting.The  'gold reset' method you provide is very important.When I need it, I will adopt it immediately.

MartyG-RealSense, 2021-11-29:
The Dynamic Calibration tool can be downloaded for Windows from the link below.

https://www.intel.com/content/www/us/en/download/645988/intel-realsense-d400-series-dynamic-calibration-tool.html

wsjlmq, 2021-11-29:
Thank you very much. I have downloaded it.You are really a helpful friend!

MartyG-RealSense, 2021-11-29:
You are very welcome  :)

wsjlmq, 2021-11-30:
Hi@MartyG-RealSense,I want to use realsense camera to display color map and depth map through c# language, and finally get 3D data.The link below is the only official information I can find.
[https://github.com/IntelRealSense/librealsense/blob/master/wrappers/csharp/Documentation/cookbook.md#get-video-stream-intrinsics](url)
So where can I find a complete example of developing realsense in c# language?

MartyG-RealSense, 2021-11-30:
The amount of official documentation for C# is limited mostly to the cookbook and the C# wrapper page and its **Tutorial** section of example programs (linked to below) , though there are a greater number of scripting resources available on this librealsense support forum.   

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/csharp/tutorial

Of the examples listed on the above link, the **Capture** one should meet your needs, as it captures depth and color and renders it to side-by-side windows.

When searching for resources on the librealsense forum, I recommend starting each search term with the word **librealsense c#**, followed by the subject that you are researching.  For example, **librealsense c# point cloud**.  By including the word 'librealsense' in the search term, you greatly increase the chances of the search engine providing results from the librealsense support forum.

wsjlmq, 2021-11-30:
I can’t thank you enough.

MartyG-RealSense, 2021-12-06:
Hi @wsjlmq  Do you require further assistance with this case, please?  Thanks!

wsjlmq, 2021-12-06:
Hi @MartyG-RealSense 
I won't pay much attention to this problem for the time being. You can close it.
Thank you for you help!

MartyG-RealSense, 2021-12-06:
Thanks very much @wsjlmq for the update.  As you suggested, I will close the issue.  Thanks again!

closed 2021-12-06 by MartyG-RealSense

wsjlmq, 2021-12-06:
Hi @MartyG-RealSense .I want use C# language to align depth and color images,but no suitable reference was found.Can you help me?

MartyG-RealSense, 2021-12-06:
https://github.com/IntelRealSense/librealsense/issues/9765#issuecomment-919249014 has another C# align script that was shared by a RealSense user.