realsenseai/librealsense #10004: CAMERA D415 x UNITY / Only one camera is SHOWN

Lewisuehjo, 2021-11-29:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D415 } | 
| Firmware Version                   | lastest | 
| Operating System & Version |   {Win (10)   | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { lastest }                          | 
| Language                            |  {C/C#/unity }                          | 
| Segment			|  {VR/AR }                          | 

### Issue Description

Hi everybody ! 

I'm actualy working on a project where I have to setup 2 x D415 in an unity scene (PointCloudDepthAndColor) from the Intel RealSense Package.

My computer detects both cameras, I duplicated the "RsDevice", I set up the serial numbers of the camera.

Now, the problem that I'm having is : When I launch the scene, only one camera is apparently detected and shown. 

I followed all the tips that you gave in 2018 but nothing is working.

Hoping someone finds my call for help.

Thanks for reading me.

I attached some picture to understand better (ofc the S/N is fake)
[image]

MartyG-RealSense, 2021-11-29:
Hi @Lewisuehjo  The approach that you have used - creating multiple RsDevice components that each have their own unique camera serial number - is the appropriate method to use.

The link below has a detailed guide shared by a RealSense user of the procedure that worked for them in creating a Unity multicam setup using this RsDevice duplication technique.

https://support.intelrealsense.com/hc/en-us/community/posts/360047049173-Community-contribution-for-using-multiple-cameras-in-RealSense-Unity-wrapper

Lewisuehjo, 2021-11-29:
Hi @MartyG-RealSense 

Thanks for the advice, I'll try.

I'll keep you informed

MartyG-RealSense, 2021-11-29:
Given that the **PointCloudDepthAndColor** example was designed for displaying data from a single camera, it may be worth checking the RsDevice object for the second camera to see whether it is detected and operating correctly even if data from that camera is not represented on the pointcloud.

Lewisuehjo, 2021-11-29:
I think they are both detected cause there is a little red LED turned on when I launch the scene on Unity.

What Scene is use to display more than 1 camera ?

MartyG-RealSense, 2021-11-29:
All of the scenes supplied with the RealSense Unity wrapper were originally designed for a single camera.  Interestingly though, I found a YouTube video where somebody created a two-camera setup with **PointCloudDepthAndColor** by creating two RsDevice objects.  Though the video does not provide much explanation, my understanding is that the video creator was able to generate on-screen imagery from two cameras separately but was not able to perform 'registration' on them to merge them together into a single combined point cloud.

https://www.youtube.com/watch?v=-JVnC8xxhuA

[image]

Lewisuehjo, 2021-12-01:
After some try, I did it ! It works, thanks a lot.

Now, if its possible, I would to know if its possible to calibrate and synchronize these cameras. Do you have some advice ?

Ty in advance

MartyG-RealSense, 2021-12-01:
The link below details a Unity project for calibrating RealSense 400 Series cameras together using **Apriltag** image tags.

https://github.com/ababilinski/realsense-apriltags-calibration-unity

MartyG-RealSense, 2021-12-13:
Case closed due to no further comments received.

closed 2021-12-13 by MartyG-RealSense

Lewisuehjo, 2021-12-14:
Hi @MartyG-RealSense, sorry I was in vacation the last week.

I got this error when I compile the calibration in unity  : "Asset Packages/babilinapps.realsense.downloader/CHANGELOG.md has no meta file, but it's in an immutable folder. The asset will be ignored."

It doesn't show me what it should show me like on the github page/example

MartyG-RealSense, 2021-12-14:
Are you using the Unity Apriltag calibration tag project mentioned earlier in this discussion, please?

https://github.com/ababilinski/realsense-apriltags-calibration-unity

It looks as though this project is designed to be loaded into Unity as a complete project.  This could be done with the following procedure.

***************

1.  On the project GitHub page, left-click on the green **Code** button and select the **Download Zip** button to download the project as a compressed zip file.

[image]

2.  Extract the project folder inside the zip file.

[image]

3.  Open the project in Unity - for example, by adding the folder to the Projects list of the Unity Hub interface.

************

Lewisuehjo, 2021-12-17:
Thanks for the answer, yea I did that, but only one camera is display and can be calibrate look this picture : 
[image]

[image]

MartyG-RealSense, 2021-12-17:
Which Unity version are you using, please?  The project https://github.com/keijiro/jp.keijiro.apriltag that the ababilinsky calibration project was adapted from was tested with Unity 2020.2, as described in the link below.

https://github.com/keijiro/jp.keijiro.apriltag#system-requirements

Lewisuehjo, 2021-12-17:
I'm using the 2020.3.4, its the problem ?

Lewisuehjo, 2021-12-17:
I gonna try

MartyG-RealSense, 2021-12-17:
I would think that using 2020.3 instead of 2020.2 would not be as much of a problem than if you were using an older major Unity branch such as 2019.4.

Lewisuehjo, 2021-12-17:
yea that didn't solve the problem...
[image]

MartyG-RealSense, 2021-12-17:
Do you have a recent version of the RealSense SDK's Unity wrapper package installed in your project, please?  The ababilinsky project instructions do not specify the need to have the SDK Unity package installed.  The last time that I saw the **Rethrow as Exception.  Couldn't resolve requests** error was in https://github.com/IntelRealSense/librealsense/issues/7068 and installing a more recent version of the SDK's Unity package corrected the error in that particular case.

You can find the appropriate Unity package for a particular SDK version by going to the SDK **Releases** page and downloading a package file called **Intel.RealSense.unitypackage**, then importing that package into your Unity project.

https://github.com/IntelRealSense/librealsense/releases

[image]

Lewisuehjo, 2021-12-17:
[image]

I already import this one... and that didn't work...

Lewisuehjo, 2021-12-17:
I don't understand how it is even working on his post... the unity project do not had 2 pipeline/RsDevice so how can 2 camera could be stream ?

MartyG-RealSense, 2021-12-17:
Yes, I have been thinking the same about RsDevice.  Usually in a multiple camera RealSense Unity project, a separate RsDevice component has to be set up for each individual camera, as described in the link below.

https://support.intelrealsense.com/hc/en-us/community/posts/360047049173-Community-contribution-for-using-multiple-cameras-in-RealSense-Unity-wrapper

I have carefully studied the assets of the project but cannot see how it integrates with the RealSense SDK.