realsenseai/librealsense #10005: realsense-viewer software rendering/processing incorrect texture

bmegli, 2021-11-29:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   | 05.13.00.50 | 
| Operating System & Version |   Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  4.15.0-162-generic                                         | 
| Platform                                 | PC  |
| SDK Version                            | 2.50.0-0~realsense0.6127 (apt package and librealsense version)                          | 
| Language                            |  realsense-viewer                          | 

With GLSL for rendering and processing

[image]

Without GLSL rendering and processing

[image]

GLSL variant only shows data where both depth and rgb exist.
Software variant renders part where color data doesn't exit with incorrect texture.

I am mainly reporting it because it is cause for other issue.

MartyG-RealSense, 2021-11-29:
Hi @bmegli   I note that your Viewer has a warning displayed about multiple udev rules.

[image]

This warning may occur if the librealsense SDK has been built from both packages and source code on the same computer.  The warning should disappear if the rule at the location **/etc/udev/rules.d/** is deleted from your computer, as described in https://github.com/IntelRealSense/librealsense/issues/6153#issuecomment-605687547

bmegli, 2021-11-29:
Hi @MartyG-RealSense,

>  I note that your Viewer has a warning displayed about multiple udev rules.

Thanks for pointing that out. To check if it affects the outcome I just confirmed on another machine:
- where I have packaged librealsense version only
- the same effect manifests (different rendering in software & GLSL)
- machine is:
  - Ubuntu 20.04
  - librealsense 2.49.0.0~realsense0.5307 (`apt-cache policy librealsense2`)

MartyG-RealSense, 2021-11-29:
I ran my own tests on a D435 with and without the GLSL options.  The results were similar to your own in terms of the more restricted field of view when the GLSL options are enabled.

**GLSL options enabled**

[image]

**GLSL options disabled**

[image]

You do not have to have both options enabled or disabled.  You can choose to disabled one or the other too (Rendering or Processing).  Having Rendering enabled and Processing disabled produced the narrow field of view, whilst having Rendering disabled and Processing enabled produced the wider field of view (at the cost of a possibly higher CPU utilization).

If the wide view is your preference, you could therefore try exporting a ply with Rendering disabled and Processing enabled to see how it affects the results.

bmegli, 2021-11-29:
Thanks for testing it out.

I am aware of some differences in libreleanse processing/rendering paths (CPU/SSE/GLSL/CUDA).

I wouldn't even bother reporting that if it wasn't for #10006.

> If the wide view is your preference, you could therefore try exporting a ply with Rendering disabled and Processing enabled to see how it affects the results.

The problem is mainly with the wide view:
- which has incorrect texture data where texture doesn't exist
- is used for for exporting to ply regardless of what user sees
  - so user possibly sees narrow FOV without incorrect texture
  - but exports wide FOV with incorrect texture

bmegli, 2021-12-02:
Yes, thank, I could probably use the script.

It's either:
- align depth to color before converting to point cloud and export
- or UV map to depth and manually filter out points with UV coordinates outside [0, 1] range during export

MartyG-RealSense, 2021-12-08:
Hi @bmegli  Do you require further assistance with this case, please?  Thanks!

bmegli, 2021-12-09:
No, I know what is happening and how to workaround the problems. Thanks.

MartyG-RealSense, 2021-12-09:
Excellent to hear that you have a workaround, @bmegli - as you do not require further assistance with this case then I will close it.  Thanks again!

closed 2021-12-15 by MartyG-RealSense