realsenseai/librealsense #10006: realsense-viewer export to ply inconsistent with visible point cloud (and wrong)

bmegli, 2021-11-29:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

Different point from visible is exported, specifically with partially incorrect texture.

The reason is #10005 
- software processing seems to be used when exporting to ply
- software processing incorrectly handles part where depth/RGB data doesn't overlap

To see:
- export point cloud to ply (Meshing and Normals disabled)
- load the ply to e.g. Meshlab
- compare exported point cloud with realsense-viewer
  - GLSL processing/rendering enabled
  - disabled

Compare below meshlab imported ply with #10005 software rendering/processing.

[image]

MartyG-RealSense, 2021-12-01:
Hi @bmegli  As this case is connected to your other case at https://github.com/IntelRealSense/librealsense/issues/10005 - are you ready to discuss this case now or would you prefer that https://github.com/IntelRealSense/librealsense/issues/10005 is resolved first, please?  Thanks!

bmegli, 2021-12-02:
I have no feelings either way.

This issue is what users face:
- exported ply (depth + color only)
- is different from what they see by default
  - by default GLSL is seen
  - with narrower FOV
  - and no area with incorrect texture

#10005 is likely a reason why this happens.

MartyG-RealSense, 2021-12-02:
As far as I am aware, this issue with GLSL and ply export from the RealSense Viewer has not been raised by other RealSense users.  So it may be worth revisiting the issue a future date if other users leave comments on this case that have the same question that you had.

MartyG-RealSense, 2021-12-08:
Case closed due to no further comments received.  The issue can be revisted at a future date if other RealSense users leave comments regarding similar questions.

closed 2021-12-08 by MartyG-RealSense