realsenseai/librealsense #10008: Cant seem to figure this error out

IJdo, 2021-11-29:
| | |
|---|---|
|**librealsense**|2.49.0 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
Program terminated due to an unrecoverable SEH exception:
Access Violation!
```

MartyG-RealSense, 2021-11-29:
Hi @IJdo  Would it be possible for you to test whether updating to the current RealSense SDK version **2.50.0** using the **Intel.RealSense.SDK-WIN10-2.50.0.3785.exe** installer file in the 'Assets' file list resolves the error, please?

https://github.com/IntelRealSense/librealsense/releases/tag/v2.50.0

IJdo, 2021-11-30:
@MartyG-RealSense That fixed it! thx!

MartyG-RealSense, 2021-11-30:
That's excellent to hear, @IJdo - thanks very much for the update   :)

closed 2021-12-06 by MartyG-RealSense