realsenseai/librealsense #10011: L515 - Can't get range over 3.5m

xyxegle, 2021-11-30:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
|Serial Number|f1062522|
| Camera Model                       | L515  | 
| Firmware Version                   | 1.05.08.01 | 
| Operating System & Version |   Win10/Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |                                      | 
| Platform                                 | x86PC |

### Issue Description
<Describe your issue / question / feature request / etc..>
My L515 was still working yesterday but suddenly I can't detect objects four meters away. I didn't change any settings, and the sensors didn't suffer any damage.  I rebooted the firmware but nothing improved.Would you please tell me how did you solve this problem?Any help would be greatly appreciated.
[image]

Thanks.

MartyG-RealSense, 2021-11-30:
Hi @xyxegle  First, please try setting the Preset to **Low Ambient Light**.  This setting uses maximum laser power like the Max Range setting does but takes account of low ambient lighting if the lighting level has reduced.

[image]

Under the **Depth Visualization** category of the options side-panel of the Viewer, also check that the **Max Distance** setting is at its default of '16'.

[image]

xyxegle, 2021-11-30:
Thank you for your reply. I just tried it again. I found that the depth data returned to normal without adjusting the parameters.
I remember that I had been using it from last night until today. Finally, it overheated and the data became abnormal. Anyway, thank you very much for your careful reply to my question, and I hope RealSense can provide better sensors soon.

MartyG-RealSense, 2021-11-30:
Thanks very much @xyxegle for the update - it's excellent to hear that your camera is operating normally.

closed 2021-12-07 by MartyG-RealSense