realsenseai/librealsense #10015: lSense error calling rs2_pipeline_start_with_config(pipe:0x55c729ea9ae0, config:0x55c729ea9f60):     xioctl(VIDIOC_S_FMT) failed Last Error: Device or resource busy

Ericloving, 2021-12-01:
Mycode is here!

#include <iostream>
#include <algorithm> 
#include <boost/date_time/posix_time/posix_time.hpp>
#include <boost/thread/thread.hpp>
#include <string>
//#include "stdafx.h"
#include <map>
#include <mutex>                    // std::mutex, std::lock_guard
#include <cmath>                    // std::ceil

// Intel Realsense Headers
#include <librealsense2/rs.hpp> // Include RealSense Cross Platform API
#include "example.hpp"   

// PCL Headers
#include <pcl/io/pcd_io.h>
#include <pcl/point_types.h>
#include <pcl/filters/passthrough.h>
#include <pcl/common/common_headers.h>
#include <pcl/features/normal_3d.h>
#include <pcl/visualization/pcl_visualizer.h>
#include <pcl/console/parse.h>
#include <boost/thread/thread.hpp>
#include <pcl/io/io.h>
#include <pcl/visualization/cloud_viewer.h>

// Include OpenCV API
#include <opencv2/opencv.hpp>   
#include "cv-helpers.hpp"

using namespace std;
using namespace cv;

typedef pcl::PointXYZRGB RGB_Cloud;
typedef pcl::PointCloud<RGB_Cloud> point_cloud;
typedef point_cloud::Ptr cloud_pointer;
typedef point_cloud::Ptr prevCloud;
boost::shared_ptr<pcl::visualization::PCLVisualizer> viewer1(new pcl::visualization::PCLVisualizer("3D Viewer"));////全局共享指针
boost::shared_ptr<pcl::visualization::PCLVisualizer> viewer2(new pcl::visualization::PCLVisualizer("3D Viewer"));////全局共享指针

////声明
std::tuple<int, int, int> RGB_Texture(rs2::video_frame texture, rs2::texture_coordinate Texture_XY);
cloud_pointer PCL_Conversion(const rs2::points& points, const rs2::video_frame& color);
//boost::shared_ptr<pcl::visualization::PCLVisualizer> rgbVis(pcl::PointCloud<pcl::PointXYZRGB>::ConstPtr cloud);

const std::string no_camera_message = "No camera connected, please connect 1 or more";
const std::string platform_camera_name = "Platform Camera";

class device_container
{
    // Helper struct per pipeline
    struct view_port
    {
        std::map<int, rs2::frame> frames_per_stream;
        rs2::colorizer colorize_frame;
        texture tex;
        rs2::pipeline pipe;
        rs2::pipeline_profile profile;
    };

public:

    void enable_device(rs2::device dev)
    {
        std::string serial_number(dev.get_info(RS2_CAMERA_INFO_SERIAL_NUMBER));
        std::lock_guard<std::mutex> lock(_mutex);

        if (_devices.find(serial_number) != _devices.end())
[... 424 lines cut ...]
        pcl::PassThrough<pcl::PointXYZRGB> Cloud_Filter; // Create the filtering object
        Cloud_Filter.setInputCloud(cloud);           // Input generated cloud to filter
        Cloud_Filter.setFilterFieldName("z");        // Set field name to Z-coordinate
        Cloud_Filter.setFilterLimits(0.0, 1.0);      // Set accepted interval values
        Cloud_Filter.filter(*newCloud);              // Filtered Cloud Outputted
        */
    }

    cout << "Exiting Program... " << endl;
    return EXIT_SUCCESS;
    
}

catch (const rs2::error & e)
{
    std::cerr << "RealSense error calling " << e.get_failed_function() << "(" << e.get_failed_args() << "):\n    " << e.what() << std::endl;
    return EXIT_FAILURE;
}
catch (const std::exception& e)
{
    std::cerr << e.what() << std::endl;
    return EXIT_FAILURE;
}

realsense2.47.0 + ubuntu 18.04 , I build the library with the sources. and the CMakelists files is offered officially. Can someone know what's the problem with the error?

MartyG-RealSense, 2021-12-01:
Hi @Ericloving   There is a lot of information in this script to consider.  As it is a device busy error though, the logical place to begin the investigation would be the emplace instruction directly after the pipeline start instruction.

The lines quoted below seem to be sourced from a script at https://github.com/IntelRealSense/librealsense/issues/1527 that was an unsuccessful attempt to pause a live camera stream (which may be what the phrase 'hold it internally' is referring to).

/```
/ Hold it internally
 _devices.emplace(serial_number, view_port{ {},{},{}, p, profile });
```

The line of the RealSense SDK's **rs-multicam** C++ example program linked to below shows how it handles the emplace.

`pipelines.emplace_back(pipe)`

https://github.com/IntelRealSense/librealsense/blob/master/examples/multicam/rs-multicam.cpp#L33

MartyG-RealSense, 2021-12-06:
Hi @Ericloving  Do you require further assistance with this case, please?  Thanks!

Ericloving, 2021-12-08:
> 

but  I still cant solve this problem...

MartyG-RealSense, 2021-12-08:
I can see in your script above that you have already tried implementing an error 'catch' in your C++ script.  That would be the usual way to check whether a camera is busy.

Going through the code above, you have two sections where the streams are being enabled.

[image]

[image]

The RealSense SDK has a set of rules called the **Multi-Streaming Model** (in the link below) that describe how the camera sensors can be accessed by processes and applications.  Basically, once a particular sensor on a particular camera is 'claimed' when the pipeline starts then another process or application cannot start that sensor until the claim on it is released.

https://github.com/IntelRealSense/librealsense/blob/master/doc/rs400_support.md#multi-streaming-model

So if the streams were claimed by one of the two pipeline starts in the images above then when the other pipeline start tried to start those streams afterwards, they would be 'busy' because they were already active.

closed 2021-12-19 by MartyG-RealSense